Arduino-IRremote/Arduino-IRremote #804: My codes are off by 20000 (HEX)

Exovel-Productions, 2021-02-21:
If you have handling problems or questions, consider to open a discussion https://github.com/Arduino-IRremote/Arduino-IRremote/discussions instead of an issue.

### Board
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] ATtiny3217 board (TinyCore)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266
* [ ] Teensy board
* [ ] Other - please specify

### IDE
* [x ] Arduino IDE

### Protocol
* [x ] NEC
* [ ] Panasonic
* [ ] RC5, RC6
* [ ] Samsung
* [ ] Sharp
* [ ] Sony
* [ ] Whynter
* [ ] Other - please specify

Please delete all unchecked lines above :-)

**Code Block:**
```

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

### Current behavior

#include <IRremote.h>

// Define sensor pin
const int RECV_PIN = 4;

// Create IR Send Object
IRsend irsend;

// Define IR Receiver and Results Objects
IRrecv irrecv(RECV_PIN);
decode_results results;

void setup(){
  // Serial Monitor @ 9600 baud
  Serial.begin(9600);
  // Enable the IR Receiver
  irrecv.enableIRIn();
}

void loop(){
  if (irrecv.decode(&results)){
    // Print Code in HEX
        Serial.println(results.value, HEX);
        irrecv.resume();
  }
}

>> 2FF00FF (power on)

Using:
irsend.sendNEC(0x2FD00FF, 32); // On

To get my device to respond to the code I needed to transmit code to:  2FD00FF,
a difference of 20000 (HEX)

I'm not great on the math so I didn't know if something was shifted.

### Expected behavior

### Error output

### Additional context

**checklist:**
- [x] I have **read** the README.md file thoroughly.
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) is used.
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [x] The title of the issue is helpful and relevant.
- [x] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

ArminJo, 2021-02-21:
Do you really have read the README.md file thoroughly???

DjordjeMandic, 2021-02-23:
Are you sure your device does not use 16bit addressing instead of 8bit or vice versa?
Normal NEC codes have 8bit address but Extended NEC codes have 16bit address.

Look at this 8bit address NEC code
[image]

Now this is 16bit address NEC code decoded as 8bit code
[image]

Now I switched decoding to 16bit address
[image]

Try sending `2FF00FF` in both 16bit and 8bit addressing mode

ArminJo, 2021-03-03:
@DjordjeMandic 
I included 2 of your fantastic images in the [readme](https://github.com/Arduino-IRremote/Arduino-IRremote#nec-encoding). Is this OK for you?
Which tool generates such images?

closed 2021-03-03 by ArminJo

DjordjeMandic, 2021-03-05:
@ArminJo sigrok PulseView with IR_NEC decoder