Arduino-IRremote/Arduino-IRremote #805: Multiple definition when using multiple files in a project

anton-shchyrov, 2021-02-22:
### IDE
* [X] Arduino IDE
* [X] JetBrains Clion

### Version
* [X] Yes I use the [latest repo version](https://github.com/ukw100/IRMP/archive/master.zip) and verified this!
* [ ] Other - please specify

I am developing a large project with classes that are described in different files. If you write `#include <IRremote.h>` in several .h files, then there are many linking errors about re-declaring the identifier

> sketch\TestClass.cpp.o (symbol from plugin): In function `irparams':
(.text+0x0): multiple definition of `irparams'
sketch\BugCheck.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
sketch\TestClass.cpp.o (symbol from plugin): In function `irparams':
(.text+0x0): multiple definition of `IRrecv::IRrecv()'
sketch\BugCheck.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
sketch\TestClass.cpp.o (symbol from plugin): In function `irparams':
(.text+0x0): multiple definition of `IRrecv::IRrecv()'

**Code Block:**
BugCheck.ino
```cpp
#include <IRremote.h>
#include "TestClass.h"

void setup() {}

void loop() {}
```

TestClass.h
```cpp
#ifndef TEST_CLASS_H
#define TEST_CLASS_H

#include <IRremote.h>

class Test {
  void test();
};

#endif  // TEST_CLASS_H
```

TestClass.cpp
```cpp
#include "TestClass.h"

void Test::test() {
}
```

To fix this error, you must rename all *.cpp.h files to *.cpp and declare global variables ih *.h files as
```cpp
extern IRrecv IrReceiver;
```

**checklist:**
- [X] I have **read** the README.md file thoroughly.
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [X] The title of the issue is helpful and relevant.
- [X] I checked, if at least one of the examples was working.

[BugCheck.zip](https://github.com/Arduino-IRremote/Arduino-IRremote/files/6021325/BugCheck.zip)

ArminJo, 2021-02-24:
Please do me a favor and include IRremoteInt.h instead of IRremote.h in all files except the one implementing the IR function.
This should give no error no more.

anton-shchyrov, 2021-02-26:
I have two classes that implement work with IRRemote. One uses IRSender and the other uses IRReceiver. Each class is in its own cpp module

ArminJo, 2021-03-02:
Have you done as I suggested? It works for me!

closed 2021-03-02 by ArminJo