Arduino-IRremote/Arduino-IRremote #808: decode() output differs between 2.5 and 3.01

Herreropl, 2021-02-23:
I have some problem with updating my project to library above version 2.5. I have spend a lot of time but I’m not able to solve a problem. I have wrote very simple code, that produce different output for different library version.

I think the new version of the library tries to decode some part of IR code as RC5 (2ACC, 2ABB values).

It is Electrolux hob2hood protocol and it send each command three times.

The working code (the old one) and output:

```
#include <IRremote.h>

// IR Receiver PIN
const int PIN_IR_RECEIVER = 2;

IRrecv irrecv(PIN_IR_RECEIVER); // create instance of 'irrecv'
decode_results results;

void setup() {
  Serial.begin(115200); //change BAUD rate as required

  Serial.println("Hob2Hood Starting ...");
  Serial.println(" ... Setup IR receiver");
  irrecv.enableIRIn(); // Start the IR receiver
  Serial.println("Hob2Hood ready ...");
}

void loop() {
  if (irrecv.decode(&results)) {

    Serial.println("Received IR command: ");
    Serial.println(results.value, HEX); // display it on serial monitor in hexadecimal

    irrecv.resume();
  }

}

```
```
Received IR command: 
55303A3
Received IR command: 
55303A3
Received IR command: 
55303A3
Received IR command: 
E208293C
Received IR command: 
E208293C
Received IR command: 
E208293C
Received IR command: 
E3C01BE2
Received IR command: 
E3C01BE2
Received IR command: 
E3C01BE2
Received IR command: 
D051C301
Received IR command: 
D051C301
Received IR command: 
C22FFFD7
Received IR command: 
C22FFFD7
Received IR command: 
C22FFFD7
Received IR command: 
B9121B29
Received IR command: 
B9121B29
Received IR command: 
55303A3
Received IR command: 
55303A3
Received IR command: 
55303A3
```

The not working code (the new one) and output:

```
#include <IRremote.h>

// IR Receiver PIN
const int IR_RECEIVE_PIN = 2;

void setup() {
  pinMode(2,INPUT_PULLUP);
  Serial.begin(115200); //change BAUD rate as required

  Serial.println("Hob2Hood Starting ...");
  Serial.println(" ... Setup IR receiver");
  IrReceiver.begin(IR_RECEIVE_PIN, DISABLE_LED_FEEDBACK);
  Serial.println("Hob2Hood ready ...");
}

void loop() {
  if (IrReceiver.decode()) {

    Serial.println("Received IR command: ");
    Serial.println(IrReceiver.decodedIRData.decodedRawData, HEX); // display it on serial monitor in hexadecimal

    IrReceiver.resume();
  }

}
```

```
Received IR command: 
E208293C
Received IR command: 
E208293C
Received IR command: 
E208293C
Received IR command: 
E3C01BE2
Received IR command: 
2AB5
Received IR command: 
E3C01BE2
Received IR command: 
D051C301
Received IR command: 
D051C301
Received IR command: 
D051C301
Received IR command: 
C22FFFD7
Received IR command: 
C22FFFD7
Received IR command: 
C22FFFD7
Received IR command: 
B9121B29
Received IR command: 
2AAC
Received IR command: 
2AAC
Received IR command: 
55303A3
Received IR command: 
2AAB
Received IR command: 
55303A3
```

### Board
* [ x] Arduino ATmega328* board (UNO, Nano)
### IDE
* [x] Arduino IDE
* [x] Other - Visual Studio Code

### Protocol
* [x] Other - hob2hood
### Version
* [x] Other - 2.5 and 3.01

ArminJo, 2021-03-02:
ib both cases you receive E208293C, so where is the problem?

closed 2021-03-02 by ArminJo

Herreropl, 2021-03-02:
The problem is that in the version 3.01 I'm receiving unpropper values In worst case I'm receiving only uncorrect values. Sometimes it happens three times in a row and I'm not able to receive proper code.

ArminJo, 2021-03-02:
Please try the IRreceiveDump and ReceiverTimingAnalysis example. If possible, with a NEC remote and look at the results.
Maybe you must adjust `MARK_EXCESS_MICROS` to compensate for your receiver module. See [here](https://github.com/Arduino-IRremote/Arduino-IRremote#compile-options--macros-for-this-library).