Arduino-IRremote/Arduino-IRremote #809: IrSender fails and locks up Serial port on MKR WIFI 1010

mdhstn, 2021-02-27:
If you have handling problems or questions, consider to open a discussion https://github.com/ukw100/IRMP/discussions instead of an issue.

### Board
* [X ] Arduino SAMD board (Zero, MKR*)

### IDE
* [ X] Arduino IDE

### Protocol
* [X ] NEC

### Version
* [X ] Yes I use the [latest repo version](https://github.com/ukw100/IRMP/archive/master.zip) and verified this!

Please delete all unchecked lines above :-)

### Pin(s) used for IR-receive, if not default

**Code Block:**
#include <IRremote.h>

uint8_t sRepeats = 0;

void setup() {
    pinMode(LED_BUILTIN, OUTPUT);

    Serial.begin(115200);
    while (!Serial) {delay(10);}

    Serial.println(F("START " __FILE__ " from " __DATE__ "\r\nUsing library version " VERSION_IRREMOTE));
    Serial.print(F("Ready to send IR signals at pin "));
    Serial.println(IR_SEND_PIN);

    IrSender.begin(true); // Enable feedback LED,
}

void loop() {
  // put your main code here, to run repeatedly:

 Serial.println("Send NEC 0xBF40FB04");
  IrSender.sendNECRaw(0xBF40FB04, sRepeats);
  delay(2000);

}

Use [a gist](gist.github.com) if the code exceeds 30 lines

### Current behavior

Using library version 3.0.1
Ready to send IR signals at pin 9
Send NEC 0xBF40FB04

Appears to halt at the IrSender.sendNECRaw, and does not proceed with loop. Also appears to lock the serial port, and can no longer communicate to the IDE. Have to reset the MKR Wifi with the on board button (double press) to recover the device.

This is not an issue on the UNO that i am running the same code on. 

### Expected behavior
Expect to transmit the 0xBF40FB04

### Additional context

**checklist:**
- [X] I have **read** the README.md file thoroughly.
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [X] The title of the issue is helpful and relevant.
- [X] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

mdhstn, 2021-02-28:
FYI the IrSender.sendNECMSB method works as expected... though the hex conversion almost broke my brain.

closed 2021-02-28 by mdhstn

ArminJo, 2021-03-01:
I committed a fix and tested it, now it works!