Arduino-IRremote/Arduino-IRremote #814: Code stopped working after upgrading to 3.0.2

barclay66, 2021-03-03:
If you have handling problems or questions, consider to open a discussion https://github.com/ukw100/IRMP/discussions instead of an issue.

### Board
[x] Arduino ATmega2560 board (Mega)

### IDE
[x] Arduino IDE

### Protocol
[x] NEC

### Version
[x] Yes I use the [latest repo version](https://github.com/ukw100/IRMP/archive/master.zip) and verified this!

### Pin(s) used for IR-receive, if not default
[x] Pin 31

**Code Block:**
`
#define USE_NO_SEND_PWM 0 //No carrier for IR signals due to direct connection. Pin 9 is connected to output of IR receiver/demodulator
#include <IRremote.h>
int IR_RECEIVE_PIN = 31; //IR Receiver input
void setup() {
  IrReceiver.begin(IR_RECEIVE_PIN, DISABLE_LED_FEEDBACK);
  IrSender.begin(false, 0); // No LED feedback
  pinMode(IR_RECEIVE_PIN, INPUT);
  analogWrite(9, 255); // Pull IR signal output to HIGH as it's active LOW
}

void loop() {
  if (IrReceiver.decode()) { //Accepts replacement and original remote
    IrReceiver.printIRResultShort(&Serial);
    if (IrReceiver.decodedIRData.decodedRawData == 4278248961 || IrReceiver.decodedIRData.decodedRawData == 4228059136) {
      IrReceiver.stop();
      delay(50);
      IrSender.sendNEC(0x1000, 0x03, 0);
      IrReceiver.start();
    }
    else if (IrReceiver.decodedIRData.decodedRawData == 4161267201 || IrReceiver.decodedIRData.decodedRawData == 4211347456) {
      IrReceiver.stop();
      delay(50);
      IrSender.sendNEC(0x1000, 0x04, 0);
      IrReceiver.start();
    }
    IrReceiver.resume();
  }
}
`

### Current behavior
Compiles without error. No reaction when pressing remote keys. The receiving part works (checked by adding 'IrReceiver.printIRResultShort(&Serial);'). It's the sending part that stopped working when upgrading the library form 3.0.1 to 3.0.2.
Did maybe the behaviour of 'USE_NO_SEND_PWM' change?

### Expected behavior
The IR code received from one remote should be translated into the corresponding code for the target device. The converted code should be transmitted via direct cable connection to the target device (therefore no carrier modulation).

### Additional context
Hardware connection as follows:
- 3-pin IR receiver connected to GND, +5V and Pin 31
- Pin 9 and GND connected to target device's IR receiver (open collector, active LOW)
[image]

**checklist:**
- [x] I have **read** the README.md file thoroughly.
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] I have browsed the examples for one, that matches my use case.
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [x] The title of the issue is helpful and relevant.
- [x] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

ArminJo, 2021-03-03:
Change 
`IrSender.begin(false, 0); `
to
`IrSender.begin(IR_SEND_PIN, false)`

barclay66, 2021-03-03:
Is this visible somewhere in the changelog?

commit 6f77f822 (2021-03-03): added assertion for USE_NO_SEND_PWM - closes #814
closed 2021-03-03 by ArminJo