Arduino-IRremote/Arduino-IRremote #815: It's not cubecell Lora compatible

hansruedi98, 2021-03-03:
The chip includes an M0 so I'm trying to change by the board to use the Zero board protocol so that it could use this option for compiling. But since it's the first time for me so no success guaranteed. That's why I ask for help.

### Board
ASR6501, ASR6502 cubecell Lora board

### IDE
* [ ✓] Arduino IDE

### Version
* [ ✓] Yes I use the [latest repo version](https://github.com/ukw100/IRMP/archive/master.zip) and verified this!
* [ ] Other - please specify

Please delete all unchecked lines above :-)

### Pin(s) used for IR-receive, if not default

**Code Block:**
```

#include <IRre

Use [a gist](gist.github.com) if the code exceeds 30 lines

### Current behavior
<!-- Paste the code or repository link, if applicable. -->

<!-- Add a the serial output which indicates the error happened. -->

<!-- Add a clear and concise description of the behavior. -->

### Expected behavior
<!-- Add a clear and concise description of what you expected to happen. -->

### Additional context
<!-- (Optional) Add any other context about the problem here. -->

**checklist:**
- [✓] I have **read** the README.md file thoroughly.
- [] I have searched existing issues to see if there is anything I have missed.
- [] I have browsed the examples for one, that matches my use case.
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [] The title of the issue is helpful and relevant.
- [] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

ArminJo, 2021-03-03:
1. This is NO Issue!
2. What do you expect from us?
3. What ist the "Zero board protocol"

closed 2021-03-03 by ArminJo

hansruedi98, 2021-03-05:
Sorry it's the first time I'm using GitHub. It would be cool to be able to use IRlibrary in the cubecell Lora boards from hHelltec. Support for the boards HTCC-AB01 and HTCC-AB02 would be very useful. It's at the moment not supported, can not compile it on Arduino IDE. Thanks