Arduino-IRremote/Arduino-IRremote #819: compilation error no matter what

kevr, 2021-03-10:
### Board

Arduino ATmega328P, Nano V3.0

### IDE

vim. I'm compiling with the same flags that Arduino IDE uses for an ATmega328P Nano.

### Version

On `master` branch, commit 1acdcce7b2d015a20a4ce0e4162a3458da669351.

### Current behavior

Currently, compiling a program with this library causes an error when apparently not **begin**ning the sender. 

```
../Arduino-IRremote/src/IRSend.cpp.h: In member function 'void IRsend::begin(bool, uint8_t)':
../Arduino-IRremote/src/IRSend.cpp.h:70:15: error: call to 'UsageError' declared with attribute error: Version > 3.0.1
   70 |     UsageError("Error: You must use begin(<sendPin>, <EnableLEDFeedback>, <LEDFeedbackPin>) if USE_SOFT_SEND_PWM or USE_NO_SEND_PWM is defined!");
```

Code involved:

```
#include <IRremote.h>

void setup(void) {
    IrReceiver.begin(15, true, 8);
    // I don't actually need this, but I attempted it to get rid of the error.
    // It did not, however, remove the compilation error.
    IrSender.begin(14, false);
}

void loop(void) {
}
```

I'm attempting to compile a program that does not use IrSender. Even if I try to run `IrSender.begin(pin, false)`, I still get the compilation error. I see that this error was introduced in 20bd179bb57a9eceaa220a0568bcdd6082035875 and it seems to not behave as expected. Additionally, it does not make much sense to me that a user would be forced to begin a sender if it is only intended to be used for receiving.

### Expected behavior

It should compile without an error.

ArminJo, 2021-03-10:
> vim. I'm compiling with the same flags that Arduino IDE uses for an ATmega328P Nano.

No, this is not true :-(

closed 2021-03-10 by ArminJo

kevr, 2021-03-10:
Looks like I was missing '-DARDUINO_LIB_DISCOVERY_PHASE' when compiling IRremote sources. Sorry!

tomaskovacik, 2021-03-15:
hello, I have same problem when using minicore:
compile of "SimpleReceiver" will failed with same error when LTO is disabled
enable LTO will fix problem