Arduino-IRremote/Arduino-IRremote #825: Joining IRremote with PCF8574

Samone1, 2021-03-20:
If you have handling problems or questions, consider to open a discussion https://github.com/Arduino-IRremote/Arduino-IRremote/discussions instead of an issue.

### Board
* [x ] Arduino ATmega328* board (UNO, Nano)

### IDE
* [x ] Arduino IDE 1.8.13

### Protocol
* [x ] NEC

### Version  version=3.0.3
* [x] Yes I use the [latest repo version]  (https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) and verified this!

### Pin(s) used for IR-receive, if not default - tryied 2,3,  and all 4-13, A1-A3

[Use code blocks](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code); **it helps us help you when we can read your code!**
Please also refrain from pasting more than 50 lines of code in a post, instead **[create a gist](https://gist.github.com/) if you need to share large snippets**
**Code Block:**

joined code :

<script src="https://gist.github.com/Samone1/bd309dc12678b2b8768dc8e6a1e91592.js"></script>

### Current behavior

 PCF :
repository=https://github.com/xreef/PCF8574_library - example readAll_Interrupt_lowMemory

IR remote : 
TinyIRReceiver.h
MinimalReceiver.ino

Compiller output :
WInterrupts.c.o (symbol from plugin): In function `attachInterrupt':
(.text+0x0): multiple definition of `__vector_2'
sketch\MinimalReceiver-PCF-pokus_z_cisteho.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1

First I would like to thank you for making this library. I appreciate your time spent on working on this library. It is really helpful.

A) I tryied to apply your library + MinimalReceiver as well as dealing with PCF8574_library for communication - also with interrupt  using sketch "readAll_Interrupt_lowMemory" from examples.

Both examples are working perferctly if alone. When I join them together, it is not compilable. There is a problem with a __Vector_2 (or __vector_1) - depends on combination of interupt pins declared
PCF interupt pin                3     2     2     3
IR interrupt pin                  2    2     3      3
-------------------------------------------------
problem :  __vector_         1     1     2      2
(yes, I know they should be different, but I did not know what else I should test #47 

Reading your info that in MinimalReceiver is not used any interrupt (readsme.md - FAQ + Minimal receiver wonder why is there this problem.

B) Going thorough your library more deeply, I finwd an info, that on ATMega328 is possible to use also other pins up to D13 and all A1-A5. Defining them as interrupt pins (it is compillable) it does not work .

I tested it so, because definig pin for interrupt higher then 3 solved above described problem, but IR does not work.

Sorry, I cannot to include the joined code PCF + IrRemote, because I do not know how. For testing purposes I purely copied one of above mentioned example into the another.

### Expected behavior

Both functions (PCF interrupt + IR interrupt) working together in one code on Arduino Nano

### Additional context

**checklist:**
- [x] I have **read** the README.md file thoroughly.
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] I have browsed the examples for one, that matches my use case.
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here - 
- [x] The title of the issue is helpful and relevant.
- [x] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

ArminJo, 2021-03-21:
> Reading your info that in MinimalReceiver is not used any interrupt (readme.md - FAQ + Minimal receiver) I wonder why is there this problem.

Please specify exactly where you read this!!!
And it is by no way an issue if there are resource problems when combined with other arbitrary libraries as long as the resources are well documented!!!

closed 2021-03-21 by ArminJo

Samone1, 2021-03-22:
Sorry, it was misunderstanding from this sentence from Readme.md - FAQ

If you can live with the NEC protocol, you can try the MinimalReceiver
example, it requires no timer.

Ok, can you please give me any advice, how to solve this problem of
combination this two features which following docujentation  should nit
have any common problem ?

Thank you very much for your advice
Aleš Brandejs

ArminJo, 2021-03-22:
I Improved the documentation. 
TL;DR; Mapping of pins to interrupts can be found [here](https://github.com/Arduino-IRremote/Arduino-IRremote/tree/master/src/TinyIRReceiver.cpp.h#L307).

ArminJo, 2021-03-22:
It is always the same! The users are right and I made a mistake 😢.

The problem I had, was not that you opened an issue, but I did not understand what your problem was beside an unspecified resource conflict.
Just get the latest versions of TinyIRReceiver.h and TinyIRReceiver.cpp.h from the repo and it should work.

Sorry for the inconvenience.
Armin

Samone1, 2021-03-22:
I understood it. I already downloaded the new full library before sending
previous email and only inputs D2+D3 were working.
When I changed to other inputs of Arduino Nano (in pure MinimalReceiver),
it does not receive any commands (it is recorded in Serial-log.txt in
previous email). Should it work for other pins or not ?
Maybee it is correct, I am not sure? In my case it does not work.

Aleš Brandejs

commit 65dee31f (2021-03-22): Fix TinyIRReceiver fixes #825