Arduino-IRremote/Arduino-IRremote #826: RC5 Wrong Duty Cycle

MarianoDesivo, 2021-03-22:
I measured Duty cycle with Proteus, simulating an Arduino UNO with IRRemote.h sending RC5

[image]
As you can see, the duty cycle is 50%

Correct duty cycle would be 25% to 33%

I test your library with my DVD Phillips and it kind of work with your 50% duty cycle, but not as good as It works with a 28% duty cycle (I made using atmega328). For example I can NOT turn on my DVD with your library

Where can I change that duty cycle in your library? So I can make the Pull request

ArminJo, 2021-03-22:
Here is a live screenshot of a 328 RC5 PWM Signal with a duty cycle of 32%.
[image]

> I test your library with my DVD Phillips and it kind of work, but not as good as It works with a well done signal

May I ask how you create this well done signal?

closed 2021-03-22 by ArminJo

MarianoDesivo, 2021-03-22:
> Here is a live screenshot of a 328 RC5 PWM Signal

So why I can't see that on Proteus? I can see it in the signal I made

> May I ask how you create this well done signal?

I will be uploading the library, I can tell you when it's done if you want. Basically I made it using Atmega Fast PWM (at 36kHz, 28%), and an interruption with counters