Arduino-IRremote/Arduino-IRremote #829: Sending old NEC 32bit

NW93, 2021-03-23:
Hello.
Can anybody explain how to convert old NEC 32bit code to 8bit and how to send it ?

/*
         * With Send sendNECMSB() you can send your old 32 bit codes.
         * To convert one into the other, you must reverse the byte positions and then reverse all positions of each byte.
         * Example:
         * 0xCB340102 byte reverse -> 0x020134CB bit reverse-> 40802CD3
         */

In this example, how 0xCB340102 becomes 40802CD3 ?

ArminJo, 2021-03-23:
You know what reverse means? and what a bit and a byte is?

closed 2021-03-23 by ArminJo

shirriff, 2021-03-27:
0xCB340102 is binary 11001011001101000000000100000010.
0x40802CD3 is binary 01000000100000000010110011010011.
If you read the first binary sequence backwards (right to left), you get the second sequence.

NW93, 2021-03-27:
Thanks for supporting bro.

NW93, 2021-03-27:
> 0xCB340102 is binary 11001011001101000000000100000010.
> 0x40802CD3 is binary 01000000100000000010110011010011.
> If you read the first binary sequence backwards (right to left), you get the second sequence.

Thanks for supporting bro.