Arduino-IRremote/Arduino-IRremote #830: no way to get ATTiny85 as IR reciever to work with library version 3.x.x

upl1nk-v01d, 2021-03-27:
Hi! I'm trying to get work ATTiny85 with 1 Mhz internal clock speed as IR receiver.
And here is what I get: https://youtu.be/z_xjA1s8BBE 
With old tiny_IRremote library it works and sounds like that: https://youtu.be/LpXEAq5Ji5c
With new IR libraries I can't get anything to work. Please help.

here's my code:
--------------------------------------

#define IR_INPUT_PIN    3 

#include "IRremote.h"

const int led =  0;      // the number of the LED pin
const int spk =  1;      // the number of the LED pin
int c = 0;

// variables will change:

void setup() {
  pinMode(spk, OUTPUT);
  pinMode(led, OUTPUT);
  IrReceiver.begin(IR_INPUT_PIN, ENABLE_LED_FEEDBACK, USE_DEFAULT_FEEDBACK_LED_PIN);
  delay(2000);
  playTone(spk,1000,50);

}

void playTone(int spk, int tone, int duration) {
  for (long i = 0; i < duration * 1000L; i += tone * 2) {
    digitalWrite(spk, HIGH);
    delayMicroseconds(tone);
    digitalWrite(spk, LOW);
    delayMicroseconds(tone);
  }
}

void loop() {

    if (IrReceiver.decode()) {
    playTone(spk,500,50);
    unsigned long _data = IrReceiver.decodedIRData.decodedRawData;
    IrReceiver.resume(); // Receive the next value
  
  if (_data == 3141861120 ) {
    digitalWrite(led, HIGH);
    delay(100);
    digitalWrite(led, LOW);
  }
  delay(10);
  }
}

AnalysIR, 2021-03-27:
I believe the default on the recent release(s) is not to use a timer to generate the Carrier...uses softPWM instead (carrier generated by code loops). Maybe try configuring the Timer instead.

ArminJo, 2021-03-27:
The 3.x version of IRremote is not a sucessor of tiny_IRremote, it is a sucessor of IRremote 2.x.

closed 2021-03-27 by ArminJo