Arduino-IRremote/Arduino-IRremote #833: Native USB uses delay in examples instead of wait-for-serial

Hyratel, 2021-04-02:
If you have handling problems or questions, consider to open a discussion https://github.com/Arduino-IRremote/Arduino-IRremote/discussions instead of an issue.

### Board
* [- ] Arduino ATmega32U4 board (Leonardo)

### IDE
* [- ] Arduino IDE

### Protocol
* [- ] Other - please specify
protocol-independent
### Version
* [- ] Yes I use the [latest repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) and verified this!

Please delete all unchecked lines above :-)

### Pin(s) used for IR-receive, if not default

[Use code blocks](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code); **it helps us help you when we can read your code!**
Please also refrain from pasting more than 50 lines of code in a post, instead **[create a gist](https://gist.github.com/) if you need to share large snippets**
**Code Block:**
```c++

#include <IRremote.h> 
void setup(){
    Serial.begin(115200);
#if defined(__AVR_ATmega32U4__) || defined(SERIAL_USB) || defined(SERIAL_PORT_USBVIRTUAL)  || defined(ARDUINO_attiny3217)
// changes start here
    while (!Serial) {    ;   } // wait for serial port to connect.
// end of changes
#endif

}

```

### Current behavior

```c++
    Serial.begin(115200);
#if defined(__AVR_ATmega32U4__) || defined(SERIAL_USB) || defined(SERIAL_PORT_USBVIRTUAL) || defined(ARDUINO_attiny3217)
    delay(4000); // To be able to connect Serial monitor after reset or power up and before first printout
#endif
```

improvement not error and I'm not familiar with making a pull request

Current behavior uses a delay, sometimes USB COMport devices get scrambled due to the way native USB resets during upload, and need to be re-assigned in arduino IDE, which takes more than the given 4 sec. 
### Expected behavior

expect init-print lines at start of serial terminal session
### Additional context

**checklist:**
- [-] I have **read** the README.md file thoroughly.
- [-] I have searched existing issues to see if there is anything I have missed.
- [-] I have browsed the examples for one, that matches my use case.
- [-] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [-] The title of the issue is helpful and relevant.
- [-] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

ArminJo, 2021-04-08:
Waiting for Serial is ommited by intent (and this is now documented 😀) . 
This enables the software to run even if no Serial Monitor is attached!

Hyratel, 2021-04-08:
ah! makes sense, and I see the value in it

closed 2021-04-08 by Hyratel