Arduino-IRremote/Arduino-IRremote #834: IR Library with depreciating errors

gjaybell, 2021-04-03:
This code works on windows desktop but not on two of my windows laptops on a Arduino UNO. Gets depreciating errors in serial monitor.

//#include <IRremote.h>
int RECV_PIN = 9;
IRrecv irrecv(RECV_PIN);

decode_results results;

void setup(){
  Serial.begin(9600);
  irrecv.enableIRIn(); 
}
  
void loop() {
if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);//  DEC or HEX, 
    irrecv.resume();
  }
}

ArminJo, 2021-04-03:
`irrecv.decode(&results)` is **deprecated**! 
Please do what the deprecation error message suggest!
And have a look at the examples 😀 .

closed 2021-04-03 by ArminJo

gjaybell, 2021-04-03:
I already did this irrecv.decode(). It doesn't work correctly.It gives a
(0) and should give a remote IR code.