Arduino-IRremote/Arduino-IRremote #835: how to send IR commands implemented in NEC2-f16 protocol

AntonioMoraes, 2021-04-06:
Hello !

I have sequences of an equipment that I decoded using a converter tool at http://irdb.tk/convert/ .

I found that the same equipment uses two sets of commands: one using NEC2 protocol and another set using NEC2-f16 protocol.
One of the commands is this  "Protocol NEC2, device 135, subdevice 36, OBC 18". The sender for this, using IRremote Arduino Library with the following  code worked perfectly (36 corresponds to 0x24, and 135 to 0x87. Side by side 0x2487);
     uint16_t sAddress = 0x2487;
     uint8_t sCommand = 0x12;
     uint8_t sRepeats = 1  ;
     IrSender.sendNEC(sAddress, sCommand, sRepeats); 
I was able to make work all commands using using NEC2 protocol...

But for any command of the other set NEC2-f16, I was not able to find a way to send.

For example, for this result below, decoded at http://irdb.tk/convert/ :

     Protocol NEC2-f16, device 135, subdevice 36, OBC 165, misc = OBC2=8

for this original sequence in raw hex:

0000 0077 0000 0022 013B 009C 0013 003A 0013 003A 0013 003A 0013 0013 0013 0013 0013 0013 0013 0013 0013 003A 0013 0013 0013 0013 0013 003A 0013 0013 0013 0013 0013 003A 0013 0013 0013 0013 0013 003A 0013 0013 0013 003A 0013 0013 0013 0013 0013 003A 0013 0013 0013 003A 0013 0013 0013 0013 0013 0013 0013 003A 0013 0013 0013 0013 0013 0013 0013 0013 0013 05BD

I tried  approaching as 16 bit address + 8 bit command + 8 bit inverted command = 2487A55A
        IrSender.sendNECRaw(0x2487A55A, 0, false);
and it did not work...

But it was clearly missing something as the code is not even using  "misc = OBC2=8"  (I do not have a clue what is it).

As I told you, I succeeded using IrSender.sendNEC(sAddress, sCommand, sRepeats) ... not sure if for IrSender.sendNECRaw(tRawData.ULong, aNumberOfRepeats, aIsRepeat) it may be missing some other set in any ".h" file.

How would be IrSender command line for it?
I really appreciate if someone could help me.
Thank you!!!

Antonio Moraes

ArminJo, 2021-04-06:
1. Please do not open an issue here, if you have problems with any obscure IR software!
2. A raw sequence can not contain 0000 values and the last value 05BD is also nonsense.
3. Use the examples of the library e.g. the SmallReceiver and you will see the received values, you can use for send.

closed 2021-04-06 by ArminJo

AntonioMoraes, 2021-04-06:
Hello Armin,
Thank you for your response!
I really didn't want to break any rules ...
Perhaps a more direct and concise question:
How to use the Arduino-IRremote library to send commands in the NEC2-f16 protocol? Is there an example for this use?
How the code would be for this case: "Protocol NEC2-f16, device 135, sub-device 36, OBC 165, misc = OBC2 = 8".

As a note, just to mention: I am trying to send IR commands to an old equipment without remote control (descontinued and very specific), but still I assessed this documentation with the sequences. Using Arduino-IRremote library I could send correct comands to this device, but just those implemented in NEC2 protocol. I have not found any example in the library for NEC2-f16 protocol, that seems to use 16bit addresses and 16 bit commands.

AntonioMoraes, 2021-04-06:
Hello Armin,

Thank you for your response!
I really didn't want to break any rules ...
Perhaps a more direct and concise question:
How to use the Arduino-IRremote library to send commands in the NEC2-f16
protocol? Is there an example for this use?
How the code would be for this case: "Protocol NEC2-f16, device 135,
sub-device 36, OBC 165, misc = OBC2 = 8".

As a note, just to mention: I am trying to send IR commands to an old
equipment without remote control (descontinued and very specific), but
still I assessed this documentation with the sequences. Using
Arduino-IRremote library I could send correct comands to this device, but
just those implemented in NEC2 protocol. I have not found any example in
the library for NEC2-f16 protocol, that seems to use 16bit addresses and 16
bit commands.

Armin , should I open another issue with just the text highlighted in blue
above?

Can you help me?

Thank you,

Antonio Moraes

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ArminJo, 2021-04-06:
There is the Discussions tab, where you can place your questions. Is it not visible for you? Maybe you have no access rights?
Please give me feedback how to guide people to use the discussions tab for such type of help requests!
 
And please look at the examples!!!!!! and also here: https://www.sbprojects.net/knowledge/ir/nec.php
Then you understand, that there is nothing like a "NEC2-f16 with OBC2" protocol and you simply can send all 256 commands of your remote for testing, since the address is constant.

AntonioMoraes, 2021-04-06:
sorry, I should have used the discussion tab.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

AntonioMoraes, 2021-04-07:
Hello Armin,

Yesterday after your definitive comment 'there is nothing like a "NEC2-f16
with OBC2" protocol' I decided to implement a sender of the PRONTO
sequences, leaving behind the ideia of sending by a protocol (I had said
they were raw, I was wrong the sequences were PRONTO sequences)... and it
worked.

I was able to send these sequences (identified as  NEC2-f16 protocol by the
online converter tool at http://irdb.tk/convert/) and the device responded
to that accordingly. (The device is an automation center that was given by
construction companies to control lights , blinds, etc, in the houses they
did at that time. Although the system had infrared sensor for receiving via
IR remote, the remotes have never been given, and the model is
descontinued; something that have never been missed by the buyers. These
houses were of high class standard ) .

But this was the first step for the intent of connecting it in an Alexa
system. So I tried to make the IoT InfraRed Universal Hub (Alexa
compatible) learn this sequence from my Arduino implementation... and
although it recognizes that it captured the sequence, when you execute the
sequence the device simply ignore. So it seems that this "malformed"
sequence is not able to fit well in any protocol that my   IoT InfraRed
Universal Hub could have as templates.

This is the  PRONTO sequence:
 0000 0077 0000 0022 013B 009C 0013 003A 0013 003A 0013 003A 0013 0013 0013
0013 0013 0013 0013 0013 0013 003A 0013 0013 0013 0013 0013 003A 0013 0013
0013 0013 0013 003A 0013 0013 0013 0013 0013 003A 0013 0013 0013 003A 0013
0013 0013 0013 0013 003A 0013 0013 0013 003A 0013 0013 0013 0013 0013 0013
0013 003A 0013 0013 0013 0013 0013 0013 0013 0013 0013 05BD

On the other hand, the sequences simply identified as NEC2 have been
perfectly learned by the  IoT InfraRed Universal Hub.

I am disappointed that  IoT InfraRed Universal Hub somehow seems to
identify a code when learning but when reproduces something different that
is not accepted by the automation center.
You are a specialist in this matter, do you think there is a workaround to
solve it.

Thank you

Antonio Moraes

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ArminJo, 2021-04-07:
Very interesting project!
The Pronto codes are Onkyo protocol codes, which is NEC with 16 bit instead of 8 bit command. (therefore the OBC2=8).
I added the protocol to the library now. Now you should be able to generate the missing signals by sendOnkyo, instead of sendPronto. Maybe it helps for your project.