Arduino-IRremote/Arduino-IRremote #848: SimpleReceiver example overflows infinite on ESP8266

Theagainmen, 2021-04-16:
If you have handling problems or questions, consider to open a discussion https://github.com/Arduino-IRremote/Arduino-IRremote/discussions instead of an issue.

### Board
*** [ X ] Other - please specify: ESP8266**

### IDE
*** [ X ] Arduino IDE**

### Protocol
*** [ X ] Unknown**

### Version
*** [ X ] Yes I use the [latest repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) and verified this!**

### Pin(s) used for IR-receive, if not default
Default pin D5

### Code
The latest `SimpleReceive.ino` example, unedited. 

### Current behavior
The ESP8266 overflows infinitely and keeps rebooting.  This is the serial output (pretty long so a pastebin):
https://pastebin.com/Jpx5VJ2J

The crash happens immediately, the UNKNOWN protocol it's showing is basically noise. 

### Expected behavior

Not crash the ESP8266, and when it stops doing that it should receive my IR signal from an IR remote. 

### Additional context
Can anyone verify if this example code works on their ESP8266?

**checklist:**
- [ X ] I have **read** the README.md file thoroughly.
- [ X ] I have searched existing issues to see if there is anything I have missed.
- [ X ] I have browsed the examples for one, that matches my use case.
- [ X ] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [ X ] The title of the issue is helpful and relevant.
- [ X ] I checked, if at least one of the examples was working (sending works)

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

ArminJo, 2021-04-17:
Thanks for reporting.
But I am very sorry, it runs for me like a charm.
These are my board settings, please compare.
I use [sloeber ](https://eclipse.baeyens.it/)a great tool for arduino pros.
[image]

Theagainmen, 2021-04-18:
Thanks for your response, I didn't really change anything and for some reason it works fine now, other examples already worked like a charm and now this one too, not sure what caused it.

However I am running in some kind of a new problem what I don't really understand. I went through all the example codes and combined SimpleReceiver, SimpleSender and UnitTest to simultanously send & receive IR data on my ESP8266. Here is the output:
```
Sending: 0x102341
Decoded raw data: E42E4006, decoded address: 0, decoded command: 0

Sending: 0x203452
Decoded raw data: ABDF7FF9, decoded address: 0, decoded command: 0

Sending: 0x304563
Decoded raw data: 79285884, decoded address: 0, decoded command: 0

Sending: 0x405674
Decoded raw data: 37A4CDCB, decoded address: 0, decoded command: 0

Sending: 0x506784
Decoded raw data: 4142D283, decoded address: 0, decoded command: 0

Sending: 0x607894
Decoded raw data: A30FA2F6, decoded address: 0, decoded command: 0

```

The send function:
```c
uint16_t sAddress = 0x0102;
uint8_t sCommand = 0x34;
uint8_t sRepeats = 1;

void send_ir_data()
{
  Serial.println();
  Serial.print(F("Sending: 0x"));    
  Serial.print(sAddress, HEX);
  Serial.print(sCommand, HEX);
  Serial.print(sRepeats, HEX);
  Serial.println();

  sAddress += 0x0101;
  sCommand += 0x11;
  sRepeats++;

  // clip repeats at 4
  if (sRepeats > 4) {
    sRepeats = 4;
  }

  // Results for the first loop to: Protocol=NEC Address=0x102 Command=0x34 Raw-Data=0xCB340102 (32 bits)
  IrSender.sendNEC(sAddress, sCommand, sRepeats);
}
```

Receive function:
```c
void receive_ir_data()
{
  if (IrReceiver.decode())
  {
    Serial.print("Decoded raw data: ");
    Serial.print(IrReceiver.decodedIRData.decodedRawData, HEX);
    Serial.print(", decoded address: ");
    Serial.print(IrReceiver.decodedIRData.address, HEX);
    Serial.print(", decoded command: ");
    Serial.println(IrReceiver.decodedIRData.command, HEX);
    IrReceiver.resume();
  }
}
```

I expect the output of the received data to be the same as the sending data, it seems to be very different. Also the `rReceiver.decodedIRData.address` and `rReceiver.decodedIRData.command` data fields are empty, even though there is decodedRawData. What am I doing wrong?

ArminJo, 2021-04-18:
Please test you board and yor setup more carefully before posting your results.
With your code, I get 
```
START Test.cpp from Apr 18 2021
Using library version 3.1.1
Ready to receive IR signals at pin D5
Ready to send IR signals at pin D6
Send signal mark duration is 7 us, pulse correction is 600 ns, total period is 26 us
20 us are subtracted from all marks and added to all spaces for decoding

address=0x102 command=0x34 repeats=1
Sending: 0x102341
Decoded raw data: BA450203, decoded address: 203, decoded command: 45

address=0x203 command=0x45 repeats=2
Sending: 0x203452
Decoded raw data: A9560304, decoded address: 304, decoded command: 56

address=0x304 command=0x56 repeats=3
Sending: 0x304563
Decoded raw data: 98670405, decoded address: 405, decoded command: 67

address=0x405 command=0x67 repeats=4
Sending: 0x405674
Decoded raw data: 87780506, decoded address: 506, decoded command: 78

address=0x506 command=0x78 repeats=4
Sending: 0x506784
Decoded raw data: 76890607, decoded address: 607, decoded command: 89

address0x607 command=0x89 repeats=4
Sending: 0x607894
Decoded raw data: 659A0708, decoded address: 708, decoded command: 9A

address=0x708 command=0x9A repeats=4
Sending: 0x7089A4
Decoded raw data: 54AB0809, decoded address: 809, decoded command: AB
```

closed 2021-04-18 by ArminJo

Theagainmen, 2021-04-18:
Hmh weird, how can I test my setup better and verify it's working? I do see the IR led sending data through my camera, I am getting raw decoded data, it's just not what I'm sending. Also the data I'm receiving stops when I disconnect the IR led, or when I disconnect the receiver LED. What could be wrong?

Theagainmen, 2021-04-18:
Update: I've tried multiple receivers and none of them is able to give me a received address and command. I'm using the default ESP8266 pins for sending and receiving and the hardware wiring is very simple. I have no clue what the issue could be..

ArminJo, 2021-04-18:
You know what LSB first protocols means? NEC is one of them!

Theagainmen, 2021-04-18:
Yes I'm aware of that. I don't see how that results in the library not being able to decode the NEC data I sent, isn't the library made to do that..?

Theagainmen, 2021-04-18:
After looking into the source code for the NEC protocol in the library it's sending and receiving by default LSB unless you use the specific MSB function. So I don't really understand your comment.

ArminJo, 2021-04-18:
What are your results for:
```
#include <Arduino.h>

/*
 * Define macros for input and output pin etc.
 */
#include "PinDefinitionsAndMore.h"

//#define EXCLUDE_EXOTIC_PROTOCOLS // saves around 240 bytes program space if IrSender.write is used
//#define SEND_PWM_BY_TIMER
//#define USE_NO_SEND_PWM

#include <IRremote.h>

#define DELAY_AFTER_SEND 2000
#define DELAY_AFTER_LOOP 5000

void setup() {
#if defined(IR_MEASURE_TIMING) && defined(IR_TIMING_TEST_PIN)
    pinMode(IR_TIMING_TEST_PIN, OUTPUT);
#endif

    Serial.begin(115200);
#if defined(__AVR_ATmega32U4__) || defined(SERIAL_USB) || defined(SERIAL_PORT_USBVIRTUAL)  || defined(ARDUINO_attiny3217)
    delay(4000); // To be able to connect Serial monitor after reset or power up and before first print out. Do not wait for an attached Serial Monitor!
#endif
    // Just to know which program is running on my Arduino
    Serial.println(F("START " __FILE__ " from " __DATE__ "\r\nUsing library version " VERSION_IRREMOTE));

    /*
     * Start the receiver, enable feedback LED and (if not 3. parameter specified) take LED feedback pin from the internal boards definition
     */
    IrReceiver.begin(IR_RECEIVE_PIN);
    IrSender.begin(IR_SEND_PIN, ENABLE_LED_FEEDBACK); // Specify send pin and enable feedback LED at default feedback LED pin

    Serial.print(F("Ready to receive IR signals at pin "));
#if defined(ARDUINO_ARCH_STM32) || defined(ESP8266)
    Serial.println(IR_RECEIVE_PIN_STRING);
#else
    Serial.println(IR_RECEIVE_PIN);
#endif
    Serial.print(F("Ready to send IR signals at pin "));
#if defined(ARDUINO_ARCH_STM32) || defined(ESP8266)
    Serial.println(IR_SEND_PIN_STRING);
#else
    Serial.println(IR_SEND_PIN);
#endif

#if FLASHEND > 0x1FFF && !defined(SEND_PWM_BY_TIMER) && !defined(USE_NO_SEND_PWM) && !defined(ESP32) // for esp32 we use PWM generation by ledcWrite() for each pin
    /*
     * Print internal signal generation info
     */
    IrSender.enableIROut(38);

    Serial.print(F("Send signal mark duration is "));
    Serial.print(IrSender.periodOnTimeMicros);
    Serial.print(F(" us, pulse correction is "));
    Serial.print((uint16_t) PULSE_CORRECTION_NANOS);
    Serial.print(F(" ns, total period is "));
    Serial.print(IrSender.periodTimeMicros);
    Serial.println(F(" us"));

    // infos for receive
    Serial.print(MARK_EXCESS_MICROS);
    Serial.println(F(" us are subtracted from all marks and added to all spaces for decoding"));
#endif
}

uint16_t sAddress = 0x0102;
uint8_t sCommand = 0x34;
uint8_t sRepeats = 1;

void send_ir_data() {
    Serial.print(F("Sending: 0x"));
    Serial.print(sAddress, HEX);
    Serial.print(sCommand, HEX);
    Serial.println(sRepeats, HEX);

    // clip repeats at 4
    if (sRepeats > 4) {
        sRepeats = 4;
    }
    // Results for the first loop to: Protocol=NEC Address=0x102 Command=0x34 Raw-Data=0xCB340102 (32 bits)
    IrSender.sendNEC(sAddress, sCommand, sRepeats);

    sAddress += 0x0101;
    sCommand += 0x11;
    sRepeats++;
}

void receive_ir_data() {
    if (IrReceiver.decode()) {
        Serial.print("Decoded raw data: ");
        Serial.print(IrReceiver.decodedIRData.decodedRawData, HEX);
        Serial.print(", decoded address: ");
        Serial.print(IrReceiver.decodedIRData.address, HEX);
        Serial.print(", decoded command: ");
        Serial.println(IrReceiver.decodedIRData.command, HEX);
        IrReceiver.resume();
    }
}

void loop() {
    /*
     * Print values
     */
    Serial.println();
    Serial.print(F("address=0x"));
    Serial.print(sAddress, HEX);
    Serial.print(F(" command=0x"));
    Serial.print(sCommand, HEX);
    Serial.print(F(" repeats="));
    Serial.println(sRepeats);
    Serial.flush();

    send_ir_data();
    delay((RECORD_GAP_MICROS / 1000) + 5);
    receive_ir_data();
}
```

Theagainmen, 2021-04-18:
Thanks for you response, here is the output:
```
START E:\xx\xx\filename.ino from Apr 18 2021
Using library version 3.1.1
Ready to receive IR signals at pin D5
Ready to send IR signals at pin D6
Send signal mark duration is 8 us, pulse correction is 300 ns, total period is 26 us
20 us are subtracted from all marks and added to all spaces for decoding

address=0x102 command=0x34 repeats=1
Sending: 0x102341
Decoded raw data: CDEB9BDD, decoded address: 0, decoded command: 0

address=0x203 command=0x45 repeats=2
Sending: 0x203452
Decoded raw data: E42E4006, decoded address: 0, decoded command: 0

address=0x304 command=0x56 repeats=3
Sending: 0x304563
Decoded raw data: ABDF7FF9, decoded address: 0, decoded command: 0

address=0x405 command=0x67 repeats=4
Sending: 0x405674
Decoded raw data: 79285884, decoded address: 0, decoded command: 0

address=0x506 command=0x78 repeats=5
Sending: 0x506785
Decoded raw data: 37A4CDCB, decoded address: 0, decoded command: 0

address=0x607 command=0x89 repeats=5
Sending: 0x607895
Decoded raw data: 4142D283, decoded address: 0, decoded command: 0

address=0x708 command=0x9A repeats=5
Sending: 0x7089A5
Decoded raw data: A30FA2F6, decoded address: 0, decoded command: 0
```

ArminJo, 2021-04-19:
Please print also the protocol!!!

Theagainmen, 2021-04-19:
> Please print also the protocol!!!

Added these lines to your code in the receive function:
```c
        Serial.print(", protocol: ");
        Serial.print(IrReceiver.decodedIRData.protocol); 
```

Output:
```
START E:\xx\xx\filename.ino from Apr 19 2021
Using library version 3.1.1
Ready to receive IR signals at pin D5
Ready to send IR signals at pin D6
Send signal mark duration is 8 us, pulse correction is 300 ns, total period is 26 us
20 us are subtracted from all marks and added to all spaces for decoding

address=0x102 command=0x34 repeats=1
Sending: 0x102341
Decoded raw data: CDEB9BDD, protocol: 0, decoded address: 0, decoded command: 0

address=0x203 command=0x45 repeats=2
Sending: 0x203452
Decoded raw data: E42E4006, protocol: 0, decoded address: 0, decoded command: 0

address=0x304 command=0x56 repeats=3
Sending: 0x304563
Decoded raw data: ABDF7FF9, protocol: 0, decoded address: 0, decoded command: 0

address=0x405 command=0x67 repeats=4
Sending: 0x405674
Decoded raw data: 79285884, protocol: 0, decoded address: 0, decoded command: 0

address=0x506 command=0x78 repeats=5
Sending: 0x506785
Decoded raw data: 37A4CDCB, protocol: 0, decoded address: 0, decoded command: 0

address=0x607 command=0x89 repeats=5
Sending: 0x607895
Decoded raw data: 4142D283, protocol: 0, decoded address: 0, decoded command: 0
```

ArminJo, 2021-04-19:
And now you found your problem!

P.s using 
`Serial.print(getProtocolString(IrReceiver.decodedIRData.protocol));` 
makes it more obvious.

Theagainmen, 2021-04-19:
Thanks for that tip `Serial.print(getProtocolString(IrReceiver.decodedIRData.protocol));` :) 

Yes I'm aware the protocol is UNKNOWN, but why is that? I'm sending NEC protocol `IrSender.sendNEC(sAddress, sCommand, sRepeats);` so why am I not receiving this? How do I solve this?

ArminJo, 2021-04-19:
Your receiver module may be wrong or your clock.
Can you remove sending and receive from a plain NEC remote???

Theagainmen, 2021-04-19:
> Your receiver module may be wrong or your clock.
> Can you remove sending and receive from a plain NEC remote???

I don't think I have a NEC remote laying around here. I can give you some background info on what I'm trying to do: I want to try some lasertag identification with IR. All I really need to send is some raw data so that I can identify a player (just send as less data as possible), I don't really care about which protocol but I assume it's necessary with IR. 

Do you know a way that I can achieve this? I have some IR remote laying around but I have no clue what protocol they use, they are cheap AliExpress remotes for a LED strip, whenever I press these I do receive raw data but it's UNKNOWN protocol again.

ArminJo, 2021-04-19:
The cheaper the more likely they use NEC, especially the LED controling ones!
So your receiver module is defect/wrong type. Please read issues about receiver problems in this repository.

Theagainmen, 2021-04-19:
Alright thank you, I'll read that, do some debugging and come back when I solved the issue or have more information about the issue :)

Theagainmen, 2021-04-19:
Well that hurts.. your comment about the clock speed made me think, turns out it doesn't work when the ESP8266 clock runs on 160MHz.. is it a must that the ESP8266 runs on 80MHz? How can I make it work when it runs on 160MHz.

```
address=0x2C2D command=0xF repeats=3
Sending: 0x2C2DF3
Decoded raw data: F00F2C2D, protocol: NEC, decoded address: 2C2D, decoded command: F

address=0x2D2E command=0x20 repeats=3
Sending: 0x2D2E203
Decoded raw data: DF202D2E, protocol: NEC, decoded address: 2D2E, decoded command: 20
```

ArminJo, 2021-04-19:
Please test to change the line 
    `timer1_write(((F_CPU / 1000000) / 16) * MICROS_PER_TICK); // 16 for TIM_DIV16 above`
in private/IRTimer.cpp.h to
    `timer1_write(((80000000 / 1000000) / 16) * MICROS_PER_TICK); // 16 for TIM_DIV16 above`
Maybe then 160 Mhz will run.

Theagainmen, 2021-04-19:
Amazing, thanks! That seems to work already :)

Theagainmen, 2021-04-21:
@ArminJo hello there, I hope I can ask another question here :). For my specific needs I would like to know the 'shortest' protocols. The NEC protocol requires ~67.5ms to send out a message with 0 repeats. Are there 'shorter' protocols that use less time than <67.5ms? The shorter the better, I hope you can give an answer or point me in the right direction!

ArminJo, 2021-04-21:
See `SONY_AVERAGE_DURATION` in ir-Sony.cpp - 21 ms. 
But be aware of `#define RECORD_GAP_MICROS   5000 `

Theagainmen, 2021-04-21:
> See `SONY_AVERAGE_DURATION` in ir-Sony.cpp - 21 ms.
> But be aware of `#define RECORD_GAP_MICROS 5000 `

Thanks a lot, by coincidence I was already testing with Sony :)! Good to see indeed that in all the protocols an average duration is defined. The 5000 micros gap is the minimum gap required between 2 signals I sent?

Theagainmen, 2021-04-25:
Alright few days later! My IR code is getting pretty big so I want to wrap up some stuff in a custom class. I fail to get IRremote working in the custom library where I include it. When you call `#include <IRremote.h>` it automatically creates the instances `IRsend` and `IRrecv` with the name `IrSender` and `IrReceiver`. When I use these in my `.cpp` and `.h` file, like I would in the `.ino`, file I get a ton of errors that I'm duplicating basically all the functions it has. I assume I should use a pointer to the existing sender and receiver object but I can not get it working. If I can wrap this up in a seperate library my code will get much cleaner and easier to read :). Do you have an example on how I can call a basic function like `IrSender.send(xx)` in a class where IRremote is included?

ArminJo, 2021-04-26:
Please look [here](https://github.com/Arduino-IRremote/Arduino-IRremote/tree/master/examples/IRremoteExtensionTest) 😀

Theagainmen, 2021-04-26:
Amazing! I didn't know there was an example! I tried the following in my sketch, but I'm sadly still getting the `multiple definition of `IRsend::BasicallyAnyFunction()` error.

I have in my `.ino` file:
```c
// IR libraries
#include <IRremote.h>
#include <PlayerIR.h>
PlayerIR IRExtension(&IrReceiver);
```

Then in my header file:
```c
#ifndef PlayerIR_h
#define PlayerIR_h

#ifndef IRremote_h
#include <IRremote.h>
#endif

class PlayerIR
{
  public:
    IRrecv *MyIrReceiver;
    IRsend *MyIrSender;
    PlayerIR(IRrecv * aIrReceiver)  {
      MyIrReceiver = aIrReceiver;
    };
    PlayerIR(IRsend * aIrSender)  {
      MyIrSender = aIrSender;
    };
};
```

Then in the .cpp file:
```c
// include this library's description file
#include "PlayerIR.h"

#include "PinDefinitionsAndMore.h"

#ifndef IRremote_h
#include <IRremote.h>
#endif

// stuff

void PlayerIR::send_data(void)
{
  MyIrSender->sendSony(sAddress, sCommand, sRepeats);   
}
```

What am I doing wrong here? 🤔

ArminJo, 2021-04-26:
Do not include `IRremote.h` except in one ino!
You can include `IRremoteInt.h` instead.

Theagainmen, 2021-04-26:
Alright I'm getting so clueless, I've tried so many things now. Here again the `.ino`:
```c
#include "PinDefinitionsAndMore.h"
#include <IRremote.h>
#include <PlayerIR.h>
PlayerIR IRExtension(&IrReceiver);
```

The header .h file:
```c
#ifndef PlayerIR_h
#define PlayerIR_h

#ifndef IRremoteInt_h
#include <IRremoteInt.h>
#endif

class PlayerIR
{
    // user-accessible "public" interface
  public:
    PlayerIR(IRrecv * aIrReceiver, IRsend * aIrSender);
    IRrecv *MyIrReceiver;        
    IRsend *MyIrSender; 
};
```

The `.cpp` file:
```c
// include this library's description file
#include "PlayerIR.h"

// IR libraries
#include "PinDefinitionsAndMore.h"

#ifndef IRremoteInt_h
#include <IRremoteInt.h>
#endif
 
PlayerIR::PlayerIR(IRrecv * aIrReceiver, IRsend * aIrSender)
{
  // initialize this instance's variables
  pinMode(LED_BUILTIN, OUTPUT);
  this->MyIrReceiver = aIrReceiver;
  this->MyIrSender = aIrSender;

  MyIrSender->begin(IR_SEND_PIN, ENABLE_LED_FEEDBACK); 
  MyIrReceiver->begin(IR_RECEIVE_PIN);

  MyIrSender->enableIROut(38);
};
```

Now I'm getting the error that `ENABLE_LED_FEEDBACK` is not defined, when I force that value to `false` I'm still getting the `multiple definition of IRsend::BasicallyAnyFunction()` error. :(

ArminJo, 2021-04-26:
`PlayerIR IRExtension(&IrReceiver);` -> `PlayerIR IRExtension(&IrReceiver, &IrSender);`

change
```
#ifndef IRremoteInt_h
#include <IRremoteInt.h>
#endif
```
to
`#include <IRremoteInt.h>`

Thanks for your testing 👍 , I moved

```
/**
 * For better readability of code
 */
#define DISABLE_LED_FEEDBACK false
#define ENABLE_LED_FEEDBACK true
#define USE_DEFAULT_FEEDBACK_LED_PIN 0
```
to IRremoteInt.h.

> I'm still getting the `multiple definition of IRsend::BasicallyAnyFunction()` error. :(

If you only include IRremote.h (which in turn includes IRSend.cpp.h) once you only compile IRSend.cpp.h / IRsend::BasicallyAnyFunction() once, if you include it twice, ....
Look carefully at the error message, where this functions may come from.

Theagainmen, 2021-04-26:
Thanks! Just a heads up, `PULSE_CORRECTION_NANOS` also seems to be undefined.

I adjusted all the things you said and came much further. The last piece was the difference in including. I included the library like `<xx>` but because my library was in the same folder I had to include it like `"xx"` and voilà it compiled, no more `double functions` error! This extension class will help me alot :).

Theagainmen, 2021-04-27:
Got it. Can you try to compile your library on an ESP32? I get errors:
`In function 'void setFeedbackLED(bool)':
E:\XX\Arduino-IRremote-master\src/private/IRFeedbackLEDDefs.h:85:42: error: 'LED_BUILTIN' was not declared in this scope
 #define FEEDBACK_LED_ON()   digitalWrite(LED_BUILTIN, HIGH)
`
When I add to `IRFeedbackLEDDefs.h`:
```c
#elif defined(ESP32)
#define LED_BUILTIN 2
#define FEEDBACK_LED_ON()   digitalWrite(LED_BUILTIN, LOW)
#define FEEDBACK_LED_OFF()  digitalWrite(LED_BUILTIN, HIGH)
```

It seems to compile, but I'm not sure whether this is right.

ArminJo, 2021-04-27:
> error: 'LED_BUILTIN' was not declared in this scope #define FEEDBACK_LED_ON() digitalWrite(LED_BUILTIN, HIGH)

Thanks for reporting. But it is already fixed in the repo.

Theagainmen, 2021-04-27:
Amazing, thanks! I am testing the ESP32 right now and all the example sketches works just fine. However when I try to implement it in my extension library (which I wrote while using the ESP8266) it doesn't seem to work anymore (compiles & uploads all fine, but I'm not actually receiving or sending anything). 
Are there certain things where I should pay attention to when using the ESP32?