Arduino-IRremote/Arduino-IRremote #855: IrRemote Returns Null Value With Panasonic Remotes using Lib v2.7

studio-suman, 2021-04-24:
If you have handling problems or questions, consider to open a discussion https://github.com/Arduino-IRremote/Arduino-IRremote/discussions instead of an issue.

### Board
* [ x] Arduino ATmega328* board (UNO, Nano)

### IDE
* [x ] Arduino IDE

### Protocol
* [ x] Panasonic

### Version
* [ ] Yes I use the [latest repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) and verified this!
* [ x] Other - please specify v2.70

Please delete all unchecked lines above :-)

### Pin(s) used for IR-receive, if not default

[Use code blocks](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code); **it helps us help you when we can read your code!**
Please also refrain from pasting more than 50 lines of code in a post, instead **[create a gist](https://gist.github.com/) if you need to share large snippets**
**Code Block:**

#define RECEIVER_PIN 11 // define the IR receiver pin
IRrecv receiver(RECEIVER_PIN); // create a receiver object of the IRrecv class
decode_results results; // create a results object of the decode_results class
unsigned long key_value = 0; // variable to store the pressed key value

#include <IRremote.h> 
Serial.begin(115200); // begin serial communication with a baud rate of 9600
  receiver.enableIRIn(); // enable the receiver
  receiver.blink13(true); // enable blinking of the built-in LED when an IR signal is received
}

void loop() {
  if (receiver.decode(&results)) {
    if (results.value == 0XFFFFFFFF) {
      results.value = key_value;
    }
    Serial.println(results.value, HEX);
    switch (results.decode_type) {
      case NEC:
        Serial.println("NEC");
        break;
      case PANASONIC:
        Serial.println("PANASONIC");
        break;
    }
    key_value = results.value;
    receiver.resume();
  }
}
### Current behavior

PANASONIC
0
PANASONIC
0

### Expected behavior

Pin2 = 0x4053888
Pin3 = 0x4053848

### Additional context

**checklist:**
- [x] I have **read** the README.md file thoroughly.
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] I have browsed the examples for one, that matches my use case.
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [x] The title of the issue is helpful and relevant.
- [x] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

ArminJo, 2021-04-30:
This may be a bug in this version, please try a recent one.

closed 2021-04-30 by ArminJo

studio-suman, 2021-05-12:
Dear @ArminJo i have tried with new version that is not compatible with my code posted above.