Arduino-IRremote/Arduino-IRremote #856: Compilation error on ATmega328 with hardware PWM on TIMER1

alexbarcelo, 2021-04-24:
### Board
* [x] Arduino ATmega328* board (UNO, Nano)

### IDE
* [x] PlatformIO IDE

### Protocol
* [x] Other - please specify

Irrelevant

### Version
* [x] Other - please specify

Version 3.2.0

### Pin(s) used for IR-receive, if not default

 * Pin 9

### Current behavior

Compilation error with the following flags:

```
-DIR_USE_AVR_TIMER1
-DIR_SEND_PIN=9
-DSEND_PWM_BY_TIMER
```

Error:

```
#...
Compiling .pio/build/ATmega328P/lib347/IRremote/ir_JVC.cpp.o
In file included from .pio/libdeps/ATmega328P/IRremote/src/IRremote.h:182:0,
                 from src/main.cpp:3:
.pio/libdeps/ATmega328P/IRremote/src/IRSend.cpp.h: In member function 'void IRsend::mark(unsigned int)':
.pio/libdeps/ATmega328P/IRremote/src/IRSend.cpp.h:365:5: error: 'ENABLE_SEND_PWM_BY_TIMER' was not declared in this scope
     ENABLE_SEND_PWM_BY_TIMER; // Enable timer or ledcWrite() generated PWM output
     ^~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/ATmega328P/IRremote/src/IRSend.cpp.h:365:5: note: suggested alternative: 'ENABLE_SEND_PWM_BY_TIMERTCNT1'
     ENABLE_SEND_PWM_BY_TIMER; // Enable timer or ledcWrite() generated PWM output
     ^~~~~~~~~~~~~~~~~~~~~~~~
     ENABLE_SEND_PWM_BY_TIMERTCNT1
.pio/libdeps/ATmega328P/IRremote/src/IRSend.cpp.h: In member function 'void IRsend::IRLedOff()':
.pio/libdeps/ATmega328P/IRremote/src/IRSend.cpp.h:405:5: error: 'DISABLE_SEND_PWM_BY_TIMER' was not declared in this scope
     DISABLE_SEND_PWM_BY_TIMER; // Disable PWM output
     ^~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/ATmega328P/IRremote/src/IRSend.cpp.h:405:5: note: suggested alternative: 'SEND_PWM_BY_TIMER'
     DISABLE_SEND_PWM_BY_TIMER; // Disable PWM output
     ^~~~~~~~~~~~~~~~~~~~~~~~~
     SEND_PWM_BY_TIMER
*** [.pio/build/ATmega328P/src/main.cpp.o] Error 1
============================================================================= [FAILED] Took 1.81 seconds =============================================================================
```

If I do not use a Timer for PWM generation or if I use TIMER2 instead, the application compiles. I have tried to follow the `#define` decision tree but I was lost after a while. Seems that there is some scenario that is not covered, maybe?

alexbarcelo, 2021-04-24:
I should say that the following two lines are somewhat fishy IMHO:

https://github.com/Arduino-IRremote/Arduino-IRremote/blob/4d44d054c2ff0c9663165c993686a6f1a2a0adea/src/private/IRTimer.cpp.h#L256

It's as if there is a whitespace missing on line 256 (between `TIMER` and `TCNT1`). And lines 256 feels as if it should define `DISABLE_SEND_PWM_BY_TIMER`. But I really have no idea what I'm doing here.

ArminJo, 2021-04-24:
Thanks for reporting 🥇 
And yes you are totally right, you found the 2 errors I corrected now. -> see changelog.md

closed 2021-04-24 by ArminJo