Arduino-IRremote/Arduino-IRremote #857: Issue with receiving data from LG air conditioner

tructran91, 2021-04-26:
### Board
* [ ] NodeMCU ESP8266

### IDE
* [ ] Arduino IDE

### Protocol
* [ ] LG

### Version
* [ ] Yes I use the [latest repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) and verified this!

Please delete all unchecked lines above :-)

### Issue
I pressed button to get the data from the remote, and this is what I've got.
```
Protocol=UNKNOWN Hash=0x0 1 bits received
rawData[2]: 
     -3276750
     +3100
Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS
uint8_t rawTicks[1] = {62};  // Protocol=UNKNOWN Hash=0x0 1 bits received

Result as microseconds array - compensated with MARK_EXCESS_MICROS
uint16_t rawData[1] = {3080};  // Protocol=UNKNOWN Hash=0x0 1 bits received

rawData[2]: 
     -3276750
     +3100
```
I try to tested on Samsung TV remote. It gave me full data and here is it's data
```
Protocol=SAMSUNG Address=0x707 Command=0x6 Raw-Data=0xF9060707 32 bits LSB first
rawData[68]: 
     -3276750
     +4500,-4550     + 550,-1700     + 550,-1700     + 550,-1700
     + 550,- 550     + 550,- 600     + 550,- 550     + 550,- 600
     + 550,- 550     + 550,-1700     + 550,-1700     + 500,-1750
     + 550,- 600     + 550,- 550     + 500,- 650     + 550,- 550
     + 550,- 600     + 550,- 550     + 500,-1750     + 550,-1700
     + 550,- 600     + 550,- 550     + 550,- 600     + 550,- 550
     + 500,- 650     + 550,-1700     + 550,- 550     + 550,- 600
     + 550,-1700     + 550,-1700     + 450,-1800     + 550,-1700
     + 450,-1800     + 550
Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS
uint8_t rawTicks[67] = {90,91, 11,34, 11,34, 11,34, 11,11, 11,12, 11,11, 11,12, 11,11, 11,34, 11,34, 10,35, 11,12, 11,11, 10,13, 11,11, 11,12, 11,11, 10,35, 11,34, 11,12, 11,11, 11,12, 11,11, 10,13, 11,34, 11,11, 11,12, 11,34, 11,34, 9,36, 11,34, 9,36, 11};  // Protocol=SAMSUNG Address=0x707 Command=0x6 Raw-Data=0xF9060707 32 bits LSB first

Result as microseconds array - compensated with MARK_EXCESS_MICROS
uint16_t rawData[67] = {4480,4570, 530,1720, 530,1720, 530,1720, 530,570, 530,620, 530,570, 530,620, 530,570, 530,1720, 530,1720, 480,1770, 530,620, 530,570, 480,670, 530,570, 530,620, 530,570, 480,1770, 530,1720, 530,620, 530,570, 530,620, 530,570, 480,670, 530,1720, 530,570, 530,620, 530,1720, 530,1720, 430,1820, 530,1720, 430,1820, 530};  // Protocol=SAMSUNG Address=0x707 Command=0x6 Raw-Data=0xF9060707 32 bits LSB first
```

**Code Block:**
```

#include <Arduino.h>

#include "PinDefinitionsAndMore.h"
#include <IRremote.h>

#if defined(APPLICATION_PIN)
#define DEBUG_BUTTON_PIN    APPLICATION_PIN // if low, print timing for each received data set
#else
#define DEBUG_BUTTON_PIN   6
#endif

// On the Zero and others we switch explicitly to SerialUSB
#if defined(ARDUINO_ARCH_SAMD)
#define Serial SerialUSB
#endif

void setup() {
    Serial.begin(9600);
#if defined(__AVR_ATmega32U4__) || defined(SERIAL_PORT_USBVIRTUAL) || defined(SERIAL_USB) || defined(ARDUINO_attiny3217)
    delay(4000); // To be able to connect Serial monitor after reset or power up and before first print out. Do not wait for an attached Serial Monitor!
#endif
    Serial.println(F("Enabling IRin..."));
    IrReceiver.begin(IR_RECEIVE_PIN, ENABLE_LED_FEEDBACK);
}

void loop() {
if (IrReceiver.decode()) {
    if (IrReceiver.decodedIRData.flags & IRDATA_FLAGS_WAS_OVERFLOW) {
      Serial.println("IR code too long. Edit IRremoteInt.h and increase RAW_BUFFER_LENGTH");
    } else {
      // Print a short summary of received data
      IrReceiver.printIRResultShort( & Serial);
      IrReceiver.printIRResultRawFormatted( & Serial, true);
      
      Serial.println(F("Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS"));
      IrReceiver.compensateAndPrintIRResultAsCArray( & Serial, false); // Output the results as uint8_t source code array of ticks
      Serial.println(F("Result as microseconds array - compensated with MARK_EXCESS_MICROS"));
      IrReceiver.compensateAndPrintIRResultAsCArray( & Serial, true); // Output the results as uint16_t source code array of micros
    }

    if (IrReceiver.decodedIRData.protocol == UNKNOWN || digitalRead(DEBUG_BUTTON_PIN) == LOW) {
      // We have an unknown protocol, print more info
      IrReceiver.printIRResultRawFormatted( & Serial, true);
    }
    Serial.println("================================================="); // blank line between entries
  }

  IrReceiver.resume();
}

```

Thank you.

ArminJo, 2021-04-26:
Looks like your receiver can not receive the LG signals, maybe of wrong carrier frequency.
Or the gap after the first 3100 us mark is longer than https://github.com/Arduino-IRremote/Arduino-IRremote/blob/5927bb0f5a001daf66aa4557ff96305c55967a78/src/IRremote.h#L134 Try to increase it to 15000.
It was exellent, that you included the Samsung cross check data 👍

tructran91, 2021-04-27:
Thanks for your help. And this's result after I change RECORD_GAP_MICROS up to 15000

```
Protocol=UNKNOWN Hash=0xEE419856 30 bits received
rawData[60]: 
     -3276750
     +3100,-9750     + 500,-1550     + 550,- 500     + 500,- 500
     + 500,- 550     + 500,-1550     + 500,- 500     + 500,- 550
     + 500,- 550     + 450,- 550     + 500,- 500     + 500,- 550
     + 450,- 550     + 500,-1550     + 500,- 500     + 500,- 550
     + 500,- 550     + 450,- 550     + 500,- 500     + 500,-1550
     + 500,-1550     + 500,- 500     + 550,-1500     + 500,- 550
     + 500,- 500     + 500,-1550     + 500,-1550     + 500,-1550
     + 500,-1550     + 500
Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS
uint8_t rawTicks[59] = {62,195, 10,31, 11,10, 10,10, 10,11, 10,31, 10,10, 10,11, 10,11, 9,11, 10,10, 10,11, 9,11, 10,31, 10,10, 10,11, 10,11, 9,11, 10,10, 10,31, 10,31, 10,10, 11,30, 10,11, 10,10, 10,31, 10,31, 10,31, 10,31, 10};  // Protocol=UNKNOWN Hash=0xEE419856 30 bits received

Result as microseconds array - compensated with MARK_EXCESS_MICROS
uint16_t rawData[59] = {3080,9770, 480,1570, 530,520, 480,520, 480,570, 480,1570, 480,520, 480,570, 480,570, 430,570, 480,520, 480,570, 430,570, 480,1570, 480,520, 480,570, 480,570, 430,570, 480,520, 480,1570, 480,1570, 480,520, 530,1520, 480,570, 480,520, 480,1570, 480,1570, 480,1570, 480,1570, 480};  // Protocol=UNKNOWN Hash=0xEE419856 30 bits received

```

Can I modify any variable to receive right protocol?

ArminJo, 2021-04-27:
Please see [here](https://github.com/Arduino-IRremote/Arduino-IRremote#protocolunknown).
And / Or change the line in ir_LG.cpp
`#define LG_UNIT                 560 // like NEC`
to 
`#define LG_UNIT                 520 // like NEC`

punj, 2021-06-15:
Hello,
I am facing the same issue. I also tried to change in ir_LG.cpp as below, but still facing the same issue.
```
#define LG_UNIT 560 // like NEC
to
#define LG_UNIT 520 // like NEC
````

Below is the Serial Monitor log:
````
16:38:56.718 -> 
16:38:56.718 -> Protocol=UNKNOWN 1 bits received
16:38:56.718 -> rawData[2]: 
16:38:56.718 ->      -3276750
16:38:56.718 ->      +3200
16:38:56.785 -> 
16:38:56.785 -> Protocol=UNKNOWN 29 bits received
16:38:56.785 -> rawData[58]: 
16:38:56.785 ->      -9500
16:38:56.785 ->      + 550,-1500     + 550,- 450     + 550,- 450     + 550,- 500
16:38:56.785 ->      + 500,-1500     + 500,- 500     + 550,- 450     + 550,- 450
16:38:56.785 ->      + 550,- 450     + 500,- 500     + 550,- 450     + 500,- 550
16:38:56.785 ->      + 500,-1500     + 500,- 500     + 500,- 450     + 600,- 450
16:38:56.785 ->      + 550,-1450     + 500,- 500     + 550,-1500     + 500,-1500
16:38:56.818 ->      + 550,- 450     + 550,-1500     + 550,- 450     + 550,- 450
16:38:56.818 ->      + 500,- 500     + 550,-1450     + 550,-1500     + 550,-1500
16:38:56.818 ->      + 500
````

Code:
I am using ReceiveDemo.ino for test purposes.

I don't know where I am wrong. Help is highly appreciated.

TIA.

punj, 2021-06-15:
> Hello,
> I am facing the same issue. I also tried to change in ir_LG.cpp as below, but still facing the same issue.
>
> ```
> #define LG_UNIT 560 // like NEC
> to
> #define LG_UNIT 520 // like NEC
> ```
>
> Below [...]

Now I am getting Log as below:
18:39:53.236 -> Protocol=PULSE_DISTANCE Address=0x0 Command=0x0 Raw-Data=0x2211011 28 bits LSB first

Below changes applied: 
1) Change in IRremote.h
`#define RECORD_GAP_MICROS   15000 //ORIGINALLY IT WAS 5000` 

2) Added below library to ReceiveDemo.ino
`#include <IRremote.h>`

3) Chang in ir_LG.cpp
`#define LG_UNIT                 520 // like NEC Originally it was 560, `

ArminJo, 2021-06-15:
Can you please post the type of Remote, maybe with a photo 😀 
And if you provide a list of remote buttons together whith its Raw-Data, I can try to build a specialized decoder.

>     2\. Added below library to ReceiveDemo.ino
>         `#include <IRremote.h>`

**Not required! It is already contained!**

> Chang in ir_LG.cpp
> #define LG_UNIT 520 // like NEC Originally it was 560,

**Not required! It is irrelevant for the PULSE_DISTANCE protocol.**

The result for the Raw data below is: `Protocol=PULSE_DISTANCE Address=0x0 Command=0x0 Raw-Data=0xE2D1011 28 bits LSB first`

punj, 2021-06-15:
> Can you please post the type of Remote, maybe with a photo 😀
> And if you provide a list of remote buttons together whith its Raw-Data, I can try to build a specialized decoder.
>
>
> **Not required! It [...]

@ArminJo Thanks for your prompt response.
Please find Remote Photo.

[Remote Photo](https://mega.nz/file/PKwU2DgQ#HMpM-yP-rC1Aha1JGuyMBL8FB-K-GkmKfecwX-PxzUM)
[Raw-Data](https://mega.nz/file/DKwmRTbY#G63QBWJqQzuxz12X8J_bo45rtedEPl4gHPBV0CvdikA)

I appreciate your support.

ArminJo, 2021-06-15:
Hi punj,
can you do me a favor and change 3 lines in ir_LG.cpp to
```
#define LG_UNIT                 500 //  or 520 ??
#define LG_HEADER_MARK          (6 * LG_UNIT) // 3000
#define LG_HEADER_SPACE         (19 * LG_UNIT)  // 9500
```
and set line 88 in IRremoteInt.h to 
`#define DEBUG // Activate this for lots of lovely debug output from the IRremote core and all protocol decoders.`

What is the result with this changes?

punj, 2021-06-16:
Hello @ArminJo Thanks again.

Well, I tried as per your suggestion and it works very well.

But now it is not recognizing the OLD LG AC's remote.
[Photo for your reference](https://mega.nz/file/iOpWwbKQ#OprGM83RAuO7V5MM95rSUKw_mgJj4NXjDtpsiuyKHC4)

Is there a way to work with both the remotes?

TIA

ArminJo, 2021-06-16:
> Is there a way to work with both the remotes?

Of course! Replace the library with the latest from the repo (3.4) and try it.
If it works, it would be fantastic if you could update your Raw-Data file from above with the new (and shorter) values.

BTW.
Can you maybe test the SendLGAirConditionerDemo? At line 35 you can switch beteen LG and LG2 protocols.

punj, 2021-06-16:
> Of course! Replace the library with the latest from the repo (3.4) and try it.
> If it works, it would be fantastic if you could update your Raw-Data file from above with the new (and shorter) [...]

Got it. thanks will update once done.

closed 2021-08-07 by ArminJo