Arduino-IRremote/Arduino-IRremote #890: [Bug]: ir_Pronto method signature issue when compiling SimpleReceiver example

thathert, 2021-09-13:
### The type of board you are using.

Teensy board

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Samsung

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [X] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. I verified that I am using library version 3.3.0. I'm on Arduino 1.8.15 and Teensyduino 1.54
2. Open File -> Examples ->  Examples from Custom Libraries -> IRRemote -> SimpleReceiver
3. Add line below `#include <Arduino.h>`: `#define SUPPRESS_ERROR_MESSAGE_FOR_BEGIN`
4. Click the checkbox to Verify
5. Errors produced in output

### What happens?

The compile fails with this output. (I replaced my username with ... and truncated a couple of the paths below):

```
/private/var/folders/2h/60_837tn5_q7f62fj_w2k5nm0000gn/T/AppTranslocation/53B704F6-3E29-40F2-8238-B54BB55BF7E6/d/Teensyduino.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-g++ -c -Os -g -Wall -ffunction-sections -fdata-sections -MMD -fno-exceptions -fpermissive -felide-constructors -std=gnu++11 -mmcu=atmega32u4 -DTEENSYDUINO=154 -DARDUINO_ARCH_AVR -DARDUINO=10815 -DARDUINO_TEENSY2 -DF_CPU=16000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/var/folders/2h/60_837tn5_q7f62fj_w2k5nm0000gn/T/arduino_build_945287/pch -I/private/var/folders/2h/60_837tn5_q7f62fj_w2k5nm0000gn/T/AppTranslocation/53B704F6-3E29-40F2-8238-B54BB55BF7E6/d/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy -I/Users/.../Documents/Arduino/libraries/IRremote/src /Users/.../Documents/Arduino/libraries/IRremote/src/ir_Template.cpp -o /var/folders/2h/60_837tn5_q7f62fj_w2k5nm0000gn/T/arduino_build_945287/libraries/IRremote/ir_Template.cpp.o

IRremote/src/ir_Pronto.cpp: In function 'bool dumpDigit(String*, unsigned int)':
IRremote/src/ir_Pronto.cpp:275:44: error: cannot convert 'String' to 'bool' in return
     return aString->concat(hexDigit(number));
                                            ^
IRremote/src/ir_Pronto.cpp: In function 'size_t dumpNumber(String*, uint16_t)':
IRremote/src/ir_Pronto.cpp:286:10: error: no match for 'operator+=' (operand types are 'size_t {aka unsigned int}' and 'String')
     size += aString->concat(' ');
     ~~~~~^~~~~~~~~~~~~~~~~~~~~~~

Multiple libraries were found for "IRremote.h"
 Used: /Users/.../Documents/Arduino/libraries/IRremote
 Not used: /private/var/folders/2h/60_837tn5_q7f62fj_w2k5nm0000gn/T/AppTranslocation/53B704F6-3E29-40F2-8238-B54BB55BF7E6/d/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/IRremote
Using library IRremote at version 3.3.0 in folder: /Users/.../Documents/Arduino/libraries/IRremote 
Error compiling for board Teensy 2.0.
```

### The serial output which indicates the error happened.

```shell
N/A
```

### What were you expecting to happen?

Successful compile.

### Additional context.

This worked with an earlier version of the IRRemote examples (I forget which version off the top of my head, I believe it was 2.x or earlier.)

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2021-09-14:
Thanks for reporting 👍 , but Teensy does not support external libraries well, see: https://forum.pjrc.com/threads/65912-Enable-Continuous-Integration-with-arduino-cli-for-3-party-libraries 😞
But nevertheless, please change line 
`    return aString->concat(hexDigit(number));`
to
`    return (bool) aString->concat(hexDigit(number));`
and
`size += aString->concat(' ');`
to
```
    if ((bool) aString->concat(' ')) {
        size++;
    }
```
and report if it worked for you.

thathert, 2021-09-14:
Thanks for the quick response, and wow, that is a troubling thread.

I tried the casts but it seems as though the compiler _really_ thinks concat returns a String instead of a bool like I'm seeing in the docs.

```
IRremote/src/ir_Pronto.cpp:275:51: error: invalid cast from type 'String' to type 'bool'
     return (bool) aString->concat(hexDigit(number));
                                                   ^
```

I am more than willing to try in another IDE or building from the command line if you can recommend a better alternative. Teensyduino is leaving a lot to be desired here.

ArminJo, 2021-09-14:
The problem is, that in the Arduino include WString.h it is defined as:
```
	// returns true on success, false on failure (in which case, the string
	// is left unchanged).  if the argument is null or invalid, the
	// concatenation is considered unsucessful.
	unsigned char concat(const String &str);
```

So try again and please change line 
`    return aString->concat(hexDigit(number));`
to
```
    aString->concat(hexDigit(number));
    return number;
```
and
`size += aString->concat(' ');`
to
```
    aString->concat(' ');
    size++;
```

thathert, 2021-09-14:
That did the trick! Thanks again for the quick assist. I'm not sure if this is something that can/should be caught/addressed in the library, so I'll defer to you whether you'd like to leave this issue open or close it out. I assume other folks using Teensy's might run into the same issue. Cheers 🙌

commit ef868f64 (2021-09-14): Closes #890
closed 2021-09-14 by ArminJo