Arduino-IRremote/Arduino-IRremote #891: Send NEC doesn't work for me

jefix96, 2021-09-16:
### The type of board you are using.

Arduino ATmega2560 board (Mega)

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

11

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [X] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [X] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [ ] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

I used the ReceiveDemo source to read the remote from the LD player.
For example, here is the output for opening the tray:

Protocol = NEC Address = 0xA8 Command = 0x19 Raw-Data = 0xE61957A8 32 bits LSB first

Raw result in internal ticks (50 us) - with leading gap
rawData [68]:
     -65535
     + 169, - 83 + 12, - 9 + 12, - 9 + 12, - 9
     + 12, - 30 + 12, - 9 + 12, - 30 + 12, - 10
     + 11, - 30 + 12, - 31 + 11, - 30 + 12, - 31
     + 11, - 10 + 11, - 30 + 12, - 10 + 11, - 31
     + 12, - 9 + 12, - 30 + 12, - 9 + 12, - 9
     + 12, - 30 + 12, - 30 + 12, - 9 + 12, - 10
     + 11, - 9 + 12, - 9 + 12, - 31 + 11, - 30
     + 12, - 10 + 12, - 9 + 12, - 30 + 12, - 30
     + 12, - 30 + 12
Raw result in microseconds - with leading gap
rawData [68]:
     -3276750
     + 8450, -4150 + 600, - 450 + 600, - 450 + 600, - 450
     + 600, -1500 + 600, - 450 + 600, -1500 + 600, - 500
     + 550, -1500 + 600, -1550 + 550, -1500 + 600, -1550
     + 550, - 500 + 550, -1500 + 600, - 500 + 550, -1550
     + 600, - 450 + 600, -1500 + 600, - 450 + 600, - 450
     + 600, -1500 + 600, -1500 + 600, - 450 + 600, - 500
     + 550, - 450 + 600, - 450 + 600, -1550 + 550, -1500
     + 600, - 500 + 600, - 450 + 600, -1500 + 600, -1500
     + 600, -1500 + 600

Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS
uint8_t rawTicks [67] = {169.83, 12.9, 12.9, 12.9, 12.30, 12.9, 12.30, 12.10, 11.30, 12.31, 11.30 , 12.31, 11.10, 11.30, 12.10, 11.31, 12.9, 12.30, 12.9, 12.9, 12.30, 12.30, 12.9, 12 , 10, 11.9, 12.9, 12.31, 11.30, 12.10, 12.9, 12.30, 12.30, 12.30, 12}; // Protocol = NEC Address = 0xA8 Command = 0x19 Raw-Data = 0xE61957A8 32 bits LSB first

Result as microseconds array - compensated with MARK_EXCESS_MICROS
uint16_t rawData [67] = {8430.4170, 580.470, 580.470, 580.470, 580.1520, 580.470, 580.1520, 580.520, 530.1520, 580.1570, 530.1520, 580.1570, 530.520, 530.1520 , 580,520, 530,1570, 580,470, 580,1520, 580,470, 580,470, 580,1520, 580,1520, 580,470, 580,520, 530,470, 580,470, 580,1570, 530,1520, 580,520, 580,470, 580,1520, 580 , 1520,580,1520,580}; // Protocol = NEC Address = 0xA8 Command = 0x19 Raw-Data = 0xE61957A8 32 bits LSB first

uint16_t address = 0xA8;
uint16_t command = 0x19;
uint32_t data = 0xE61957A8;

Pronto Hex as string
char ProntoData [] = "0000 006D 0022 0000 0146 009F 0018 0011 0018 0011 0018 0011 0018 0039 0018 0011 0018 0039 0018 0012 0016 0039 0018 003B 0016 0039 0018 003B 0016 0012 0016 0039 0018 0012 0016 003B 0018 0011 0018 0039 0018 00 0018 0039 0018 0039 0018 0011 0018 0012 0016 0011 0018 0011 0018 003B 0016 0039 0018 0012 0018 0011 0018 0039 0018 0039 0018 0039 0018 06C3 "

I made a transmitter and used code like this:

#include <IRremote.h>
IRsend irsend;

uint16_t ir [67] = {8430, 4170, 580, 470, 580, 470, 580, 470, 580, 1520, 580, 470, 580, 1520, 580, 520, 530, 1520, 580, 1570, 530, 1520 , 580, 1570, 530, 520, 530, 1520, 580, 520, 530, 1570, 580, 470, 580, 1520, 580, 470, 580, 470, 580, 1520, 580, 1520, 580, 470, 580 , 520, 530, 470, 580, 470, 580, 1570, 530, 1520, 580, 520, 580, 470, 580, 1520, 580, 1520, 580, 1520, 580};

void setup () {
}

void loop () {

  for (int i = 0; i <1; i ++) {
    irsend.sendRaw (ir, sizeof (ir) / sizeof (ir [0]), 32);
    delay (40);
  }
  delay (3000);
  for (int i = 0; i <2; i ++) {
    irsend.sendRaw (ir, sizeof (ir) / sizeof (ir [0]), 32);
    delay (40);
  }
  delay (3000);
  for (int i = 0; i <3; i ++) {
    irsend.sendRaw (ir, sizeof (ir) / sizeof (ir [0]), 32);
    delay (40);
  }
  delay (3000);

  irsend.sendNECRaw (0xE61957A8,0);
  delay (3000);
  irsend.sendNECRaw (0xE61957A8,1);
  delay (3000);
  irsend.sendNECRaw (0xE61957A8,2);
  delay (3000);
  irsend.sendNECRaw (0xE61957A8,3);
  delay (3000);

}

### What happens?

None of the commands above worked.

I read my transmitter and I got the correct output.

Can you tell me what the problem might be? Am I using the wrong delays between retries?

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2021-09-16:
# Sending IR codes
Please do not use the old send*Raw functions for sending like e.g. `IrSender.sendNECRaw (0xE61957A8,2)`,
even if this functions are used in a lot of **(old)** tutorials. They are only kept for backward compatibility and unsupported and error prone.<br/>
**Much better** is to use the **new structured functions** with address and command parameters like e.g. `IrSender.sendNEC(0xA8, 0x19, 2)`.
Especially if you are able to receive these remote codes. You will find that **the address is a constant** and the commands sometimes are sensible grouped.

jefix96, 2021-09-17:
I tried submitting a different code and it worked perfectly! SendRaw, SendNEC, SendNECRaw work equally well.

Why is the tray open code not working? I noticed that two buttons from the remote control are read with the same code (open/close & last memory). How is this possible? I tried another receiver but got the same result.

Perhaps yesterday I was transmitting the "last memory" signal instead of the expected "open/close".

The remote control works fine. This is not a bug in the remote control.

ArminJo, 2021-09-17:
So please record the IR signal with printIRResultShort() using 115200 baud.
Take a look at the repeats.
You can post the results if you want.

jefix96, 2021-09-17:
Here is a complete dump. The first is "open/close", the second is "last memory".

Protocol=NEC Address=0xA8 Command=0x19 Raw-Data=0xE61957A8 32 bits LSB first

Raw result in internal ticks (50 us) - with leading gap
rawData[68]: 
     -65535
     + 169,-  82     +  12,-   9     +  12,-   9     +  12,-   9
     +  12,-  31     +  11,-   9     +  12,-  31     +  11,-   9
     +  12,-  30     +  13,-  29     +  13,-  30     +  11,-  30
     +  12,-  10     +  11,-  30     +  13,-   9     +  11,-  30
     +  13,-   8     +  12,-  30     +  13,-   9     +  11,-  10
     +  12,-  30     +  12,-  30     +  12,-   9     +  12,-   9
     +  12,-   9     +  12,-   9     +  12,-  30     +  12,-  30
     +  12,-   9     +  12,-   9     +  12,-  30     +  12,-  30
     +  12,-  30     +  12
Raw result in microseconds - with leading gap
rawData[68]: 
     -3276750
     +8450,-4100     + 600,- 450     + 600,- 450     + 600,- 450
     + 600,-1550     + 550,- 450     + 600,-1550     + 550,- 450
     + 600,-1500     + 650,-1450     + 650,-1500     + 550,-1500
     + 600,- 500     + 550,-1500     + 650,- 450     + 550,-1500
     + 650,- 400     + 600,-1500     + 650,- 450     + 550,- 500
     + 600,-1500     + 600,-1500     + 600,- 450     + 600,- 450
     + 600,- 450     + 600,- 450     + 600,-1500     + 600,-1500
     + 600,- 450     + 600,- 450     + 600,-1500     + 600,-1500
     + 600,-1500     + 600

Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS
uint8_t rawTicks[67] = {169,82, 12,9, 12,9, 12,9, 12,31, 11,9, 12,31, 11,9, 12,30, 13,29, 13,30, 11,30, 12,10, 11,30, 13,9, 11,30, 13,8, 12,30, 13,9, 11,10, 12,30, 12,30, 12,9, 12,9, 12,9, 12,9, 12,30, 12,30, 12,9, 12,9, 12,30, 12,30, 12,30, 12};  // Protocol=NEC Address=0xA8 Command=0x19 Raw-Data=0xE61957A8 32 bits LSB first

Result as microseconds array - compensated with MARK_EXCESS_MICROS
uint16_t rawData[67] = {8430,4120, 580,470, 580,470, 580,470, 580,1570, 530,470, 580,1570, 530,470, 580,1520, 630,1470, 630,1520, 530,1520, 580,520, 530,1520, 630,470, 530,1520, 630,420, 580,1520, 630,470, 530,520, 580,1520, 580,1520, 580,470, 580,470, 580,470, 580,470, 580,1520, 580,1520, 580,470, 580,470, 580,1520, 580,1520, 580,1520, 580};  // Protocol=NEC Address=0xA8 Command=0x19 Raw-Data=0xE61957A8 32 bits LSB first

uint16_t address = 0xA8;
uint16_t command = 0x19;
uint32_t data = 0xE61957A8;

Pronto Hex as string
char ProntoData[] = "0000 006D 0022 0000 0146 009D 0018 0011 0018 0011 0018 0011 0018 003B 0016 0011 0018 003B 0016 0011 0018 0039 001A 0037 001A 0039 0016 0039 0018 0012 0016 0039 001A 0011 0016 0039 001A 000F 0018 0039 001A 0011 0016 0012 0018 0039 0018 0039 0018 0011 0018 0011 0018 0011 0018 0011 0018 0039 0018 0039 0018 0011 0018 0011 0018 0039 0018 0039 0018 0039 0018 06C3 "

Protocol=NEC Address=0xA8 Command=0x19 Raw-Data=0xE61957A8 32 bits LSB first

Raw result in internal ticks (50 us) - with leading gap
rawData[68]: 
     -52830
     + 169,-  82     +  12,-   9     +  12,-   9     +  12,-   9
     +  12,-  30     +  12,-   9     +  12,-  30     +  12,-   9
     +  12,-  30     +  12,-  30     +  12,-  30     +  12,-  30
     +  12,-   9     +  12,-  30     +  12,-   9     +  12,-  30
     +  12,-   9     +  12,-  30     +  12,-   9     +  12,-   9
     +  12,-  30     +  12,-  30     +  12,-   9     +  12,-  10
     +  11,-   9     +  12,-   9     +  12,-  31     +  12,-  29
     +  12,-  10     +  11,-   9     +  12,-  31     +  11,-  30
     +  13,-  29     +  12
Raw result in microseconds - with leading gap
rawData[68]: 
     -2641500
     +8450,-4100     + 600,- 450     + 600,- 450     + 600,- 450
     + 600,-1500     + 600,- 450     + 600,-1500     + 600,- 450
     + 600,-1500     + 600,-1500     + 600,-1500     + 600,-1500
     + 600,- 450     + 600,-1500     + 600,- 450     + 600,-1500
     + 600,- 450     + 600,-1500     + 600,- 450     + 600,- 450
     + 600,-1500     + 600,-1500     + 600,- 450     + 600,- 500
     + 550,- 450     + 600,- 450     + 600,-1550     + 600,-1450
     + 600,- 500     + 550,- 450     + 600,-1550     + 550,-1500
     + 650,-1450     + 600

Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS
uint8_t rawTicks[67] = {169,82, 12,9, 12,9, 12,9, 12,30, 12,9, 12,30, 12,9, 12,30, 12,30, 12,30, 12,30, 12,9, 12,30, 12,9, 12,30, 12,9, 12,30, 12,9, 12,9, 12,30, 12,30, 12,9, 12,10, 11,9, 12,9, 12,31, 12,29, 12,10, 11,9, 12,31, 11,30, 13,29, 12};  // Protocol=NEC Address=0xA8 Command=0x19 Raw-Data=0xE61957A8 32 bits LSB first

Result as microseconds array - compensated with MARK_EXCESS_MICROS
uint16_t rawData[67] = {8430,4120, 580,470, 580,470, 580,470, 580,1520, 580,470, 580,1520, 580,470, 580,1520, 580,1520, 580,1520, 580,1520, 580,470, 580,1520, 580,470, 580,1520, 580,470, 580,1520, 580,470, 580,470, 580,1520, 580,1520, 580,470, 580,520, 530,470, 580,470, 580,1570, 580,1470, 580,520, 530,470, 580,1570, 530,1520, 630,1470, 580};  // Protocol=NEC Address=0xA8 Command=0x19 Raw-Data=0xE61957A8 32 bits LSB first

uint16_t address = 0xA8;
uint16_t command = 0x19;
uint32_t data = 0xE61957A8;

Pronto Hex as string
char ProntoData[] = "0000 006D 0022 0000 0146 009D 0018 0011 0018 0011 0018 0011 0018 0039 0018 0011 0018 0039 0018 0011 0018 0039 0018 0039 0018 0039 0018 0039 0018 0011 0018 0039 0018 0011 0018 0039 0018 0011 0018 0039 0018 0011 0018 0011 0018 0039 0018 0039 0018 0011 0018 0012 0016 0011 0018 0011 0018 003B 0018 0037 0018 0012 0016 0011 0018 003B 0016 0039 001A 0037 0018 06C3 "

ArminJo, 2021-10-02:
The repeats are still missing

closed 2021-10-02 by ArminJo