Arduino-IRremote/Arduino-IRremote #892: [Bug]: ESP32 wifi wont reconnect unless IrReceiver.stop()

htmltiger, 2021-09-17:
Wifi works fine if done hard reset or software reboot but if it disconnects for any reason it will not reconnect.
I could not find why this was happening and changed boards but did not help.
the same code works fine for sender and commented out IrReceiver.decode() in loop and reconnects automatically.

 
The workaround is
```
#define LED_BUILTIN  42 // Dummy
#define IR_SEND_PIN 4
#define IR_RECEIVE_PIN          34  // D15
#define tone(a,b,c) void()      // no tone() available on ESP32
#define noTone(a) void()
#define TONE_PIN                42 // Dummy for examples using it
#define APPLICATION_PIN         16 // RX2 pin
#include <IRremote.h>

void loop(){
static byte iron=1;
if ( WiFi.isConnected() ){
   if( !iron ){iron=1; IrReceiver.resume();}
}else{
  if( iron ){iron=0; IrReceiver.stop();}
}

if (iron && IrReceiver.decode()){
 ...
}

}
```

Could it be some timer in the library that conflicts with wifi connection handshake?

ArminJo, 2021-09-17:
This code toggles constantly iron and IrReceiver if WiFi is connected!
So the error may be in the other parts of your code and not in the library .

htmltiger, 2021-09-17:
> This code toggles constantly iron and IrReceiver if WiFi is connected!
> So the error may be in the other parts of your code and not in the library .

Its static variable and only toggles once for the change.

ArminJo, 2021-09-17:
Yes you are right! Sorry.
Is the connect done before enabling IRremote? 
How is the reconnect performed?

htmltiger, 2021-09-17:
Please try this code and disconnect it from router or reboot wifi
```
#include <WiFi.h>
#define LED_BUILTIN  42 // Dummy
#define IR_SEND_PIN 4
#define IR_RECEIVE_PIN          34  // D15
#define tone(a,b,c) void()      // no tone() available on ESP32
#define noTone(a) void()
#define TONE_PIN                42 // Dummy for examples using it
#define APPLICATION_PIN         16 // RX2 pin
#include <IRremote.h> //Version Sep 16, 2021

volatile byte gotip=0;

void WiFiEvent(WiFiEvent_t event) {
	Serial.print("WIFI:");Serial.println(event);
	switch(event) {
		case SYSTEM_EVENT_STA_GOT_IP:{
			gotip=1;
			int sig=WiFi.RSSI(); sig=(sig<=-100)?0:(sig>=-50)?100:(2*(sig+100));
			Serial.print(sig);Serial.print("% Wifi IP: ");
			Serial.println(WiFi.localIP());
			break;
		}
		case SYSTEM_EVENT_STA_DISCONNECTED:{
			gotip=0; break;
		}
		default:{break;}
	}
}

void wifiMon(void *pvParameters){
	const char* ssid = "YOURSSID";
	const char* password = "YOURPASS";
	WiFi.onEvent(WiFiEvent);
	for(;;){
		if(WiFi.status() == WL_CONNECTED && gotip){Serial.print("y");vTaskDelay(5000); continue;}
		Serial.println("Reset");
		WiFi.disconnect();
		delay(100); 
		WiFi.mode(WIFI_STA);
		WiFi.begin(ssid, password);
		byte i = 0;
		while (!gotip && (i++ < 40)){
			Serial.print(".");
			vTaskDelay(200);
		}
		vTaskDelay(5000);
	}
}

void setup(){
	Serial.begin(115200);
	xTaskCreatePinnedToCore(wifiMon, "wifiMon", 12192, NULL, 1, NULL, 0);
	delay(2000);
	IrReceiver.begin(IR_RECEIVE_PIN, DISABLE_LED_FEEDBACK);
	IrSender.begin(IR_SEND_PIN, DISABLE_LED_FEEDBACK);
}

void loop(){

	static byte iron=0;

	//if (!gotip && iron){iron=0; Serial.println("\nIrReceiver.stop()"); IrReceiver.stop();}
	
	if (gotip && !iron){iron=1; Serial.println("\nIrReceiver.resume()"); IrReceiver.resume();}
	
	if (iron && IrReceiver.decode()){char buf[100];
		snprintf( buf, 100, "IR Mf %d Cmd %#x %#x", IrReceiver.decodedIRData.protocol, IrReceiver.decodedIRData.command, IrReceiver.decodedIRData.decodedRawData );
		Serial.println(buf);
		IrReceiver.resume();
	}

}
```

softhack007, 2022-02-09:
Just an idea - at least on my ESP32 board, the pins related to ADC2 sometimes behave strangely while Wifi is active - using them as digital i/o sometimes killed the wifi connection. Also they seem to have some potential (current) towards ground, even when used as digital input.

* GPIO0, GPIO2, GPIO4, GPIO12 to GPIO15, GOIO25 to GPIO27

Could you try to use other pins in your code? If this improves the situation, then maybe its a bug in the esp32 libraries.

See also https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#adc-limitations .

htmltiger, 2022-05-12:
This has been fixed with esp32 library 2.0.0, 
2.0.3 has possible issues with timerAlarmEnable or other timer related bugs.

closed 2022-05-12 by htmltiger