Arduino-IRremote/Arduino-IRremote #898: [minor] defines vs compilation

roger-tannous, 2021-09-29:
Say we want to use the library with a Samsung device:: 

#define DECODE_SAMSUNG

#define EXCLUDE_EXOTIC_PROTOCOLS 1
#define EXCLUDE_UNIVERSAL_PROTOCOLS 1

#include <IRremote.h>

The compiler still compiles ir_Sony.cpp, ir_NEC.cpp, etc.

Issue marked as minor as it may not affect program storage space.

roger-tannous, 2021-09-29:
These defines don't need a value, so it could be like:: 

#define DECODE_SAMSUNG

#define EXCLUDE_EXOTIC_PROTOCOLS
#define EXCLUDE_UNIVERSAL_PROTOCOLS

#include <IRremote.h>

ArminJo, 2021-09-29:
Can you please suggest a solution for this and test it 😀 .

roger-tannous, 2021-09-29:
Not really.
I thought there could be a way, like the magic lines we often learn from in makefiles :)

closed 2021-09-29 by roger-tannous

ArminJo, 2021-09-29:
Please do not open an issue for a problem which cannot be solved in Arduino!

roger-tannous, 2021-09-29:
Kindly understand that if I knew how everything can be fixed in Arduino, I would have created my own Arduino library, which makes you, at least "here", superior (superior is expected to be kinder :P).

Anyway, I respect your passion and hard work, you're really making a difference.

ArminJo, 2021-09-29:
You are welcome. 
Maybe you did not know that there is a (quite new) Discussions tab in Github.