Arduino-IRremote/Arduino-IRremote #900: Irremote for arduino RP2040

mugiwara28, 2021-10-02:
Hello, I am trying to use the librairy with arduino RP2040, I got the message below :

```
In file included from C:\Users\leled\Documents\Arduino\libraries\Arduino-IRremote-master\src/IRremote.h:188:0,
                 from C:\Users\leled\Documents\Arduino\libraries\Arduino-IRremote-master\examples\SimpleReceiver\SimpleReceiver.ino:44:
C:\Users\leled\Documents\Arduino\libraries\Arduino-IRremote-master\src/private/IRTimer.hpp:1237:2: error: #error Internal code configuration error, no timer functions implemented for this CPU / board
 #error Internal code configuration error, no timer functions implemented for this CPU / board
  ^~~~~
exit status 1
Erreur de compilation pour la carte Arduino Nano RP2040 Connect
```
arduino rp2040 work with mbed_nano.
 Arduino-IRremote/library.properties  mention : architectures=avr,megaavr,samd,esp8266,esp32,stm32,STM32F1,mbed,mbed_nano

Please help

ArminJo, 2021-10-02:
Thanks for reporting 👍 
Please try the latest version from this repository and give me feedback if it works for you.

mugiwara28, 2021-10-03:
Hello,
with the latest version this is what i get 👍 

```
In file included from C:\Users\leled\Documents\Arduino\libraries\Arduino-IRremote-master\src/IRremote.h:198:0,
                 from C:\Users\leled\Documents\Arduino\libraries\Arduino-IRremote-master\examples\SimpleReceiver\SimpleReceiver.ino:44:
C:\Users\leled\Documents\Arduino\libraries\Arduino-IRremote-master\src/IRSend.hpp: In member function 'void IRsend::begin(bool, uint8_t)':
C:\Users\leled\Documents\Arduino\libraries\Arduino-IRremote-master\src/IRSend.hpp:87:15: error: call to 'UsageError' declared with attribute error: Version > 3.0.1
     UsageError(
     ~~~~~~~~~~^
             "Error: You must use begin(<sendPin>, <EnableLEDFeedback>, <LEDFeedbackPin>) if SEND_PWM_BY_TIMER is not defined or USE_NO_SEND_PWM is defined or enable lto or activate the line #define SUPPRESS_ERROR_MESSAGE_FOR_BEGIN in IRremote.h.");
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Nano RP2040 Connect.
```
regards

mugiwara28, 2021-10-03:
by activating the line #define SUPPRESS_ERROR_MESSAGE_FOR_BEGIN in IRremote.h." i get no error message, I will test the code with the sensor and see if all it ok

ArminJo, 2021-10-14:
Any new results?