Arduino-IRremote/Arduino-IRremote #906: How the IrSender.sendSony() work

LeeFuuChang, 2021-10-30:
I've connected IRrecv and IRsend
after using the following code to send "0xa90"
```
IrSender.sendSony(0xa90, 12);
```
the recv side prints out "95" (Hex)
how does this work?

1. How 0xa90 represents/convert to 95(Hex)?

2. Is there any documentation I can read for better understanding?

3. How do I convert the code I got from the recv side, to the code which I can send using the IRsender, and will get the same code when the recv side received and print out?

ArminJo, 2021-10-30:
Please show the output and use 
```
/*
 * Repeat commands should be sent in a 45 ms raster.
 * There is NO delay after the last sent command / repeat!
 * @param numberOfBits if == 20 send 13 address bits otherwise only 5 address bits
 */
void IRsend::sendSony(uint16_t aAddress, uint8_t aCommand, uint_fast8_t aNumberOfRepeats, uint8_t numberOfBits) {
```

LeeFuuChang, 2021-10-30:
IRsend Code
```
#include <IRremote.h>
IRsend irsend;
int Send_PIN = 3;
int Code = 0xa90;

void setup()
{
 Serial.begin(9600);
 IrSender.begin(Send_PIN, ENABLE_LED_FEEDBACK);
}

void loop() 
{
  Serial.print("Send： ");
  Serial.println(Code);
  IrSender.sendSony(Code, 12); // Sony TV power code
  delay(500);
}
```

IrReceiver Code
```
#include <IRremote.h>

//Fan
//ON / OFF    = 7C483B
//Weak wind   = A045FB
//Strong wind = 8047FB

int IRreceivePIN = 7;

void setup(){
  Serial.begin(9600);

  pinMode(LED_BUILTIN, OUTPUT);

  Serial.println("Starting IR-Receiver . . .");

  IrReceiver.begin(IRreceivePIN, DISABLE_LED_FEEDBACK);

  Serial.println("IR-Receiver activated!");

  digitalWrite(LED_BUILTIN, LOW);
}

void loop(){
  while(IrReceiver.decode()){
    Serial.println(IrReceiver.decodedIRData.decodedRawData, HEX);
    IrReceiver.begin(IRreceivePIN, DISABLE_LED_FEEDBACK);
    switch(IrReceiver.decodedIRData.protocol){
      case NEC: 
        Serial.println("NEC"); 
        break;
      case SONY: 
        Serial.println("SONY"); 
        break;
      case RC5: 
        Serial.println("RC5"); 
        break;
      case RC6: 
        Serial.println("RC6"); 
        break;
      case DISH: 
        Serial.println("DISH"); 
        break;
      case SHARP: 
        Serial.println("SHARP"); 
        break;
      case JVC: 
        Serial.println("JVC"); 
        break;
      case SAMSUNG: 
        Serial.println("SAMSUNG"); 
        break;
      case LG: 
        Serial.println("LG"); 
        break;
      case WHYNTER: 
        Serial.println("WHYNTER"); 
        break;
      case PANASONIC: 
        Serial.println("PANASONIC"); 
        break;
      case DENON: 
        Serial.println("DENON"); 
        break;
    default:
      case UNKNOWN: 
        Serial.println("UNKNOWN"); 
        break;
    }
  }
}
```

IRsend Output
```
Send: 2704
Send: 2704
Send: 2704
Send: 2704
Send: 2704
.
.
.
.
.
```

IrReceiver Output
```
95
Sony
95
Sony
95
Sony
95
Sony
.
.
.
.
.
.
```

> Please show the output and use
>
> ```
> /*
>  * Repeat commands should be sent in a 45 ms raster.
>  * There is NO delay after the last sent command / repeat!
>  * @param numberOfBits if == 20 send 13 address [...]

Can't understand what you mean 'use'
but here's the output and my code

ArminJo, 2021-10-30:
You must see a deprecated warning!!! turn on File > Preferences Show verbose output during compilation!

> Can't understand what you mean 'use'

Look into the SendDemo example how to send Sony.

LeeFuuChang, 2021-10-30:
Output
```
Protocol=UNKNOWN 13 bits received
 rawData[26]: 
      -496350
      +2350,- 650     +1150,- 650     + 550,- 650     +1150,- 650
      + 550,- 650     +1150,- 650     + 550,- 650     + 600,- 600
      +1200,- 600     + 550,- 650     + 550,- 650     + 600,- 600
      + 600
```

do i use sendraw to send this?

ArminJo, 2021-10-30:
> (with sony enabled!)

You read this?

closed 2021-10-30 by LeeFuuChang