Arduino-IRremote/Arduino-IRremote #911: [Bug]: ATtiny 2313 support

vintagepc, 2021-11-17:
Is it possible to use this library with the ATTiny2313?

Using ATTinyCore and selecting this board results in a compile error:

```
Arduino/libraries/IRremote/src/private/IRTimer.hpp:792:2: error: #error Internal code configuration error, no timer functions implemented for this AVR CPU / board
 #error Internal code configuration error, no timer functions implemented for this AVR CPU / board
```

Is that something that just requires modifying the definitions, or is it because the hardware has insufficient capabilities? (Timer speed/resolution)

Thanks!

closed 2021-11-17 by ArminJo

vintagepc, 2021-11-17:
FYI you can convert issues to discussions rather than closing them.

ArminJo, 2021-11-17:
And you can open a discussion rather ttan open an issue.

vintagepc, 2021-11-17:
Sheesh, no need to be snarky about it  because of an unintentional oversight on my part. A few clicks from you and a suitable apology from me for having a derp moment and we could have been having a fruitful discussion on the actual question at hand.

If you prefer to use discussions for such things that's fine, but until github makes them more prominently featured you might want to consider adding a link-template to your github templates config.yml to explicitly tell users you'd prefer to have anything that is not obviously a bug opened there. Peoples' github usage patterns vary wildly based on how the other projects they interact with operate, and I happened to wander down the common "Don’t see your issue here? Open a blank issue." pathway after using search and not finding a suitable matching result.

https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository#configuring-the-template-chooser

ArminJo, 2021-11-17:
Thanks a lot for this hint 🥇 .
Since you seems to have more knowledge than a script kiddy, so what is the background of using a ATTiny2313 and do you have any examples of using a periodic timer on this chip and what board for Arduino support? Then maybe is is not much effort to extend the lib.

vintagepc, 2021-11-17:
Glad we could turn it into something positive :) 

The 2313 is what I happen to have on hand. I'm essentially looking to receive some panasonic or NEC IR codes and drive some IO pins in response to add remote control capability to a device. From looking at the datasheet it does have a 16 bit timer on timer1 (and the code compiles if I hack in a few defines to bypass the various MCU macros to test)

Let me poke at it some more and tinker in the library code a bit and I'll see what I can put together to determine whether support is possible/practical. I'll see if I can get it working at least crudely as a proof-of-concept that more formal support might be possible.

ArminJo, 2021-11-17:
Try the minimalReceiver code it uses only pin change interrupts and receives NEC, but not yet panasonic 😞 .

vintagepc, 2021-11-17:
Thanks - the protocol isn't fixed; the driving remote is universal so I could just pick a different unused codeset to drive my device that is in NEC protocol.

vintagepc, 2021-11-19:
Early findings: The minimal example does compile and appears to be running, though it's very close to the memory limitations of the chip (2K flash, 128b RAM)

The following defines are required for the compile:
```

#define __digitalPinToPortReg(__pin)    (((__pin) <= 8) ? &PORTD : &PORTB)
#define __digitalPinToDDRReg(__pin)     (((__pin) <= 8) ? &DDRD : &DDRB)
#define __digitalPinToPINReg(__pin)     (((__pin) <= 8) ? &PIND : &PINB)
#define __digitalPinToBit(__pin)        (((__pin) <= 1) ? (__pin) : (((__pin) >= 4 && (__pin) <= 8) ? (__pin) - 2 : (__pin) - 9))

```

I'm hoping to get some time this weekend to actually wire up the chip DIO and test it. (only have bare chips and an ISP so I'm just dead-bugging it instead of in a devboard)

vintagepc, 2021-12-10:
given the limited space on this micro is barely sufficient for the example code after stripping out several things like printf and digitalwrite I needed in the interest of time and getting the project done I've opted to move to a different micro instead.