Arduino-IRremote/Arduino-IRremote #921: [Bug]: Nano Every: strange selection of SEND_PIN with SEND_PWM_BY_TIMER

softhack007, 2021-12-01:
### The type of board you are using.

Arduino megaAVR board (NanoEvery)

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 5

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

Created a simple Sketch with SEND_PWM_BY_TIMER defined (see below)
Compiled on Nano Every (megaavr core), no register emulation
Uploaded to check which IR_SEND_PIN is selected by the library

```C++
// just some random pick
#define IR_RECEIVE_PIN 5
#define IR_SEND_PIN    11

#define SEND_PWM_BY_TIMER
  // Nano Every: IR_USE_AVR_TIMER_B => expected send pin = pin 6
  //  auto select = PIN18 = A4
#include <IRremote.h>

void setup() {
  Serial.begin(115200);
  // initialize IR objects
  IrReceiver.begin(IR_RECEIVE_PIN, DISABLE_LED_FEEDBACK); // Start the receiver, enable feedback LED, take LED feedback pin from the internal boards definition
  IrSender.begin(IR_SEND_PIN, DISABLE_LED_FEEDBACK); // Specify send pin and enable feedback LED at default feedback LED pin

#ifdef USE_NO_SEND_PWM
  Serial.println("USE_NO_SEND_PWM defined");
#endif
#ifdef SEND_PWM_BY_TIMER
  Serial.println("SEND_PWM_BY_TIMER defined");
#endif
#ifdef IR_USE_AVR_TIMER_B
  Serial.println("IR_USE_AVR_TIMER_B defined");
#endif
#ifdef IR_USE_AVR_TIMER_A
  Serial.println("IR_USE_AVR_TIMER_A defined");
#endif

  // write PINs to serial
  Serial.print("IR_RECEIVE_PIN = ");
  Serial.println(IR_RECEIVE_PIN);
  Serial.print("IR_SEND_PIN = ");
  Serial.println(IR_SEND_PIN);
}
void loop() {
  delay(1000);
}

```

### What happens?

The Sketch produces the output below.
As you can see, **PIN 18** is selected, which is PIN_A4 - A4 is usually used for I2C, so not a good selection on Every. if I understand IRTimer.hpp correctly, PIN 6 would be expected.

```
SEND_PWM_BY_TIMER defined
IR_USE_AVR_TIMER_B defined
IR_RECEIVE_PIN = 5
IR_SEND_PIN = 18
```

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2021-12-01:
Thanks for doing this wonderful Quality Assurance! 🥇 
I corrected it in the sources, pin 6 is right, so please replace the A4 in your version of IRTimer.hpp by 6.

closed 2021-12-01 by ArminJo

softhack007, 2021-12-01:
Thanks for the fast reaction 👍 will try out the fix in the next few days, and let you know if it works for me.

softhack007, 2021-12-02:
work for me now, thanks again 👍

commit 3afa4ef3 (2021-12-01): Documentation and fix of ATmega4209 send pin bug. Closes #921
commit 823c53f3 (2021-12-01): Documentation and fix of ATmega4209 send pin bug. Closes #921