Arduino-IRremote/Arduino-IRremote #923: [Bug]: Program with IrReceive not compilable for an Atmega88

DoImant, 2021-12-01:
### The type of board you are using.

Other - please specify below

### What IDE are you using?

PlatformIO IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [ ] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

Use Platform IO 

platformio.ini
[platformio]
description = Infrarot Empfang mit dem TSOP 31238
default_envs = ATmega88P

[env:ATmega88P]
platform = atmelavr
board = ATmega88P
framework = arduino

platform_packages = 
lib_deps = z3t0/IRremote@^3.1.1
build_type = release
build_flags = -Os

Copy the ReceiveDemo Example into a main.cpp file and try to compile it. The errors below are displayed.

### What happens?

In file included from .pio\libdeps\ATmega88P\IRremote\src/IRremote.hpp:216:0,
                 from src\main.cpp:76:
.pio\libdeps\ATmega88P\IRremote\src/private/IRTimer.hpp:778:2: error: #error Internal code configuration error, no timer functions implemented for this AVR CPU / board
 #error Internal code configuration error, no timer functions implemented for this AVR CPU / board
  ^~~~~
In file included from .pio\libdeps\ATmega88P\IRremote\src/IRremote.hpp:221:0,
                 from src\main.cpp:76:
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp: In member function 'void IRrecv::enableIRIn()':
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp:157:5: error: 'timerConfigForReceive' was not declared in this scope
     timerConfigForReceive();
     ^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp:158:5: error: 'TIMER_ENABLE_RECEIVE_INTR' was not declared in this scope
     TIMER_ENABLE_RECEIVE_INTR;  // Timer interrupt enable
     ^~~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp:159:5: error: 'TIMER_RESET_INTR_PENDING' was not declared in this scope
     TIMER_RESET_INTR_PENDING;   // NOP for most platforms
     ^~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp: In member function 'void IRrecv::disableIRIn()':
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp:173:5: error: 'TIMER_DISABLE_RECEIVE_INTR' was not declared in this scope
     TIMER_DISABLE_RECEIVE_INTR;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\users\kai\.platformio\packages\framework-arduino-avr-minicore\cores\MiniCore/Arduino.h:30:0,
                 from src\main.cpp:34:
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp: At global scope:
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp:1342:1: error: expected unqualified-id before 'void'
 ISR () // for functions definitions which are called by separate (board specific) ISR
 ^
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp:1342:1: error: expected ')' before 'void'
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp:1342:1: error: expected unqualified-id before 'void'
 ISR () // for functions definitions which are called by separate (board specific) ISR
 ^
.pio\libdeps\ATmega88P\IRremote\src/IRReceive.hpp:1342:1: error: expected ')' before 'void'
*** [.pio\build\ATmega88P\src\main.cpp.o] Error 1

### What were you expecting to happen?

The program should be compilable in the main program without the addition #define IR_USE_AVR_TIMER2.

### Additional context.

If you insert #define IR_USE_AVR_TIMER2 it works.

Is it possible that in the file IRTimer.hpp at the line #if defined (__ AVR_ATmega328P__) || defined (__ AVR_ATmega328PB__) || defined (__ AVR_ATmega168__)  a || (__AVR_ATmega88P__) is missing?

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2021-12-01:
If you explicitely confirm, that receiving with #IR_USE_AVR_TIMER2 inserted, works sucessfully (can you post the serial output?) I can add it as proposed. I have no means to check it for myself.

DoImant, 2021-12-03:
This ist a Screenshot of the ReceiveDemo output with an Atmega88P with 8Mhz, internal clock.

board_build.mcu = atmega88P
board_hardware.oscillator = internal
board_build.f_cpu = 8000000L

[image]

ArminJo, 2021-12-03:
Thanks, I checked it in.

closed 2021-12-03 by ArminJo