Arduino-IRremote/Arduino-IRremote #944: [Bug]: Samsung decoder often fails, the library misinterprets received signals as unknown 34 bit codes

roromo, 2022-01-12:
### The type of board you are using.

Arduino ATmega32U4 board (Leonardo, etc.)

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Samsung

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 3

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [X] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [X] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [ ] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

-  try to do this in a room with low ambient light (no ambient light is fine too) so that the IR receiver AGC is high, to reproduce this quicker than in a room with sunlight
- load one of the IR receive demos on the board, like the ReceiveDump demo
- press buttons on a Samsung TV remote control
- randomly, but very often, the sent IR code will not be recognized, the demo will print that it received a 34 bit code of an unknown protocol

This was reproduced with both IR receivers I have: a Vishay TSOP4838 and a Vishay TSOP38238.
It seems to happen for any MARK_EXCESS_MICROS value. I tried 0, 20, 50, values above 100 and as far as up to 350.

### What happens?

Often Samsung IR signals are not recognized, the library misinterprets them as a 34 bit unknown protocol codes. This makes it impossible to have uninterrupted volume up or down actions for instance.

### The serial output which indicates the error happened.

```shell
2:14:51.563 -> Protocol=UNKNOWN Hash=0xFE2E4130 34 bits received
12:14:51.563 -> 
12:14:51.563 -> Raw result in internal ticks (50 us) - with leading gap
12:14:51.563 -> rawData[68]: 
12:14:51.563 ->      -899
12:14:51.563 ->      +  89,-  90     +  11,-  34     +  11,-  34     +  11,-  34
12:14:51.563 ->      +  11,-  12     +  10,-  15     +   8,-  13     +  11,-  12
12:14:51.563 ->      +  11,-  12     +  11,-  34     +  11,-  34     +  11,-  34
12:14:51.563 ->      +  11,-  12     +  11,-  12     +  10,-  13     +  11,-  12
12:14:51.563 ->      +  11,-  12     +  11,-  13     +   9,-  13     +  10,-  12
12:14:51.563 ->      +  12,-  33     +  13,-  33     +  10,-  12     +  12,-  34
12:14:51.563 ->      +  11,-  11     +  11,-  36     +   9,-  34     +  12,-  33
12:14:51.563 ->      +  12,-  13     +   9,-  13     +  11,-  34     +  11,-  12
12:14:51.563 ->      +  11,-  33     +  12
12:14:51.563 -> Raw result in microseconds - with leading gap
12:14:51.611 -> rawData[68]: 
12:14:51.611 ->      -44950
12:14:51.611 ->      +4450,-4500     + 550,-1700     + 550,-1700     + 550,-1700
12:14:51.611 ->      + 550,- 600     + 500,- 750     + 400,- 650     + 550,- 600
12:14:51.611 ->      + 550,- 600     + 550,-1700     + 550,-1700     + 550,-1700
12:14:51.611 ->      + 550,- 600     + 550,- 600     + 500,- 650     + 550,- 600
12:14:51.611 ->      + 550,- 600     + 550,- 650     + 450,- 650     + 500,- 600
12:14:51.611 ->      + 600,-1650     + 650,-1650     + 500,- 600     + 600,-1700
12:14:51.611 ->      + 550,- 550     + 550,-1800     + 450,-1700     + 600,-1650
12:14:51.611 ->      + 600,- 650     + 450,- 650     + 550,-1700     + 550,- 600
12:14:51.611 ->      + 550,-1650     + 600
12:14:51.611 -> 
12:14:51.611 -> Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS=20
12:14:51.611 -> uint8_t rawTicks[67] = {89,90, 11,34, 11,34, 11,34, 11,12, 10,15, 8,13, 11,12, 11,12, 11,34, 11,34, 11,34, 11,12, 11,12, 10,13, 11,12, 11,12, 11,13, 9,13, 10,12, 12,33, 13,33, 10,12, 12,34, 11,11, 11,36, 9,34, 12,33, 12,13, 9,13, 11,34, 11,12, 11,33, 12};  // Protocol=UNKNOWN Hash=0xFE2E4130 34 bits received
12:14:51.611 -> 
12:14:51.611 -> Result as microseconds array - compensated with MARK_EXCESS_MICROS=20
12:14:51.611 -> uint16_t rawData[67] = {4430,4520, 530,1720, 530,1720, 530,1720, 530,620, 480,770, 380,670, 530,620, 530,620, 530,1720, 530,1720, 530,1720, 530,620, 530,620, 480,670, 530,620, 530,620, 530,670, 430,670, 480,620, 580,1670, 630,1670, 480,620, 580,1720, 530,570, 530,1820, 430,1720, 580,1670, 580,670, 430,670, 530,1720, 530,620, 530,1670, 580};  // Protocol=UNKNOWN Hash=0xFE2E4130 34 bits received
12:14:51.659 -> 
12:14:51.659 -> Pronto Hex as string
12:14:51.659 -> char prontoData[] = "0000 006D 0022 0000 00AC 00AC 0016 0041 0016 0041 0016 0041 0016 0016 0014 001C 0010 0018 0016 0016 0016 0016 0016 0041 0016 0041 0016 0041 0016 0016 0016 0016 0014 0018 0016 0016 0016 0016 0016 0018 0012 0018 0014 0016 0018 003F 001A 003F 0014 0016 0018 0041 0016 0014 0016 0044 0012 0041 0018 003F 0018 0018 0012 0018 0016 0041 0016 0016 0016 003F 0018 06C3 ";
```

### Additional context.

I did find the probable cause, and I have a fix for this issue which works 100% of the time.

The library does two things which seem to cause this behavior:
- is using in its computation of the duration of mark/spaces/etc 38 KHz as the carrier frequency for Samsung IR signals, instead of the actual frequency which is 37.9 KHz
- is using truncated values for one duration to compute another duration (a truncated value of SAMSUNG_UNIT to compute the SAMSUNG_HEADER_MARK duration, for instance), and then does a truncation of the computed value as well, which means we have not only error propagation but additional errors introduced in the computation of derived durations

What I did to fix this was to use the actual 37.9 KHz frequency for the computation of durations, and to avoid using truncated values in the computation of derived durations (SAMSUNG_HEADER_MARK duration is computed as "(8 * 21) / 37 900" for instance, instead of "8 * int(21 / 38 000)".

In other words, I used these defines in the ir_Samsung.hpp file instead of the existing values:
// actual frequency is 37.9 KHz
// all times in microseconds
#define SAMSUNG_UNIT                554 // 21 periods of 37.9 kHz (554,0897)
#define SAMSUNG_HEADER_MARK         4433 // (8 * SAMSUNG_UNIT) (4432.717)
#define SAMSUNG_HEADER_SPACE        4433 // (8 * SAMSUNG_UNIT) (4432.717)
#define SAMSUNG_BIT_MARK            SAMSUNG_UNIT
#define SAMSUNG_ONE_SPACE           1662 // (3 * SAMSUNG_UNIT) (1662.2691)
#define SAMSUNG_ZERO_SPACE          SAMSUNG_UNIT

With these changes, I have recorded no misinterpreted Samsung IR codes anymore using the default value of MARK_EXCESS_MICROS=20 for days, on any of the 52 buttons on my remote. Presses on buttons with repeat functionality such as "Volume Up/Down" works fine all the time, the action is no longer randomly interrupted by a misinterpreted received signal.

It also works fine most of the time with MARK_EXCESS_MICROS=0, but very, very rarely misinterpretation of the received signal still occurs. It occurs much less often than with the unmodified durations, but still occurs.

I did not change the durations used for sending, as the sending seems to be using a wrong carrier frequency anyway. Any signal that should be sent with a 38 KHz frequency is sent using a 38.46 KHz signal. In case of the Samsung IR protocol which should use a 37.9 KHz carrier, the delta between what should be used and what is used is even greater.

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2022-01-12:
Your main problem is, that you have a negative MARK_EXCESS_MICROS, which you should see as a result of the ReceiverTimingAnalysis.
i.e. all our marks are always shorter than the short spaces.
See [here](https://github.com/Arduino-IRremote/Arduino-IRremote#protocolunknown)
`If you see timings like + 500,- 550 + 450,- 550 + 500,- 500 + 500,-1550, then marks are generally shorter than spaces and therefore MARK_EXCESS_MICROS (specified in your ino file) should be **negative** to compensate for this at decoding.`
And you had luck to be able to compensate it with your new timings!

closed 2022-01-12 by ArminJo

ArminJo, 2022-01-12:
I added a [chapter in the readme](https://github.com/Arduino-IRremote/Arduino-IRremote#how-we-decode-signals), mabe you can do a quality assurance?

roromo, 2022-01-12:
ReceiverTimingAnalysis never displayed negative values for MARK_EXCESS_MICROS in my tests.

In fact, ReceiverTimingAnalysis almost never displays two consecutive identical values. I got values in between 0 and 350 with it, that's why I said I used those values in my tests. And this with both my receivers. (the documentation claims that 50 is the optimal value for TSOP4838, even).

Here it is the same issue reproduces just now with MARK_EXCESS_MICROS= -20:

```
4:29:47.658 -> Protocol=UNKNOWN Hash=0x6A05DBD2 34 bits received
14:29:47.658 -> 
14:29:47.658 -> Raw result in internal ticks (50 us) - with leading gap
14:29:47.658 -> rawData[68]: 
14:29:47.658 ->      -898
14:29:47.658 ->      +  90,-  90     +  11,-  34     +  12,-  32     +  13,-  33
14:29:47.658 ->      +  11,-  11     +  12,-  12     +  11,-  12     +  12,-  11
14:29:47.658 ->      +  11,-  11     +  12,-  34     +  11,-  34     +  12,-  32
14:29:47.658 ->      +  12,-  12     +  11,-  12     +  10,-  13     +  11,-  11
14:29:47.658 ->      +  12,-  11     +  11,-  12     +  12,-  11     +  11,-  34
14:29:47.658 ->      +  11,-  12     +  11,-  12     +  11,-  12     +  11,-  33
14:29:47.658 ->      +  17,-   6     +  12,-  33     +  12,-  34     +  12,-  11
14:29:47.658 ->      +  12,-  33     +  12,-  33     +  12,-  33     +  12,-  11
14:29:47.658 ->      +  12,-  34     +  11
14:29:47.658 -> Raw result in microseconds - with leading gap
14:29:47.658 -> rawData[68]: 
14:29:47.658 ->      -44900
14:29:47.658 ->      +4500,-4500     + 550,-1700     + 600,-1600     + 650,-1650
14:29:47.658 ->      + 550,- 550     + 600,- 600     + 550,- 600     + 600,- 550
14:29:47.658 ->      + 550,- 550     + 600,-1700     + 550,-1700     + 600,-1600
14:29:47.706 ->      + 600,- 600     + 550,- 600     + 500,- 650     + 550,- 550
14:29:47.706 ->      + 600,- 550     + 550,- 600     + 600,- 550     + 550,-1700
14:29:47.706 ->      + 550,- 600     + 550,- 600     + 550,- 600     + 550,-1650
14:29:47.706 ->      + 850,- 300     + 600,-1650     + 600,-1700     + 600,- 550
14:29:47.706 ->      + 600,-1650     + 600,-1650     + 600,-1650     + 600,- 550
14:29:47.706 ->      + 600,-1700     + 550
14:29:47.706 -> 
14:29:47.706 -> Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS=-20
14:29:47.706 -> uint8_t rawTicks[67] = {90,90, 11,34, 12,32, 13,33, 11,11, 12,12, 11,12, 12,11, 11,11, 12,34, 11,34, 12,32, 12,12, 11,12, 10,13, 11,11, 12,11, 11,12, 12,11, 11,34, 11,12, 11,12, 11,12, 11,33, 17,6, 12,33, 12,34, 12,11, 12,33, 12,33, 12,33, 12,11, 12,34, 11};  // Protocol=UNKNOWN Hash=0x6A05DBD2 34 bits received
14:29:47.706 -> 
14:29:47.706 -> Result as microseconds array - compensated with MARK_EXCESS_MICROS=-20
14:29:47.706 -> uint16_t rawData[67] = {4520,4480, 570,1680, 620,1580, 670,1630, 570,530, 620,580, 570,580, 620,530, 570,530, 620,1680, 570,1680, 620,1580, 620,580, 570,580, 520,630, 570,530, 620,530, 570,580, 620,530, 570,1680, 570,580, 570,580, 570,580, 570,1630, 870,280, 620,1630, 620,1680, 620,530, 620,1630, 620,1630, 620,1630, 620,530, 620,1680, 570};  // Protocol=UNKNOWN Hash=0x6A05DBD2 34 bits received
14:29:47.706 -> 
14:29:47.706 -> Pronto Hex as string
14:29:47.706 -> char prontoData[] = "0000 006D 0022 0000 00AC 00AE 0014 0042 0016 003E 0018 0040 0014 0016 0016 0018 0014 0018 0016 0016 0014 0016 0016 0042 0014 0042 0016 003E 0016 0018 0014 0018 0012 001A 0014 0016 0016 0016 0014 0018 0016 0016 0014 0042 0014 0018 0014 0018 0014 0018 0014 0040 0020 000C 0016 0040 0016 0042 0016 0016 0016 0040 0016 0040 0016 0040 0016 0016 0016 0042 0014 06C3 ";
```

Here with MARK_EXCESS_MICROS= -50:
```
4:33:01.893 -> Protocol=UNKNOWN Hash=0xE11B891D 34 bits received
14:33:01.893 -> 
14:33:01.893 -> Raw result in internal ticks (50 us) - with leading gap
14:33:01.893 -> rawData[68]: 
14:33:01.893 ->      -4252
14:33:01.893 ->      +  90,-  89     +  12,-  34     +  11,-  34     +  11,-  34
14:33:01.893 ->      +  11,-  12     +  11,-  12     +  11,-  11     +  12,-  11
14:33:01.893 ->      +  12,-  11     +  12,-  33     +  12,-  34     +  12,-  33
14:33:01.893 ->      +  11,-  12     +  11,-  11     +  11,-  12     +  12,-  11
14:33:01.893 ->      +  12,-  11     +  12,-  10     +  13,-  33     +  11,-  12
14:33:01.893 ->      +  11,-  34     +  12,-  11     +  12,-  11     +  12,-  33
14:33:01.893 ->      +  12,-  11     +  11,-  34     +  11,-  11     +  12,-  34
14:33:01.893 ->      +  12,-  10     +  12,-  34     +  14,-  31     +  11,-  12
14:33:01.893 ->      +  12,-  33     +  12
14:33:01.893 -> Raw result in microseconds - with leading gap
14:33:01.893 -> rawData[68]: 
14:33:01.893 ->      -212600
14:33:01.893 ->      +4500,-4450     + 600,-1700     + 550,-1700     + 550,-1700
14:33:01.940 ->      + 550,- 600     + 550,- 600     + 550,- 550     + 600,- 550
14:33:01.940 ->      + 600,- 550     + 600,-1650     + 600,-1700     + 600,-1650
14:33:01.940 ->      + 550,- 600     + 550,- 550     + 550,- 600     + 600,- 550
14:33:01.940 ->      + 600,- 550     + 600,- 500     + 650,-1650     + 550,- 600
14:33:01.940 ->      + 550,-1700     + 600,- 550     + 600,- 550     + 600,-1650
14:33:01.940 ->      + 600,- 550     + 550,-1700     + 550,- 550     + 600,-1700
[... 34 lines cut ...]
14:34:43.783 ->      + 600,- 550     + 600,-1650     + 600,-1700     + 550,-1700
14:34:43.783 ->      + 550,- 600     + 550,- 550     + 700,- 450     + 550,- 600
14:34:43.783 ->      + 600,- 550     + 600,-1650     + 600,-1650     + 600,-1700
[... cut ...]  0019 0011 0019 0013 0045 0013 0043 0013 0043 0013 0019 0011 0045 0013 06C3 ";
14:34:44.119 -> 
14:34:44.119 -> PULSE_DISTANCE: HeaderMarkMicros=4500 HeaderSpaceMicros=4450 MarkMicros=600 OneSpaceMicros=1650 ZeroSpaceMicros=550 NumberOfBits=32 DecodedRawData: 0xB9460707
```

This is with the TSOP38238 that I currently have installed.

MARK_EXCESS_MICROS is a blunt weapon that can't always hide issues caused by wrong expected mark/space durations.

Please don't close this issue, it's still there.

ArminJo, 2022-01-12:
What are the results of ReceiverTimingAnalysis?
Can you enable line 37 `#define DEBUG // Activate this for lots of lovely debug output from this decoder.` and post results?
And you read and understand the new chapter?

I just checked with my samsung remote and my TSOP31238 and all was fine.

> 14:29:47.658 ->      +  17,-   6     +  12,-  33     +  12,-  34     +  12,-  11

What do you expect to decode with this variation in timing (from 6 to 11 or 12)?

> Please don't close this issue, it's still there.

Please do not claim the library to be errorneous, if you have such an "random" input signal from your receiver!
Welcome to real world signal processing. 😀