Arduino-IRremote/Arduino-IRremote #958: [Bug]: Didn't work with ESP32 C3

Skywalkerf34, 2022-01-26:
### The type of board you are using.

ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 20

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

I am using ARDUINO 1.8.13
IR Library just fresh downloaded
Board def is ESP32 Expressif systems version 2.0.1
Board is set with ESP32C3 dev
The file compiles well but cannot get any value from the receiver

With the same wiring changing to another ESP32 WROOM board ( and target), it works very well

### What happens?

No value for IR remote

### The serial output which indicates the error happened.

```shell
no error
```

### What were you expecting to happen?

got IR code

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

AnalysIR, 2022-01-26:
> Board is set with ESP32C3 dev
> .......
> With the same wiring changing to another ESP32 WROOM board ( and target), it works very well

I suspect the C3 (RISC based MCU) is not the same as the original WROOM ESP32 (XTENSA MCU) and would require support to be added as a new platform/architecture.

Of course, if Espressif have made timer use identical(unlikely), then this would not be the case.

_{Might be a good idea to update the README to cover this}_

Skywalkerf34, 2022-01-26:
Hi,

Thanks for this prompt response,
For your information, I was using board package 2.0.1 from Espressif
now I just upgrade to very last 2.0.2 and it works !
This ESP32 C3 is still quite new and lot of libs give me trouble up to now

AnalysIR, 2022-01-26:
good to know :)

Did you check IR sending or just receiving?

Skywalkerf34, 2022-01-26:
@AnalysIR 
Just receiving for that time
I don't have IR Led available to try it for now
I am also discovering that this board reference changing now block some other part in my Firmware  that was working well before :-(  like the  sigmaDelta pwm

ArminJo, 2022-01-26:
@Skywalkerf34 
Thanks for the Info 👍 , I added a remark in the readme (coming soon).

closed 2022-01-26 by ArminJo