Arduino-IRremote/Arduino-IRremote #961: Error with including IRremote.hpp within a .h file

ArminJo, 2022-01-28:
### Discussed in https://github.com/Arduino-IRremote/Arduino-IRremote/discussions/960

<div type='discussions-op-text'>

<sup>Originally posted by **AnthonyEdmonds** January 28, 2022</sup>
I have an old program that was made using version 2 of IRremote, and I was looking to update it with version 3.

The program is structured like this, with the remote parts held in two files:

- display-cabinets.ino
  - remote.h
  - remote.cpp
  - _... other unrelated .h and .cpp files_

`remote.h` and `remote.cpp` held the `#include <IRremote.h>` line, which worked.

Attempting to include the new `<IRremote.hpp>` in `remote.h` and `remote.cpp` does not work, due to the documented multiple definitions error.

However, attempting to include `<IRremote.hpp>` in `display-cabinets.ino` and `<IRremoteInt.h>` in `remote.h` and `remote.cpp` also does not work with the following error: `Seems you use #include IRremoteInt.h in a file not containing main().` with further complaints about the `RAW_BUFFER_LENGTH`, which I tried to define in `remote.h` to no effect.

For reference, none of my programs of files contain a `main()` method.

I do not understand what the reference in the readme about the multiple definitions error is asking me to do; I've muddled my way through C/C++ so far, but this feels above my level.

I just want to include IRremote to receive IR signals in `remote.h`, like I did in version 2.

Any thoughts would be useful.

---

Here's my non-working code, set back to how it was prior to attempting to define `RAW_BUFFER_LENGTH`.

### display-cabinets.ino
```cpp
#include <IRremote.hpp>

#include "cabinets.h"
#include "colours.h"
#include "effects.h"
#include "leds.h"
#include "palettes.h"
#include "power.h"
#include "remote.h"
#include "timers.h"

void setup()
{
    Serial.begin(9600);
    Serial.println("Display Cabinets online!");
      
    setupRemote();
    setupLeds();
}

void loop()
{
    handleTimers();
}
```

### remote.h
```cpp
#ifndef REMOTE_H
    #define REMOTE_H

    #include <IRremoteInt.h>

    const byte IR_PIN = 2;
    const unsigned long POWER_CODE = 1585296348;
    const unsigned long UP_CODE = 102834823;
    const unsigned long UP_ALT_CODE = 86057202;
    const unsigned long DOWN_CODE = 3728385585;
    const unsigned long LEFT_CODE = 3931694629;
    const unsigned long LEFT_ALT_CODE = 2270411356;
    const unsigned long RIGHT_CODE = 1770599057;
    const unsigned long CENTRE_CODE = 1990685718;
    const unsigned long HOME_CODE = 4168409336;
    const unsigned long BACK_CODE = 3948472246;
    const unsigned long MENU_CODE = 4118076481;
    const unsigned long PLUS_CODE = 2007463339;
    const unsigned long MINUS_CODE = 2220078501;

    extern bool receivingCodes;
    extern unsigned long firstReceivedCode;
    extern unsigned long secondReceivedCode;
    extern unsigned long thirdReceivedCode;

    void setupRemote();

    void handleRemote();

    void processReceivedCode();

    void processReceivedCodes();

    void handleButton();

    unsigned long determineButtonCode();
#endif
```

### remote.cpp
```cpp
#include "remote.h"

#include <IRremoteInt.h>
#include "adjustments.h"
#include "effects.h"
#include "palettes.h"
#include "power.h"

bool receivingCodes = false;
unsigned long firstReceivedCode = 0;
unsigned long secondReceivedCode = 0;
unsigned long thirdReceivedCode = 0;    

// Function definitions, etc...
```</div>

ArminJo, 2022-01-28:
You can safely delete the 3 lines
```
#if !defined(RAW_BUFFER_LENGTH)
#error Seems you use #include IRremoteInt.h in a file not containing main(). Please define RAW_BUFFER_LENGTH with the same value as in the main program and check if the macros IR_SEND_PIN and SEND_PWM_BY_TIMER are defined in the main program.
#endif
```
You may now also use:
```
#define USE_IRREMOTE_HPP_AS_PLAIN_INCLUDE
#include <IRremote.hpp>
```

ArminJo, 2022-01-28:
Thanks again for reporting 🥇 .

closed 2022-01-28 by ArminJo

AnthonyEdmonds, 2022-01-28:
I presume you mean to delete the lines in the library file; I'll give that a go tonight, thanks for the reply!

commit 7a84ab80 (2022-01-28): Closes #961
commit 8622513f (2022-01-28): Closes #961