Arduino-IRremote/Arduino-IRremote #973: [Bug]: Using IRremote together with Wifi crashes my esp8266 boards

hneel, 2022-02-17:
### The type of board you are using.

Other - please specify below

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Unknown

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 4

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

I'm using a Wemos Lolin D1  ESP8266 board. Originally I was working on a project o PlatformIO, consisting of Wifi, Mqtt, HeatpumpIR . When I wanted to add a IRremote reciever I experienced crashes. 
I was able to reproduce this problem with a simple sketch on Arduino:

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <IRremote.hpp>
#include "secrets.h"

#define PIN_IRTX    13  // D7
#define PIN_IRRX    4   // D2

static const char*  wifi_ssid     = SECRET_WIFI_SSID;
static const char*  wifi_pass     = SECRET_WIFI_PASS;

void setup() 
{
  pinMode( LED_BUILTIN, OUTPUT );
[... 355 lines cut ...]
11:05:32.261 -> tail 4
11:05:32.261 -> chksum 0xcc
11:05:32.261 -> load 0x3fff20b8, len 40, room 4 
11:05:32.261 -> tail 4
11:05:32.261 -> chksum 0xc9
11:05:32.261 -> csum 0xc9
11:05:32.261 -> v00042c90
11:05:32.261 -> ~ld
```

### What were you expecting to happen?

I was expecting no such crashes.

### Additional context.

Perhaps related to this issue?
https://github.com/crankyoldgit/IRremoteESP8266/issues/1073

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2022-02-17:
Please remove the long dump, it is useless for all people which do not have access to the build .elf file. Use the ESP Exception decoder with this dump, and post its result!

hneel, 2022-02-17:
OK, here is the result:

```
Exception 0: Illegal instruction
PC: 0x40201084
EXCVADDR: 0x00000000

Decoding stack results
0x40100b9c: malloc(size_t) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 821
0x401004b4: __digitalWrite(uint8_t, uint8_t) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 87
0x40100290: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 37
0x401002d8: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 44
0x40100429: millis() at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring.cpp line 193
0x4020bea4: do_memp_malloc_pool at core/memp.c line 255
0x40100dfc: calloc(size_t, size_t) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1176
0x40100f66: __wrap_spi_flash_read(uint32_t, uint32_t*, size_t) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_phy.cpp line 309
0x4020207a: loop_task(ETSEvent*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 213
0x40100b4c: umm_init() at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 476
0x401000ac: app_entry_redefinable() at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 325
0x401010a0: _stopPWM(uint8_t) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_waveform_pwm.cpp line 264
0x401010a0: _stopPWM(uint8_t) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_waveform_pwm.cpp line 264
0x40100290: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 37
0x401004b4: __digitalWrite(uint8_t, uint8_t) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 87
0x401010a0: _stopPWM(uint8_t) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_waveform_pwm.cpp line 264
0x401010a0: _stopPWM(uint8_t) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_waveform_pwm.cpp line 264
0x40100290: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 37
0x401004b4: __digitalWrite(uint8_t, uint8_t) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 87
0x40100230: ets_intr_unlock() at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 167
0x40100290: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 37
0x402010b2: setFeedbackLED(bool) at D:\Documenten\Arduino\libraries\IRremote\src/IRFeedbackLED.hpp line 115
0x40100230: ets_intr_unlock() at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 167
0x401001d9: IRTimerInterruptHandler() at D:\Documenten\Arduino\libraries\IRremote\src/IRReceive.hpp line 1437
0x40100290: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 37
0x401002d8: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 44
0x401002db: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 44
0x401002d8: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 44
0x40100290: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 37
0x40100258: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x401002db: timer1_isr_handler(void*, void*) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_timer.cpp line 44
0x40100258: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100258: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
[... cut ...]
0x40201158: IRrecv::begin(unsigned char, bool, unsigned char) at D:\Documenten\Arduino\libraries\IRremote\src/IRReceive.hpp line 102
0x40100258: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100279: esp_schedule() at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 128
0x40202242: loop_wrapper() at C:\Users\Hans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 205

```

hneel, 2022-02-17:
I noticed something about the feedback LED in the output, and I tried the DISABLE_LED_FEEDBACK for a change. This seems to help. No more crashes so far.

ArminJo, 2022-02-17:
Thanks a lot for reporting 👍.
Can you do me a favor and test again with feedback LED enabled, but line 105 in IRFeedback led.hpp
changed from:
```
#if defined(ESP32)
IRAM_ATTR
#endif
void setFeedbackLED(bool aSwitchLedOn) {
```
to: 
```
#if defined(ESP32) || defined(ESP8266)
IRAM_ATTR
#endif
void setFeedbackLED(bool aSwitchLedOn) {
```
?
Thanks in advance.
Armin

ArminJo, 2022-02-17:
Thanks, it is fixed now 😀

commit 0ef17d23 (2022-02-18): Fixes #973 ESP8266 problem with Feedback LED. Thanks to hneel!