Arduino-IRremote/Arduino-IRremote #997: [Bug]: ESP32 Arduino development release 2.0.x doesn't work

tomaculum, 2022-06-26:
### The type of board you are using.

ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

send: 4

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [X] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [X] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. add the development release link from [espressif](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) to your additional boards manager urls: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json

2. load and run the SimpleSender Example

### What happens?

The code compiles and runs but there are no ir signals emmited by the ir diode. If I downgrade to the stable release, it works, therefore there should be nothing wrong with my wiring.

### The serial output which indicates the error happened.

```shell
there is no error message or anything related, it just runs as it would emit signals and displays all the output as expected
```

### What were you expecting to happen?

the esp32 should send the ir commands

### Additional context.

I did try to use platform.io since it seemed to work there but I noticed, that they are using the stable release. If I switch to the development release, it also doesn't work.

Another library I use in my project needs the 2.0.x version to work.

Hopefully this is just something small and easy to fix (:

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2022-06-26:
What ist your definition of stable release?

tomaculum, 2022-06-26:
I am using the terms of their installing guide:

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

There is one link for esp32 boards for a stable release and one for a development release. The stable release is something 1.x and the development version is 2.x.

NickTucker42, 2022-06-27:
I am also seeing the very same issues.  ESP32, IR on pin 4 and Library version 3.7.0 (Downloaded just now).  SimpleSender does not appear to transmit any data.
Downgrading to Library version 2.9 works with no issue.

ArminJo, 2022-06-28:
I can confirm this problem. ESP core 2.0.1 still works, but 2.0.3 fails. 
The reason is, that for 2.0.2 Ledc was refactored to use ESP-IDFs api.

NickTucker42, 2022-06-28:
Thanks Armin for looking into this..  

I have a question though.  Does this mean 2.9 does not use ledc?  
All I am doing is swapping out the 3.7.x version for the 2.9 version in my arduino\libraries folder to get it to work, I'm not changing any of the board\core versions.

ArminJo, 2022-06-28:
I do not know any 2.9 version, latest 2 is 2.8.1 and this version uses ledc.
**If you disable line 270** 
`#define SEND_PWM_BY_TIMER // We do not have pin restrictions for this CPU's, so lets use the hardware PWM for send carrier signal generation`
in PinDefinitionsAndMore.h, **you can avoid this bug**.

NickTucker42, 2022-06-28:
I was going by this line in README.md --> **### [Version 2.9.0](https://github.com/z3t0/Arduino-IRremote/releases) - work in progress**
and this one works with no issue.

tomaculum, 2022-07-03:
The new v.3.7.1 version fixed this issue, thank you very much!

closed 2022-07-03 by tomaculum