Arduino-IRremote/Arduino-IRremote #999: sendSamsungRepeat() sending packets but not registering with Samsung TV

ArminJo, 2022-07-10:
### Discussed in https://github.com/Arduino-IRremote/Arduino-IRremote/discussions/998

<div type='discussions-op-text'>

<sup>Originally posted by **zachleedogg** July 10, 2022</sup>
I have a Samsung TV  "The Frame" responding to normal single commands. But for some reason it is ignoring repeat commands. I'll need to set up a second receiver to properly capture the repeat timings, but I'm not sure what exactly the repeat protocol should look like.

I am using sendSamsung(address,cmd, x) where x is the number to repeat. Not actually calling the sendSamsungRepeat explicitly, though I have tried it with timer and had the same results.

See this stack overflow thread I just started before I realized I could discuss here.

https://stackoverflow.com/questions/72924539/why-does-samsung-tv-ir-sensor-ignore-repeat-codes</div>

ArminJo, 2022-07-10:
@zachleedogg Thanks for reporting!
I removed the special send repeat handling of Samsung, because I could not find any reference for it any more, but 2 references for regular repeat handling (including my ownn Samsung remote).
So I changed it ( ir_Samsung.hpp, IRremoteInt.h, IRSend.hpp) . 
Please get the version from repository (at least this 3 files and test again.
Thanks
Armin

zachleedogg, 2022-07-10:
Dude @ArminJo you are a rockstar. Can't believe the quick turn-around on this one. Can confirm it works exactly as intended. TV responds to a hold correctly now.

closed 2022-07-13 by ArminJo