Arduino-IRremote/Arduino-IRremote #1001: [Bug]: Pi Pico not working

DaleMitchell, 2022-07-12:
### The type of board you are using.

Other - please specify below

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

LG

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 1 (supposedly GP1)

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [X] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [X] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [X] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [X] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. Installed the latest Arduino IDE and latest RP2040 Arduino core (https://github.com/earlephilhower/arduino-pico).
2. Changed the IR_RECEIVE_PIN by editing the `PinDefinitionsAndMore.h` file in the example:
```
#elif defined(ARDUINO_ARCH_RP2040) // Arduino Nano Connect, Pi Pico with arduino-pico core https://github.com/earlephilhower/arduino-pico
#define IR_RECEIVE_PIN 1
```
3. Connected GP1 to the gate of a N-channel MOSFET with an IR LED connected to the MOSFET appropriately. Tested pin 1 using `digitalWrite(1, HIGH)`, which works. But the example does not ever change the state of GP1. I tested with an oscilloscope, using single shot trigger.

### What happens?

The selected pin never changes the state of the GPIO.

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2022-07-13:
Does the original example with `#define IR_RECEIVE_PIN      15  // to be compatible with the Arduino Nano RP2040 Connect (pin3)` work for you?

DaleMitchell, 2022-07-15:
No, I did not change the pin for receiving, I changed it for sending.

If I run the example completely unmodified, using 16 as the send pin, it does not work either. I confirmed the pin is working with a simple blink sketch. This library is simply not working for the Pi Pico (RP2040). I tested a different IR library, https://github.com/ukw100/IRMP, at the most [recent commit](https://github.com/ukw100/IRMP/tree/33abc2d822f75b4e09f0f2949e99593bf8901a1f) as of me typing this comment, and it works just fine. I will have to use IRMP instead of this Arduino-IRremote library.

daniloc, 2022-07-16:
I am finding the same issue with a Pi Pico.

I note, @ArminJo, that you're handling the problem. Any chance you can advise on a workaround? Thanks for your help.

ArminJo, 2022-07-16:
Is fixed, but not for all platforms. RC5X is still not working.

daniloc, 2022-07-16:
@ArminJo Thank you so much for your generous and prompt attention.

I can confirm everything is working perfectly on my Pico. Appreciate your help with my project!

ArminJo, 2022-07-16:
It is fixed and tested now.

closed 2022-07-16 by ArminJo