Arduino-IRremote/Arduino-IRremote #1003: [Bug]: Errors compiling for SAMD51 ('volatile struct TC_STATUS_Type::<unnamed>' has no member named 'SYNCBUSY' and others)

JetForMe, 2022-07-25:
### The type of board you are using.

Arduino SAMD board (Zero, MKR*)

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Other - please specify below

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 4

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

Attempt to compile the ReceiveDemo example, or this code

```
#include <Arduino.h>
#include <IRremote.hpp>
```

(Literally just that in a `.ino` file.)

For Adafruit Feather M4 Express.

### What happens?

I have a very simple sketch that fails to compile with missing symbols in the header. It's not clear to me how to ensure the right headers are being included. The board is a Feather M4 Express. The IRremote headers seem to have properly identified it as an ATSAMD (it's a SAMD51), as the first error occurs in the neighborhood of some ATSAMD preprocessor checks.

My code is literally just this:

```
#include <Arduino.h>
#include <IRremote.hpp>
```

The errors:

```
/Users/rmann/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -Os -Werror=return-type -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD "-D__SKETCH_NAME__=\"\"\"target.ino\"\"\"" -DF_CPU=120000000L -DARDUINO=10813 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M4 Express\"" -I/Users/rmann/Library/Arduino15/packages/adafruit/hardware/samd/1.7.10/libraries/Adafruit_TinyUSB_Arduino/src/arduino -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M4 Express\"" -I/Users/rmann/Library/Arduino15/packages/adafruit/hardware/samd/1.7.10/libraries/Adafruit_TinyUSB_Arduino/src/arduino -I/Users/rmann/Library/Arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/Core/Include/ -I/Users/rmann/Library/Arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/DSP/Include/ -I/Users/rmann/Library/Arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.2/CMSIS/Device/ATMEL/ -I/Users/rmann/Library/Arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino -I/Users/rmann/Library/Arduino15/packages/adafruit/hardware/samd/1.7.10/variants/feather_m4 -I/Users/rmann/Documents/Arduino/libraries/IRremote/src /var/folders/16/jfb809_s2fz01ql7k494f6pc0000gn/T/arduino_build_756234/sketch/target.ino.cpp -o /var/folders/16/jfb809_s2fz01ql7k494f6pc0000gn/T/arduino_build_756234/sketch/target.ino.cpp.o
In file included from /Users/rmann/Documents/Arduino/libraries/IRremote/src/IRremote.hpp:284,
                 from /Users/rmann/Projects/Clients/SAVC/LaserTargetGame/target/target.ino:7:
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp: In function 'void setTimerFrequency(unsigned int)':
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp:1184:27: error: 'volatile struct TC_STATUS_Type::<unnamed>' has no member named 'SYNCBUSY'
 1184 |     while (TC->STATUS.bit.SYNCBUSY == 1) {
      |                           ^~~~~~~~
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp: In function 'void timerConfigForReceive()':
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp:1193:5: error: 'REG_GCLK_CLKCTRL' was not declared in this scope; did you mean 'REG_GCLK_GENCTRL0'?
 1193 |     REG_GCLK_CLKCTRL = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | IR_SAMD_TIMER_ID);
      |     ^~~~~~~~~~~~~~~~
      |     REG_GCLK_GENCTRL0
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp:1193:36: error: 'GCLK_CLKCTRL_CLKEN' was not declared in this scope; did you mean 'GCLK_CLKCTRL_IDs'?
 1193 |     REG_GCLK_CLKCTRL = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | IR_SAMD_TIMER_ID);
      |                                    ^~~~~~~~~~~~~~~~~~
      |                                    GCLK_CLKCTRL_IDs
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp:1193:57: error: 'GCLK_CLKCTRL_GEN_GCLK0' was not declared in this scope; did you mean 'GCLK_PCHCTRL_GEN_GCLK0'?
 1193 |     REG_GCLK_CLKCTRL = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | IR_SAMD_TIMER_ID);
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~
      |                                                         GCLK_PCHCTRL_GEN_GCLK0
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp:1162:29: error: 'GCLK_CLKCTRL_ID_TCC2_TC3' was not declared in this scope
 1162 | #define IR_SAMD_TIMER_ID    GCLK_CLKCTRL_ID_TCC2_TC3
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp:1193:82: note: in expansion of macro 'IR_SAMD_TIMER_ID'
 1193 |     REG_GCLK_CLKCTRL = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | IR_SAMD_TIMER_ID);
      |                                                                                  ^~~~~~~~~~~~~~~~
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp:1194:18: error: 'struct Gclk' has no member named 'STATUS'
 1194 |     while (GCLK->STATUS.bit.SYNCBUSY == 1) {
      |                  ^~~~~~
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp:1202:27: error: 'volatile struct TC_STATUS_Type::<unnamed>' has no member named 'SYNCBUSY'
 1202 |     while (TC->STATUS.bit.SYNCBUSY == 1) {
      |                           ^~~~~~~~
/Users/rmann/Documents/Arduino/libraries/IRremote/src/private/IRTimer.hpp:1214:46: error: 'TC_CTRLA_WAVEGEN_MFRQ' was not declared in this scope; did you mean 'TCC_WAVE_WAVEGEN_MFRQ'?
 1214 |     TC->CTRLA.reg |= TC_CTRLA_MODE_COUNT16 | TC_CTRLA_WAVEGEN_MFRQ | TC_CTRLA_PRESCALER_DIV64 | TC_CTRLA_ENABLE;
      |                                              ^~~~~~~~~~~~~~~~~~~~~
      |                                              TCC_WAVE_WAVEGEN_MFRQ
Using library IRremote at version 3.7.1 in folder: /Users/rmann/Documents/Arduino/libraries/IRremote 
exit status 1
Error compiling for board Adafruit Feather M4 Express (SAMD51).
```

### The serial output which indicates the error happened.

```shell
n/a
```

### What were you expecting to happen?

I expect the library to compile.

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2022-07-25:
Sorry, but SAMD51 is not supported. 
I changed the readme, to make it more explicitly.

JetForMe, 2022-07-25:
To make it more explicit? It *explicitly* said SAMD 51 was supported.

What would it take to get support for SAMD51? I spent some time already integrating this into one part of my project that uses SAMD21, assuming I'd be able to use it for the other part that uses SAMD51.

ArminJo, 2022-07-26:
> To make it more explicit? It _explicitly_ said SAMD 51 was supported.

Better ask "it" than open an issue here!

JetForMe, 2022-07-27:
> Better ask "it" than open an issue here!

Sorry, I mis-read the Readme. It just says "SAMD," which could reasonably be interpreted to mean all SAMD families. In any case, what would it take get support for the 51 series?

ArminJo, 2022-07-27:
I can try to provide code, but I have no means of testing it.
So If you are ready to test, we can try...

JetForMe, 2022-07-27:
For sure, thank you!

ArminJo, 2022-07-29:
I added the code, please test it now.

JetForMe, 2022-07-30:
That seems to compile and load, but I must not understand how sending works, as I'm unable to send a raw code to then receive, so I can't test if it can properly decode. I'll have to spend more time on this later. But at least it compiles!

closed 2022-08-07 by ArminJo