Arduino-IRremote/Arduino-IRremote #1007: [Bug]: Arduino Nano RP2040 Connect receive doesn't work

lucki1000, 2022-08-07:
### The type of board you are using.

Other - please specify below

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Unknown

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. Compile & flash ReceiveDemo without any changes on rp2040 
2. Physical connect v(on rp2040) with v(receiver), gnd(rp2040) with gnd(receiver) and signal(D3/GPIO15 rp2040) with signal(receiver)
3. run serial monitor 
4. I tested it with 2 TV remote controls 

### What happens?

Nothing.

### What were you expecting to happen?

The IR code that sended by the TV remote control

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2022-08-07:
Please test the new 3.8.0 version

lucki1000, 2022-08-07:
I have already added 3.8 as zip to Arduino ide, and compiled & flashed it to the rp2040.

ArminJo, 2022-08-07:
Please reinstall it via the arduino library manager and post the serial output of the ReceiveDemo.

lucki1000, 2022-08-07:
Done, 
[image]

ArminJo, 2022-08-07:
Which board and which core/fqbn? And please post the COMPLETE output, no a picture.

lucki1000, 2022-08-07:
Never Mind, sorry it looks like anything goes correctly, excepted the 5 Volt pin on my arduino just not.
But thank you for your support @ArminJo & nice rest sunday.

closed 2022-08-07 by lucki1000