Arduino-IRremote/Arduino-IRremote #1015: [Bug]: Ir_MagiQuest not returning data.

thanneman79, 2022-08-20:
### The type of board you are using.

Arduino ATmega328* board (UNO, Nano, etc.)

### What IDE are you using?

Other - please specify below

### What IR protocol are you using?

Other - please specify below

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

Simple recieve signal from MagiQuest wand. I'm working in VS2022, however it is the same with Arduino studio.

### What happens?

ir_MagiQuest.hpp appears passes data.cmd.magnitude to decodedIRData.extra on line 175 and  data.cmd.wand_id to decodedIRData.decodedRawData on line 176. However neither data.cmd.magnitude or data.cmd.wand_id are ever calculated so you just get 0 returned.

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2022-08-20:
You know how C unions work?
And please add a line
`#define DEBUG `
at the beginning of ir_MagiQuest.hpp and post the output.

thanneman79, 2022-08-22:
Apologies, was not familiar with union, I've educated myself on it.

defined DEBUG in ir_MagiQuest.hpp and tested with 4 different wands. To be safe I tested a simple remote to rule out wiring or the IR sensor itself.

Using library version 3.8.0
Ready to receive IR signals of protocols: NEC, Panasonic/Kaseikyo, Denon/Sharp, Sony, RC5, RC6, LG, JVC, Samsung, Whynter, Lego Power Functions, Bosewave , MagiQuest, Pulse Distance, Hash at pin 2
09:01:14:924 -> 00000000000000000000000000000000000000000000000000000000
Protocol=MagiQuest Address=0x0 Command=0x0 Extra=0x0 Raw-Data=0x0 55 bits MSB first
Send with: IrSender.sendMagiQuest(0x0, 0x0, <numberOfRepeats>, 0x0);

09:03:19:098 -> 00000000000000000000000000000000000000000000000000000000
Protocol=MagiQuest Address=0x0 Command=0x0 Extra=0x0 Raw-Data=0x0 55 bits MSB first
Send with: IrSender.sendMagiQuest(0x0, 0x0, <numberOfRepeats>, 0x0);

09:03:29:315 -> 00000000000000000000000000000000000000000000000000000000
Protocol=MagiQuest Address=0x0 Command=0x0 Extra=0x0 Raw-Data=0x0 54 bits MSB first
Send with: IrSender.sendMagiQuest(0x0, 0x0, <numberOfRepeats>, 0x0);

09:03:31:099 -> 00000000000010000000000000000000000000000000000000000000
Protocol=MagiQuest Address=0x0 Command=0x0 Extra=0x0 Raw-Data=0x4000000 55 bits MSB first
Send with: IrSender.sendMagiQuest(0x0, 0x0, <numberOfRepeats>, 0x0);

09:03:31:425 -> 00000000000000000000000000000000000000000000000000000000
Protocol=MagiQuest Address=0x0 Command=0x0 Extra=0x0 Raw-Data=0x0 55 bits MSB first
Send with: IrSender.sendMagiQuest(0x0, 0x0, <numberOfRepeats>, 0x0);

09:06:40:783 -> Protocol=NEC Address=0x0 Command=0xC Raw-Data=0xF30CFF00 32 bits LSB first
Send with: IrSender.sendNEC(0x0, 0xC, <numberOfRepeats>);

Protocol=NEC Address=0x0 Command=0xC Repeat gap=39500us

09:06:45:347 -> Protocol=NEC Address=0x0 Command=0x16 Raw-Data=0xE916FF00 32 bits LSB first
Send with: IrSender.sendNEC(0x0, 0x16, <numberOfRepeats>);

Protocol=NEC Address=0x0 Command=0x16 Repeat gap=39500us

09:06:49:565 -> Protocol=NEC Address=0x0 Command=0x5E Raw-Data=0xA15EFF00 32 bits LSB first
Send with: IrSender.sendNEC(0x0, 0x5E, <numberOfRepeats>);

09:06:49:677 -> Protocol=NEC Address=0x0 Command=0x5E Repeat gap=39500us

ArminJo, 2022-08-22:
This looks strange!
Can you add a line 
`IrReceiver.printIRResultRawFormatted(&Serial, true);`
after the line 
`IrReceiver.printIRSendUsage(&Serial);`
and post the extended output of 2 wands?

And thank you for posting the NEC output!!!

thanneman79, 2022-08-22:
Here you go!

07:49:05:966 -> �MDPR_START C:\Users\Tomato\source\repos\MagiQuestPictureFrame\IRTest\IRTest.ino from Aug 22 2022
Using library version 3.8.0
Ready to receive IR signals of protocols: NEC, Panasonic/Kaseikyo, Denon/Sharp, Sony, RC5, RC6, LG, JVC, Samsung, Whynter, Lego Power Functions, Bosewave , MagiQuest, Pulse Distance, Hash at pin 2
07:50:15:130 -> 0000000000000000000000000000000000000000000000000000000M
Protocol=MagiQuest Address=0x0 Command=0x0 Extra=0x0 Raw-Data=0x0 55 bits MSB first
Send with: IrSender.sendMagiQuest(0x0, 0x0, <numberOfRepeats>, 0x0);
rawData[112]: 
 -3276750
 + 250,- 800 + 250,- 850 + 250,- 850 + 250,- 850
 + 250,- 850 + 300,- 800 + 250,- 850 + 250,- 850
 + 300,- 800 + 300,- 800 + 550,- 600 + 550,- 650
 + 250,- 850 + 500,- 650 + 500,- 650 + 250,- 850
 + 250,- 850 + 250,- 850 + 500,- 650 + 300,- 800
 + 500,- 650 + 300,- 800 + 550,- 650 + 250,- 850
 + 500,- 650 + 250,- 850 + 500,- 650 + 500,- 650
 + 500,- 650 + 300,- 800 + 300,- 800 + 300,- 850
 + 250,- 850 + 250,- 850 + 250,- 850 + 250,- 850
 + 300,- 800 + 250,- 850 + 500,- 650 + 250,- 850
 + 250,- 850 + 250,- 850 + 250,- 850 + 250,- 850
 + 250,- 850 + 250,- 850 + 250,- 850 + 500,- 700
 + 500,- 650 + 500,- 650 + 500,- 650 + 200,- 900
 + 250,- 850 + 500,- 650 + 300,- 800 + 500

07:50:38:044 -> 0000000000000000000000000000000000000000000000000000000M
Protocol=MagiQuest Address=0x0 Command=0x0 Extra=0x0 Raw-Data=0x0 55 bits MSB first
Send with: IrSender.sendMagiQuest(0x0, 0x0, <numberOfRepeats>, 0x0);
rawData[112]: 
 -3276750
 + 200,- 900 + 200,- 900 + 250,- 850 + 250,- 850
 + 250,- 900 + 250,- 850 + 250,- 850 + 250,- 900
 + 200,- 900 + 250,- 850 + 550,- 650 + 500,- 650
 + 250,- 850 + 550,- 650 + 200,- 900 + 550,- 600
 + 550,- 650 + 550,- 600 + 250,- 900 + 500,- 650
 + 550,- 600 + 550,- 650 + 550,- 600 + 250,- 900
 + 500,- 650 + 250,- 850 + 550,- 650 + 250,- 850
 + 250,- 850 + 550,- 650 + 250,- 850 + 250,- 850
 + 250,- 900 + 200,- 900 + 250,- 850 + 250,- 850
 + 250,- 900 + 200,- 900 + 550,- 600 + 250,- 900
 + 250,- 850 + 250,- 850 + 250,- 850 + 250,- 900
 + 250,- 850 + 250,- 850 + 550,- 650 + 200,- 900
 + 250,- 850 + 550,- 650 + 200,- 900 + 250,- 850
 + 250,- 850 + 550,- 650 + 250,- 850 + 550

thanneman79, 2022-08-26:
Sorry it took awhile to get back to you. 

Made changes and tested all 5 wands we have. All had rawData of 112 and  8 minus values of over 700. As a side note I tried out a second IR receiver of a different build and brand.

samuelbarrionuevo, 2022-08-27:
> ### The type of board you are using.
>
> Arduino ATmega328* board (UNO, Nano, etc.)
> ### What IDE are you using?
>
> Other - please specify below
> ### What IR protocol are you using?
>
> Other - please specify [...]

ArminJo, 2022-08-27:
Thanks a lot for testing.
The improved Magiquest decoder is now in the repo and will be contained in the forthcoming 3.9.release.

closed 2022-08-27 by ArminJo

thanneman79, 2022-08-28:
@ArminJo thanks for taking the time to work with me. I did a quick test with the changes and it is working for all five wands. Ever need a tester again I'd be happy to help!

ArminJo, 2022-08-28:
@thanneman79 Thanks for testing the release version 👍