Arduino-IRremote/Arduino-IRremote #1031: [Bug]: ESP32 "HW TIMER NEVER INIT ERROR" with MicroGirs

whc2001, 2022-10-04:
### The type of board you are using.

ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Other - please specify below

### Pin(s) used for IR-receive, if not default.

ESP32 pin number: D15

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. Change the receive pin of MicroGirs example to D15
2. Compile and download
3. Send "r" with serial monitor
4. Press remote button

### What happens?

An error message is emitted before the data.

### The serial output which indicates the error happened.

```shell
E (4388) timer_group: timer_isr_callback_remove(244): HW TIMER NEVER INIT ERROR
+9050 -4400 +650 -1650 +600 -1650 +600 -1650 +600 -1650 +600 -1600 +600 -1650 +600 -1650 +600 -500 +650 -500 +600 -500 +650 -500 +600 -500 +650 -500 +600 -500 +650 -450 +650 -1650 +600 -500 +600 -500 +650 -500 +600 -500 +650 -500 +600 -500 +650 -500 +600 -500 +650 -1650 +600 -1650 +600 -1650 +600 -1600 +600 -1650 +600 -1650 +600 -1650 +600 -1650 +600 -40000
OK
```

### What were you expecting to happen?

The error should not happen, or to say at least should not be output into the serial.

### Additional context.

The error message makes it unusable with IrScrutinizer which the message is regarded as garbage and will trigger exception. Testing with `ReceiveDemo`, the error message won't appear. I have found [a discussion before](https://github.com/Arduino-IRremote/Arduino-IRremote/discussions/993) about this message, but the current release version is already 3.9.0. I also tried the latest version directly from the repo and the problem persists. Not sure if I am missing something.

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

whc2001, 2022-10-04:
Just found that if I don't stop the receiver after successfully receiving a signal, the error won't appear... But will that cause any problem?

Commenting out this line and the problem is away:
https://github.com/Arduino-IRremote/Arduino-IRremote/blob/e36bcaa7b12331ca7de906cde823b04bad2bb6cb/examples/MicroGirs/MicroGirs.ino#L290

ArminJo, 2022-10-04:
Which ESP32 core version do you use? 
There are bugs in the timer functions in some 2.x versions. See [here](https://github.com/espressif/arduino-esp32/issues/6730)

whc2001, 2022-10-04:
Ahh it's an upstream problem, no wonder. I am using 2.0.4 and the latest 2.0.5 also does not work. Downgrading to 2.0.1 solves the problem.

ArminJo, 2022-10-04:
Thanks for reporting! 👍

ArminJo, 2022-10-04:
BTW, can you please test if 2.0.1 and 2.0.5 are working if you change the line 1122 in IRTimer.hpp from
`#define TIMER_DISABLE_RECEIVE_INTR  if (s50usTimer != NULL) {timerEnd(s50usTimer); timerDetachInterrupt(s50usTimer);}`
to
`#define TIMER_DISABLE_RECEIVE_INTR  if (s50usTimer != NULL) {timerEnd(s50usTimer);}`
this could then be a stable fix to this problem.

whc2001, 2022-10-04:
Seems like the second fix won't work on 2.0.1 but works on 2.0.5. The first fix would work on both 2.0.1 and 2.0.5.

closed 2022-10-04 by ArminJo