Arduino-IRremote/Arduino-IRremote #1036: TinyIRReceiver does not work with ESP32C3 [Bug]:

0x0fe, 2022-10-20:
### The type of board you are using.

ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

RC5, RC6

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 8

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [ ] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

The TinyIRReceiver version does not work with ESP32C3 and is even pulling the OUTPUT pin high, which is very strange. I checked the source and found nothing that the output was intialized as an INPUT_PULLUP, i changed to INPUT to desiable the internal pullup, but the problem is still there.
I could not find anythin suspicious in the librart itself so i can only assume it has to do with the "fast" read function,not sure? Anyway it doesnt worK.
As a side note i wonder why the RMT peripheral is not used for the ESP32 implementation since this hardware block is on all ESP32 chips, including the C3 and certainly offloads the IR reception, the current implementation with interrupts and timer to measure intervals being less than ideal, especially in an RTOS where interrupt intensive tasks should be avoided.

### What happens?

the input pin is somehow set to an HIGH state, which prevents anything to work and may even damage some IR receivers. I changed the intialisation that was cvery strangely set to INPUT_PULLUP (a pullup on an input pin should, at the very least, be an option disabled by default). but it did not fixed the problem.

### The serial output which indicates the error happened.

```shell
nothing in the uart output
```

### What were you expecting to happen?

well, that the INPUT would not be driven, and that the library would work, IRremote works, this problem is only with TinyIRReceiver.

### Additional context.

none

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2022-11-08:
I just tested it wiht an ESP32 DevkitV1 and it works for me!
So it must be an error in the C3 core :-(.

closed 2022-11-08 by ArminJo

ArminJo, 2022-11-10:
It is the MinimalReceiver example!

0x0fe, 2022-11-10:
oh, strange, i also don't understand why the input is set as pullup.

ArminJo, 2022-11-11:
You can debug it in the simulator there are the latest sources and input is NOT set as pullup any more..