Arduino-IRremote/Arduino-IRremote #1052: [Bug]:

mouseart, 2022-11-29:
### The type of board you are using.

Other - please specify below

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Unknown

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

#include <Arduino.h>
#include <IRremote.h>

// const byte IR_RECEIVE_PIN=7; 

void setup() {
    Serial.begin(115200);
    Serial.println(F("Enabling IRin"));
    IrReceiver.begin(IR_RECEIVE_PIN,ENABLE_LED_FEEDBACK); //启动红外解码
    Serial.print(F("Ready to receive IR signals at pin "));
    delay(1000);
}

void loop() {

}

### What happens?
[... 454 lines cut ...]
            ^~~~~
/Users/mouseart/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.3/cores/arduino/Print.h:69:12: note: candidate: size_t Print::print(unsigned int, int)
     size_t print(unsigned int, int = DEC);
            ^~~~~
/Users/mouseart/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.3/cores/arduino/Print.h:70:12: note: candidate: size_t Print::print(long int, int)
     size_t print(long, int = DEC);
            ^~~~~
/Users/mouseart/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.3/cores/arduino/Print.h:71:12: note: candidate: size_t Print::print(long unsigned int, int)
     size_t print(unsigned long, int = DEC);
            ^~~~~
/Users/mouseart/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.3/cores/arduino/Print.h:72:12: note: candidate: size_t Print::print(double, int)
     size_t print(double, int = 2);
            ^~~~~

exit status 1

Compilation error: exit status 1
```

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2022-11-29:
Thanks for reporting, but please answer all questions of the issue form.

ArminJo, 2022-11-29:
Cores for 32 bit platforms, which lack the print function for 64 bit integer `size_t println(unsigned long long, int = DEC)` are no longer supported. E.g. seeduino core for SAMD21. Please open an issue for the core to support printing of 64 bit integer or activate the line [#define LAZY_32_BIT_CORE](https://github.com/Arduino-IRremote/Arduino-IRremote/blob/master/src/IRremoteInt.h#L116).

closed 2022-11-29 by ArminJo