Arduino-IRremote/Arduino-IRremote #1058: [Bug]: IR_USE_AVR_TIMER1 doesn`t work in attiny167

freskpe, 2022-12-12:
### The type of board you are using.

Other - please specify below

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

attiny167(AttinyCore by spence konde)

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

not receiver enable

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [X] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [X] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. Open SimpleSender sketch and erase all serial related lines code.
2. uncomment the next lines:
#define SEND_PWM_BY_TIMER         // Disable carrier PWM generation in software and use (restricted) hardware PWM.
#define IR_USE_AVR_TIMER1
#define IR_SEND_PIN 13  // pin used in the attiny167

3. Change file IRTimer.hpp at lines 293 (change to PB5  (pin 13)) and line 301 (change OC1BU to OC1BW)

4. compile and upload the sketch to your attiny167. 

### What happens?

using Timer1 in attiny167 doesn`t work. 

### The serial output which indicates the error happened.

```shell
no serial error output.
```

### What were you expecting to happen?

a NEC protocol command send.

### Additional context.

no additional context 

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2022-12-12:
Thanks for reporting 👍.  Bug is now fixed 😀

closed 2022-12-12 by ArminJo

freskpe, 2022-12-12:
Now, the timer selection its ok, but the sent data is random, i`m trying to send IrSender.Samsung(0x707, 0x7F, 0) and receiving

`Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Raw result in internal ticks (50 us) - with leading gap
rawData[2]: 
 -65535
 +1461
Sum: 1461
Raw result in microseconds - with leading gap
rawData[2]: 
 -3276750
 +73050
Sum: 73050

Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS=20
uint8_t rawTicks[1] = {255};  // Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Result as microseconds array - compensated with MARK_EXCESS_MICROS=20
uint16_t rawData[1] = {73030};  // Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Pronto Hex as string
char prontoData[] = "0000 006D 0001 0000 0AFA 06C3 ";

Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Raw result in internal ticks (50 us) - with leading gap
rawData[2]: 
 -284
 +88
Sum: 88
Raw result in microseconds - with leading gap
rawData[2]: 
 -14200
 +4400
Sum: 4400

Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS=20
uint8_t rawTicks[1] = {88};  // Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Result as microseconds array - compensated with MARK_EXCESS_MICROS=20
uint16_t rawData[1] = {4380};  // Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Pronto Hex as string
char prontoData[] = "0000 006D 0001 0000 00AA 06C3 ";

Protocol=UNKNOWN Hash=0x0 2 bits (incl. gap and start) received

Raw result in internal ticks (50 us) - with leading gap
rawData[4]: 
 -283
 +93,-94
 +89
Sum: 276
Raw result in microseconds - with leading gap
rawData[4]: 
 -14150
 +4650,-4700
 +4450
Sum: 13800

Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS=20
uint8_t rawTicks[3] = {93,94, 89};  // Protocol=UNKNOWN Hash=0x0 2 bits (incl. gap and start) received

Result as microseconds array - compensated with MARK_EXCESS_MICROS=20
uint16_t rawData[3] = {4630,4720, 4430};  // Protocol=UNKNOWN Hash=0x0 2 bits (incl. gap and start) received

Pronto Hex as string
char prontoData[] = "0000 006D 0002 0000 00B4 00B4 00AC 06C3 ";

Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Raw result in internal ticks (50 us) - with leading gap
rawData[2]: 
 -108
 +80
Sum: 80
Raw result in microseconds - with leading gap
rawData[2]: 
 -5400
 +4000
Sum: 4000

Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS=20
uint8_t rawTicks[1] = {80};  // Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Result as microseconds array - compensated with MARK_EXCESS_MICROS=20
uint16_t rawData[1] = {3980};  // Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Pronto Hex as string
char prontoData[] = "0000 006D 0001 0000 009B 06C3 ";

Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Raw result in internal ticks (50 us) - with leading gap
rawData[2]: 
 -304
 +68
Sum: 68
Raw result in microseconds - with leading gap
rawData[2]: 
 -15200
 +3400
Sum: 3400

Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS=20
uint8_t rawTicks[1] = {68};  // Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Result as microseconds array - compensated with MARK_EXCESS_MICROS=20
uint16_t rawData[1] = {3380};  // Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Pronto Hex as string
char prontoData[] = "0000 006D 0001 0000 0084 06C3 ";

Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Raw result in internal ticks (50 us) - with leading gap
rawData[2]: 
 -103
 +84
Sum: 84
Raw result in microseconds - with leading gap
rawData[2]: 
 -5150
 +4200
Sum: 4200

Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS=20
uint8_t rawTicks[1] = {84};  // Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Result as microseconds array - compensated with MARK_EXCESS_MICROS=20
uint16_t rawData[1] = {4180};  // Protocol=UNKNOWN Hash=0x0 1 bits (incl. gap and start) received

Pronto Hex as string
char prontoData[] = "0000 006D 0001 0000 00A2 06C3 ";`

the arduino code is:

`
#include <Arduino.h>

#define SEND_PWM_BY_TIMER         // Disable carrier PWM generation in software and use (restricted) hardware PWM.
#define IR_USE_AVR_TIMER1         // Tested #define USE_TIMER_CHANNEL_B too... and commented line.
#define NO_LED_FEEDBACK_CODE

#include <IRremote.hpp>

uint16_t sAddress = 0x707;
uint8_t sCommand = 0x7F;
uint8_t sRepeats = 0;

void setup() {
    IrSender.begin(); // Start with IR_SEND_PIN as send pin and if NO_LED_FEEDBACK_CODE is NOT defined, enable feedback LED at default feedback LED pin
}

void loop() {
    IrSender.sendNEC(sAddress, sCommand, sRepeats);
    delay(500);  // delay must be greater than 5 ms (RECORD_GAP_MICROS), otherwise the receiver sees it as one long signal
}`

Environment:
bareboard attiny167 
commented lines 297 and 315 of file IRTimers.hpp 
uncommented lines 299 and 317 of file IRTimers.hpp

freskpe, 2022-12-12:
Environment:
bareboard attiny167
commented lines 297 and 315 of file IRTimers.hpp
uncommented lines 299 and 317 of file IRTimers.hpp

ArminJo, 2022-12-12:
It works for me with Digispark Pro and SendDemo!

freskpe, 2022-12-12:
i will test tomorrow flashing the attiny167 with Digispark Pro and will comment to you. thank you very much!

freskpe, 2022-12-12:
my attiny167 is running at 8MHz, Digispark Pro Run at 16MHz could be this the problem?

ArminJo, 2022-12-12:
Digispark has no PB5 available!

freskpe, 2022-12-12:
ok, i will change the esp8266 Receiver by an arduino nano and will make another tests. i will check the bareboard and connections to discard hardware issues.

freskpe, 2022-12-13:
> It works for me with Digispark Pro and SendDemo!

are you using the PinDefinitionsAndMore.h file to test the Digispark Pro? can you test with another pin like PB1?

ArminJo, 2022-12-13:
I tested it with PB1 and PB3

freskpe, 2022-12-14:
The problem was the attiny167 fuses... i compare Digispark Pro Fuses with attiny167 and default fuses divide the bus clock by 8... 

in another order... if i comment the line #define  DISABLE_CODE_FOR_RECEIVER the device stop workin, i will continue working and notice you anything...  i dont know how to close a issue in github. sorry and thank you