Arduino-IRremote/Arduino-IRremote #1061: [Bug]: Sony command not sending correctly

ntzb, 2022-12-14:
### The type of board you are using.

Arduino ATmega328* board (UNO, Nano, etc.)

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Sony

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [X] ReceiveDemo
- [X] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [X] SimpleSender
- [X] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] MinimalReceiver
- [ ] ReceiveAndSend
- [X] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [ ] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

record a Sony TV "ON" command, and send it

### What happens?

the command is not picked up by the TV

### Additional context.

I'm trying to use IRremote to control a Sony TV. using release 3.9.0.
I used the rcv examples and got this in the serial:
`use IrSender.sendSony(0x1, 0x15, <numberOfRepeats>);`
I tried using it with 0 repeats, but the TV doesn't respond. (also noticed there's a setting of address size and number of bits, but the address is small, and the serial also reported 12 bits).

I previously had IRMP (which you maintain as well @ArminJo) on this board, and there, the rcv saw the code as:
```
tvOn.protocol = IRMP_SIRCS_PROTOCOL;
tvOn.address = 0x0;
tvOn.command = 0x95;
```
which worked without issues.
when switching back and forth between IRMP and IRremote on the same board, the IRMP sketch works, while the IRremote one doesn't.
am I missing something?
will appreciate any help.

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

bengtmartensson, 2022-12-14:
Sony devices in general require three copies of the signal to react. (My Sony TV (AF9) is happy with two, but it definitely ignores one.) So, my suggestion is to try to send three of them.

ntzb, 2022-12-14:
wow, that was actually it!
I tried 1 repetition, but it worked only about 30% of the times.
I tried 2 repetitions and it works all the time.

was this documented somewhere and I missed it?

closed 2022-12-14 by ntzb