Arduino-IRremote/Arduino-IRremote #1086: [Bug]: IR stops receiving after while

AhmedHamam001, 2023-02-08:
### The type of board you are using.

Arduino ATmega328* board (UNO, Nano, etc.)

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

nano

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

i used simple relieve demo skitch 
and setup my codes and buttons 
after pressing some buttons the receiving stops 
note that the loop still works fine 

### What happens?

i have a code and it works fine with ir remote but at some point ir stops working The loop function is running time after time as expected but at some point the library decode function stops working until I reset the board. Some times this happens fast and other times it is after many receptions. the board isn,t frozen but the ir code is , i alredy use the latest vertion of the liberary

### What were you expecting to happen?

the code should run as expected and not stop receiving after while

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2023-02-08:
The example runs endlessly at my Nano, so it must be an error in your board or code.

closed 2023-02-08 by ArminJo

AhmedHamam001, 2023-02-08:
i have searched this problem and i found another person got the same as me 
https://arduino.stackexchange.com/questions/13189/arduino-irremote-stops-working

ArminJo, 2023-02-08:
He is using the old deprecated an not supported version of decode()!!!

AhmedHamam001, 2023-02-12:
is ir library interfere with adfruit neo pixels library ?
is there any problem using ir library with it on Arduino nano or is there any restrictions should i take into consideration?
the problem happens while using both libraries  

i am using this code for loop function 

void loop() {

  unsigned long currenttime = millis();

  //displaymood(1);

 readTheTime();
    displayTheTime();
  if (IrReceiver.decode()) {

    if (IrReceiver.decodedIRData.address == 0) {

      RGB_Remote();
    }
  }

and this is the  RGB_Remote(); fun which i call inside the loop and decode function 

void RGB_Remote() {  //  Functions for IR Remote
delay(x);
  switch (IrReceiver.decodedIRData.command) {  //  IR Remote buttons

    case 0x16:                       // num 0
      clockMinuteColour = 0x333399;  // blue white
      clockHourColour = 0x333399;    // blue white
      delay(x);
      stripClock.show();
      delay(x);
      IrReceiver.resume();

      break;

    case 0x0C:  // num 1

      clockMinuteColour = 0xFF0000;  // pure red
      clockHourColour = 0xFF0000;    // pure green
       delay(x);
      stripClock.show();
      delay(x);
      IrReceiver.resume();

      break;

    case 0x18:                       // num 2
      clockMinuteColour = 0x00FF00;  // pure green
      clockHourColour = 0xC00FF00;   // pure green
       delay(x);
      stripClock.show();
      delay(x);
      IrReceiver.resume();
      break;

    case 0x5E:                       // num 3
      clockMinuteColour = 0x0000ff;  // pure blue
      clockHourColour = 0x0000ff;    // pure blue
       delay(x);
      stripClock.show();
      delay(x);
      IrReceiver.resume();
      break;

    case 0x8:                        // num 4
      clockMinuteColour = 0xDCA21B;  // orange
      clockHourColour = 0xDCA21B;    // orange
       delay(x);
      stripClock.show();
      delay(x);
      IrReceiver.resume();
      break;

    case 0x1C:                       // num 5
      clockMinuteColour = 0xC800C8;  // magenta
      clockHourColour = 0xC800C8;    // magenta
       delay(x);
      stripClock.show();
      delay(x);
      IrReceiver.resume();
      break;

    case 0x5A:                       // num 6
      clockMinuteColour = 0xFFC800;  // yellow
      clockHourColour = 0xFFC800;    // yellow
       delay(x);
      stripClock.show();
      delay(x);
      IrReceiver.resume();
      break;

    case 0x42:                       // num 7
      clockMinuteColour = 0x00FFFF;  // cayan
      clockHourColour = 0x00FFFF;    // cayan
       delay(x);
      stripClock.show();
      delay(x);
      IrReceiver.resume();
      break;

    case 0x52:                       // num 8
      clockMinuteColour = 0xA06EE6;  // violet light
      clockHourColour = 0xA06EE6;    // violet light
       delay(x);
      stripClock.show();
      delay(x);
     IrReceiver.resume();
      break;

    case 0x4A:                       // num 9
      clockMinuteColour = 0xFA5A5A;  // onion
      clockHourColour = 0xFA5A5A;    // onion
       delay(x);
      stripClock.show();
      delay(x);
     IrReceiver.resume();
      break;

      case 0x45:                       // power
      clockMinuteColour = 0x000000;  // off
      clockHourColour = 0x000000;    // off
      delay(x);
      stripClock.show();
      delay(x);

     IrReceiver.resume();
      break;
      
case 0x7:  // EQ   ON / OFF DOWN LIGHT 
  
 ledState = !ledState;
 if (ledState == LOW) {
    
        stripDownlighter.fill(16777215, 0, LEDDOWNLIGHT_COUNT);
        delay(x);
     stripDownlighter.show();
     delay(x);
      } else {
       stripDownlighter.fill(00000000, 0, LEDDOWNLIGHT_COUNT);
     delay(x);
     stripDownlighter.show();
     delay(x);
      }

     delay(x);

      break;

      default:
     delay(2*x);
      IrReceiver.resume();
     
      break;

    
  

//  Ready to receive the next input
}

delay(x);  //  short delay
IrReceiver.resume();
}  //  End of My Remote   */

AhmedHamam001, 2023-02-12:
everything works fine but sometimes the RGB_Remote function don,t do anything and the rest of the code runs fine I mean the Arduino itself isn't frozen completely 
and when I reset Arduino  everything works fine again until i start changing colors again and it stops at some point 

note the x value is 50 so i have done 100 ms delay total 

I tried it without delay at all but it was freezing much faster 
I am still searching for the reason why this happened and if the problem was related to the Arduino nano itself not specifically mine but the nano gears and power like CPU power etc

ArminJo, 2023-02-13:
Have you read the readme?

AhmedHamam001, 2023-02-13:
thanks for the help,i am still new to dealing with neopixels and  I have read the instructions before and I did,t know that fast led library and adfruit have something in common i mean the interrupt problem so i ignored the line which checks if ir recive is ideal or not before strip. show  as I am not using the fast led library but after butting this in my code I think everything is just work fine now, not 100 % perfect but it,s way better than before 
thanks for your great work :)

ArminJo, 2023-02-14:
Thanks for your feedback!