Arduino-IRremote/Arduino-IRremote #1096: [Bug]: SoftPWM not working

GralfR, 2023-02-17:
### The type of board you are using.

Arduino ATmega328* board (UNO, Nano, etc.)

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Unknown

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [X] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [X] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [ ] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. connect a normal LED and Resistor between send-pin (3) and GND
2. connect an IR-receiver like TSOP1736 to 5V and GND on the Arduino
3. connect a normal LED with Resistor between 5V and signal-out of IR receiver
4. do not connect signal out of receiver to Arduino, just use the receiver and LED as an IR-indicator
5. flash this simple script to the Arduino:
```
void setup() {

}
void loop() {
analogWrite(3,128);
}
```

### What happens?

You will see both LEDs light up. The first LED on pin 3 is light up because of the analogWrite, the second LED indicates an IR-signal present. Where does this signal come from? There is no IR LED sending.

### What were you expecting to happen?

I've investigated a lot, thinking of any noise, tried some capacitors, covered the sending and the receiving LED,...
Nothing helps. I came across this problem, because my IR circuit had this problem. I build a circuit, which shall receive, learn and repeat/send the learned codes. So far for the first step I just send hardcoded RAW IR codes and disabled the receiving code. Because I was not using any IR-LEDs for testing, I was wondering, why the receiver-LED was indicating a signal when sending. I reduced the circuit, even went from breadboard to a soldered board to reduce wire length. I even get rid of any sending LED, just put a resistor to ground on pin 3. The receiver is still receiving noise, which only stops, if pin 3 is left open. To make a long story short, I fouund out, the analogWrite-function is the problem, not the IR library in general.

That's why I was looking for a way to not use the hardware timer and PWM and expected the SEND_PWM_BY_TIMER to do this. I tried `#undef SEND_PWM_BY_TIMER` but that did not help. Only the `#define USE_NO_SEND_PWM` did not produce any noise., but obviously does not work with IR receivers because of the missing modulation. So, what did I miss? Is a software-PWM implemented?

### Additional context.

If I misunderstood the compiler-options, there was a [modification](https://sui77.wordpress.com/2011/05/27/use-any-arduino-pin-for-transmission-with-the-irremote-libary-by-soft-pwm/) of the library, but on a very old version. I wasn't succeccfull to implement it, yet, but I hope it has already been done because of the SEND_PWM_BY_TIMER - option. Maybe there is something missing inside the documentation? I didn't find any example for sending RAW IR without using analogWrite().

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2023-02-18:
What do you expect should be changed in the source code to fix tyour problem?

GralfR, 2023-02-19:
Well, SEND_PWM_BY_TIMER seams to be defined = enabled by default, so the options should be changed first, to the opposit like USE_SOFT_PWM.
Then, if USE_SOFT_PWM is defined, any port can be used as send-pin and the function for sending uses digitalWrite instead of analogWrite like in the given modification-link in my first post.

At least, that's what I expected but maybe because of the inverted option SEND_PWM_BY_TIMER I was not able to get this activated.

ArminJo, 2023-02-19:
Have you ever looked into the implementation of analogWrite?

GralfR, 2023-02-21:
Sorry, no. I started programming AVR on assembler, so very close to the hardware. I use arduino just for "quick&dirty"-coding by implementing libraries and getting to a working POC quickly. Because libraries tend to have a lot of overhead using a lot of flash and RAM, which can be optimized by hand-coding in assembler. But that takes time - especially if external interfaces/protocolls are needed. So, I hoped, there is someone, that already dived deeper into this library and may explain to me, if I missed something or got something wrong. I don't understand, if SEND_PWM_BY_TIMER is working as intended or if it is a leftover of older versions, because it's now enabled by default, which does not make sense to me concerning the naming.