Arduino-IRremote/Arduino-IRremote #1110: [Bug]: Doesn't compile for Teensy 4.1

maxgerhardt, 2023-03-15:
### The type of board you are using.

Teensy board

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

Teensy 4.1

### The board manager URL used for unspecified board.

See ArduinoTeensy

### What IDE are you using?

PlatformIO IDE

### What IR protocol are you using?

Unknown

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. Create new PlatformIO project for Teensy 4.1
2. add library
```ini
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
lib_deps = https://github.com/Arduino-IRremote/Arduino-IRremote.git
```
3. Add example code from https://github.com/Arduino-IRremote/Arduino-IRremote/tree/master/examples/SimpleReceiver

### What happens?

Fails to compile.
```
In file included from .pio/libdeps/teensy41/IRremote/src/IRremote.hpp:274:0,
                 from src/main.cpp:67:
.pio/libdeps/teensy41/IRremote/src/private/IRTimer.hpp: In function 'void timerEnableReceiveInterrupt()':
.pio/libdeps/teensy41/IRremote/src/private/IRTimer.hpp:1240:43: error: 'pwm1_3_isr' was not declared in this scope
     attachInterruptVector(IRQ_FLEXPWM1_3, pwm1_3_isr);
                                           ^
```

### The serial output which indicates the error happened.

```shell
None relevant
```

### What were you expecting to happen?

Compilation os OK.

### Additional context.

Although https://github.com/Arduino-IRremote/Arduino-IRremote/pull/928 was merged, this fails to compile.

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2023-03-15:
Sorry Max, 
but there is no further support possible, see: https://forum.pjrc.com/threads/65912-Enable-Continuous-Integration-with-arduino-cli-for-3-party-libraries 😞
If you find a fix we can include it 😀

maxgerhardt, 2023-03-15:
Is it acceptable to use PlatformIO for the CI with Teensy boards? That's like a 2 minute job. I don't feel like massagin a non-conforming Arduino core to work with arduino-cli when PlatformIO works fine.

Of course the build will fail though without code modifications.

maxgerhardt, 2023-03-15:
Actually, hold up.  Looking at https://www.pjrc.com/teensy/td_download.html it looks like it installs regularly now without the custom installer just with the board package URL. So.. it should already work with arduino-cli?

ArminJo, 2023-03-15:
How do you do the CI with Platform IO? It might be acceptable, but I doubt the 2 minutes 😀

maxgerhardt, 2023-03-15:
You do it with thte template given in thet official documentation: https://docs.platformio.org/en/latest/integration/ci/github-actions.html

This is literally just installing Python, PlatformIO and executing the **dedicated** `pio ci` with a pointer to the source file(s) and for which board to compile. Let me just cook up something.

maxgerhardt, 2023-03-15:
Yes, the `.github/workflows` files are visible as normal.

closed 2023-08-12 by ArminJo