Arduino-IRremote/Arduino-IRremote #1112: [Bug]: Issue with Mi TV remote

NXC-Controls, 2023-03-18:
### The type of board you are using.

Other - please specify below

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

Atmega23

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

PlatformIO IDE

### What IR protocol are you using?

Unknown

### Pin(s) used for IR-receive, if not default.

Arduino pin number: INT0 / PD2 / 10

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [X] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

im using Mi TV remote 
remote picture URL : [(https://www.amazon.in/Mi-Remote-Suitable-Trend-Trail/dp/B09B4DDYDH)]

` if (IrReceiver.decode()) {
    if (IrReceiver.decodedIRData.flags & IRDATA_FLAGS_WAS_OVERFLOW) {
      Serial.println("IR code too long. Edit IRremoteInt.h and increase RAW_BUFFER_LENGTH");
    } else {
      // Print a short summary of received data
      IrReceiver.printIRResultShort( & Serial);
      IrReceiver.printIRResultRawFormatted( & Serial, true);
      
      Serial.println(F("Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS"));
      IrReceiver.compensateAndPrintIRResultAsCArray( & Serial, false); // Output the results as uint8_t source code array of ticks
      Serial.println(F("Result as microseconds array - compensated with MARK_EXCESS_MICROS"));
      IrReceiver.compensateAndPrintIRResultAsCArray( & Serial, true); // Output the results as uint16_t source code array of micros
    }

    if (IrReceiver.decodedIRData.protocol == UNKNOWN ) {
      // We have an unknown protocol, print more info
      IrReceiver.printIRResultRawFormatted( & Serial, true);
    }
    Serial.println("================================================="); // blank line between entries
  }

  IrReceiver.resume();`

### What happens?

getting Protocol=UNKNOWN error and every time received data is different for the same button.

`#define RECORD_GAP_MICROS   15000 ` but no luck

followed this issue but unable to understand , what to change for my remote [https://github.com/Arduino-IRremote/Arduino-IRremote/issues/857]

### The serial output which indicates the error happened.

```shell
Protocol=UNKNOWN Hash=0x72C500CF 48 bits (incl. gap and start) received
rawData[96]: 
 -3276750
 +1050,- 550
 + 600,-1200 + 550,- 600 + 550,- 900 + 550,-1200
 + 550,- 600 + 600,- 600 + 550,-1450 + 600,- 850
 + 600,- 550 + 600,-1450 + 600,-13200 +1000,- 550
 + 600,-1150 + 600,- 600 + 600,- 850 + 600,-1150
 + 600,- 550 + 600,- 550 + 600,-1450 + 600,- 850
 + 600,- 550 + 600,-1450 + 600,-13200 +1000,- 550
 + 600,-1200 + 550,- 600 + 600,- 850 + 600,-1150
 + 600,- 550 + 650,- 550 + 600,-1400 + 600,- 900
 + 550,- 600 + 600,-1400 + 600,-13200 +1000,- 600
 + 600,-1150 + 600,- 550 + 600,- 850 + 600,-1150
 + 600,- 550 + 600,- 600 + 600,-1450 + 600,- 800
 + 600,- 600 + 600,-1450 + 550
Sum: 108850
Result as internal ticks (50 us) array - compensated with MARK_EXCESS_MICROS
uint8_t rawTicks[95] = {21,11, 12,24, 11,12, 11,18, 11,24, 11,12, 12,12, 11,29, 12,17, 12,11, 12,29, 12,255, 20,11, 12,23, 12,12, 12,17, 12,23, 12,11, 12,11, 12,29, 12,17, 12,11, 12,29, 12,255, 20,11, 12,24, 11,12, 12,17, 12,23, 12,11, 13,11, 12,28, 12,18, 11,12, 12,28, 12,255, 20,12, 12,23, 12,11, 12,17, 12,23, 12,11, 12,12, 12,29, 12,16, 12,12, 12,29, 11};  // Protocol=UNKNOWN Hash=0x72C500CF 48 bits (incl. gap and start) received

Result as microseconds array - compensated with MARK_EXCESS_MICROS
uint16_t rawData[95] = {1030,570, 580,1220, 530,620, 530,920, 530,1220, 530,620, 580,620, 530,1470, 580,870, 580,570, 580,1470, 580,13220, 980,570, 580,1170, 580,620, 580,870, 580,1170, 580,570, 580,570, 580,1470, 580,870, 580,570, 580,1470, 580,13220, 980,570, 580,1220, 530,620, 580,870, 580,1170, 580,570, 630,570, 580,1420, 580,920, 530,620, 580,1420, 580,13220, 980,620, 580,1170, 580,570, 580,870, 580,1170, 580,570, 580,620, 580,1470, 580,820, 580,620, 580,1470, 530};  // Protocol=UNKNOWN Hash=0x72C500CF 48 bits (incl. gap and start) received
```

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2023-03-18:
What do you expect us to do for you?

NXC-Controls, 2023-03-19:
Hello arminjo,

If possible add support for this kind of remote or help me to create a decoding protocol for this remote.
I have very less knowledge of this so you need to guide me on how can i create a library file with all button decoding.

ArminJo, 2023-03-19:
And why do you consider this as an issue?