Arduino-IRremote/Arduino-IRremote #1139: [Bug]: REPEAT in SimpleReceiver does not work with JVC protocol

amc57, 2023-05-25:
### The type of board you are using.

Arduino ATmega328* board (UNO, Nano, etc.)

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

JVC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 2

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [ ] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

Test config: Arduino nano with ATmega328P, sensor module IRM-3638T
JVC IR Remotes: RM-SA911U, RM-SX463U 

### What happens?

SimpleReceiver does not work when a key on remote is being pressed and hold for long time.
There is REPEAT code detected in the beginning and after 2 seconds it failed with UNKNOWN protokol.
After that, no more code is recognized until the key is released.

There is no issue with RC5 protocol.

Trace log is attached. 

### The serial output which indicates the error happened.

```shell
14:20:33.535 -> START D:\Arduino\Test_JVC\Test_JVC.ino from May 25 2023
14:20:33.535 -> Using library version 4.1.0
14:20:33.535 -> Ready to receive IR signals of protocols: RC5, JVC, at pin = 2
14:20:39.787 -> Protocol=JVC Address=0xA3 Command=0x1E Raw-Data=0x1EA3 16 bits LSB first
14:20:39.787 -> Send with: IrSender.sendJVC(0xA3, 0x1E, <numberOfRepeats>);
14:20:39.787 -> 
14:20:39.822 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20300us
14:20:39.822 -> 
14:20:39.868 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:39.868 -> 
14:20:39.914 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20300us
14:20:39.914 -> 
14:20:39.961 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20300us
14:20:39.961 -> 
14:20:40.007 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20300us
14:20:40.007 -> 
14:20:40.038 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.038 -> 
14:20:40.070 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.116 -> 
14:20:40.162 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20300us
14:20:40.162 -> 
14:20:40.208 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20300us
14:20:40.208 -> 
14:20:40.254 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.254 -> 
14:20:40.301 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.301 -> 
14:20:40.348 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.348 -> 
14:20:40.395 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.395 -> 
14:20:40.441 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.441 -> 
14:20:40.488 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.488 -> 
14:20:40.534 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.534 -> 
14:20:40.580 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.580 -> 
14:20:40.627 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20300us
14:20:40.627 -> 
14:20:40.659 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:40.659 -> 
.
.
.
14:20:41.030 -> 
14:20:41.077 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:41.077 -> 
14:20:41.123 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20400us
14:20:41.123 -> 
14:20:41.169 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20400us
14:20:41.169 -> 
14:20:41.216 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:41.216 -> 
14:20:41.262 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:41.262 -> 
14:20:41.308 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:41.308 -> 
14:20:41.355 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:41.355 -> 
14:20:41.402 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:41.402 -> 
14:20:41.448 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:41.448 -> 
14:20:41.495 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20350us
14:20:41.495 -> 
14:20:41.541 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20400us
14:20:41.541 -> 
14:20:41.588 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20400us
14:20:41.588 -> 
14:20:41.635 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20400us
14:20:41.635 -> 
14:20:41.682 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20450us
14:20:41.682 -> 
14:20:41.729 -> Protocol=JVC Address=0xA3 Command=0x1E Repeat gap=20450us
14:20:41.729 -> 
14:20:41.761 -> Protocol=UNKNOWN 15 bits (incl. gap and start) received
14:20:41.761 -> Received noise or an unknown (or not yet enabled) protocol
14:20:41.761 -> rawData[30]: 
14:20:41.761 ->  -20500
14:20:41.761 ->  + 400,-1700
14:20:41.761 ->  + 400,-1700 + 400,- 650 + 400,- 650 + 400,- 650
14:20:41.761 ->  + 400,-1700 + 350,- 700 + 350,-1750 + 350,- 700
14:20:41.761 ->  + 350,-1750 + 350,-1750 + 300,-1850 + 250,-1850
14:20:41.807 ->  + 250,- 750 + 150
14:20:41.807 -> Sum: 23250
14:20:41.807 ->
```

### What were you expecting to happen?

There is no time limit for press and hold a key, so this behaviour is not correct

### Additional context.

It was still working about two months ago.

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2023-05-25:
Try `#define MARK_EXCESS_MICROS    100`

amc57, 2023-05-25:
No improvement. Unfortunately the behavior is the same

ArminJo, 2023-05-29:
Weak battery?

amc57, 2023-05-30:
Nope. The same behaviour with two different JVC remotes (s. Test config)