Arduino-IRremote/Arduino-IRremote #1140: [Bug]: Failing to work with Arduino-MBED-Core-RP2040, works with Arduino-PICO core.

chandradharkoneti, 2023-05-29:
### The type of board you are using.

Other - please specify below

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

Raspberry Pi Pico

### The board manager URL used for unspecified board.

https://github.com/arduino/ArduinoCore-mbed

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Other - please specify below

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [X] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [X] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. Compile the attached Code
2. Upload the Code
3. Expect desired behaviour and get disappointed.

### What happens?

It flashes the wrong pin with wrong timings and it doesn't work on PICO using Arduino's MBED OS.
The same code works for arduino UNO and MEGA(if using v3.9.0).
[... 305 lines cut ...]

Sketch uses 97796 bytes (4%) of program storage space. Maximum is 2097152 bytes.
Global variables use 44904 bytes (16%) of dynamic memory, leaving 225432 bytes for local variables. Maximum is 270336 bytes.
```

Compiles and Uploads, but **Does not work.** GP-6 flashes, no other pin works. GP-6 also does not provide any control signal.

It works well when using Earle F. Philhower, III's Board (3.9.0 and 4.1.2[with sendNEC removed]).
>[Raspberry Pi Pico/RP2040 (Earle F. Philhower, III) - v3.2.1](https://github.com/earlephilhower/arduino-pico)

I am using Arduino's Mbed OS RP2040 since that is published by Arduino and is the ***Official*** Board for RP2040.

I am having many more issues when using the following board:
Raspberry Pi Pico/RP2040 (Earle F. Philhower, III) - v3.2.1
for example: https://github.com/PaulStoffregen/OneWire/issues/123

Whenever I flash my Program that used OneWire to read temperatures from DS18b20 sensors, it bricks my PICO.
[PICO__LG_Control.zip](https://github.com/Arduino-IRremote/Arduino-IRremote/files/11590515/PICO__LG_Control.zip)

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2023-05-29:
Is SimpleSender working with the latest version from the repository?

chandradharkoneti, 2023-06-07:
From the repository? You mean not installed from arduino IDE?
I haven't tried that. Will try by tomorrow and let you know.