Arduino-IRremote/Arduino-IRremote #1152: [Bug]: BluePill (STM32F103C8) not sending data trough IR diode

milan1984, 2023-07-31:
### The type of board you are using.

STM32F1 board (BluePill)

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

PlatformIO IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [X] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [X] SimpleSender
- [X] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [X] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [X] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

RECEIVER PART
1 . Load examples\SimpleReceiver\SimpleReceiver.ino 
2. Build
3. Upload **into BluePill no.1**
4. Open the Serial Monitor on **BluePill no.1**

SENDER PART
5. Load  examples\SimpleReceiver\SimpleReceiver.ino
6. Build
7. Upload **into BluePill no.2**
8. Open the Serial monitor  **on BluePill no.2**

### What happens?

Sender (BluePill no. 2 )sends one package of date and stops (no repeat in "loop").
**No serial terminal was available!**

The receiver (BluePill no.1) received one packet of data and nothing more from the sender).
The data is printed as "UNKNOWN" protocol with no matching to what was sent.

For testing ( when the receiver stops receiving messages ), using a remote from LG TV, pressing some buttons, and pointing the remote in the receiver's direction, the receiver continues to correctly receive and decodes messages.
  
Below is the screenshot with the explanation described above
[image]

### The serial output which indicates the error happened.

```shell
Protocol=UNKNOWN 30 bits (incl. gap and start) received
Received noise or an unknown (or not yet enabled) protocol
rawData[60]:
 -3276750
 + 850,-4500
 + 500,- 550 + 450,- 550 + 450,- 550 + 450,- 550
 + 450,- 550 + 450,- 550 + 450,- 550 + 450,- 550
 + 450,-1700 + 550,-3450 + 550,-3450 + 550,-3450
 + 550,-2300 + 450,- 550 + 450,-1700 + 550,- 550
 + 600,-1650 + 600,-1650 + 600,- 550 + 550,- 550
 + 600,-1650 + 200,-1700 + 550,- 550 + 600,-1650
 + 600,- 550 + 550,- 550 + 600,-1650 + 550,-1700
 + 200
Sum: 55850

Protocol=UNKNOWN 1 bits (incl. gap and start) received
Received noise or an unknown (or not yet enabled) protocol
rawData[2]:
 -1000050
 +1950
Sum: 1950

Protocol=NEC Address=0x4 Command=0x2 Raw-Data=0xFD02FB04 32 bits LSB first
Send with: IrSender.sendNEC(0x4, 0x2, <numberOfRepeats>);

Protocol=NEC Address=0x4 Command=0x2 Raw-Data=0xFD02FB04 32 bits LSB first
Send with: IrSender.sendNEC(0x4, 0x2, <numberOfRepeats>);

Protocol=NEC Address=0x4 Command=0x3 Raw-Data=0xFC03FB04 32 bits LSB first
Send with: IrSender.sendNEC(0x4, 0x3, <numberOfRepeats>);

Protocol=NEC Address=0x4 Command=0x3 Repeat gap=40250us

Protocol=NEC Address=0x4 Command=0x3 Raw-Data=0xFC03FB04 32 bits LSB first
Send with: IrSender.sendNEC(0x4, 0x3, <numberOfRepeats>);

Protocol=NEC Address=0x4 Command=0x3 Repeat gap=40250us
```

### What were you expecting to happen?

The sender is expected to send messages and the receiver is expected to decode and print the same ones as in the screenshot below
[image]

### Additional context.

With Arduino Nano, with the same code - **everything is working fine as expected (data was sent  and printed out every 1000ms)**, see a screenshot from the receiver terminal below 

[image]

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

milan1984, 2023-08-01:
Just tested with tag 3.9.0, - everything is working fine

closed 2023-08-01 by ArminJo

ArminJo, 2023-08-01:
@milan1984 Can you please check with latest version from repository and give feedback? Afterwards I can release the fix 😀 .

commit 2a801a6b (2023-08-01): Fixes #1152
commit c2a3858a (2023-08-01): Fixes #1152