Arduino-IRremote/Arduino-IRremote #1165: Not working on Arduino Arduino r4 wifi

meeeeek, 2023-08-31:
### The type of board you are using.

Arduino ATmega328* board (Uno, Nano, etc.)

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

Arduino r4 wifi

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [X] ReceiveAndSend
- [ ] SimpleSender
- [X] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

Used many different sample codes.
Loaded correctly in arduino 
When arduino should send instruction via IR led (3 pin), the module's red led works(just if i use digitalWrite()) but this doesn't send the Instruction.
Tried with different devices and different IR leds.
I'm almost sure that the problem is the board compatibility 

### What happens?

same as i wrote before

### What were you expecting to happen?

Turning on tv via hex code

### Additional context.

//RECEIVER INPUT 
//Protocol=NEC Address=0xBF00 Command=0xD Raw-Data=0xF20DBF00 32 bits LSB first

#include <IRremote.h>   
IRsend irsend;
                               
void setup()  
{  
  pinMode(3,OUTPUT); // i also tried without this line 
}  
                               
void loop()  
{  
  irsend.sendNEC(0xF20DBF00, 32); // Turn on tv
  digitalWrite(3,HIGH);
  delay(2000); 
  irsend.sendNEC(0xF20DBF00, 32);
    digitalWrite(3,LOW);

  delay(2000);
 } 
[image]

} 
[image]

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2023-08-31:
If you send me a R4, I can test it.

closed 2023-08-31 by ArminJo

meeeeek, 2023-09-01:
> If you send me a R4, I can test it.

if you send me 50 $ i'll send you